openthread/openthread #8625: Crash on publishing on-link prefix in posix

thcu-gp, 2023-01-06:
**Describe the bug** 
When running an  rcp posix build, the following   happens:
```
Jan  6 14:12:57 raspberrypi-3177 otbr-agent[187]: 00:00:40.706 [I] BorderRouter--: Start evaluating routing policy, scheduled in 429 milliseconds
Jan  6 14:12:58 raspberrypi-3177 otbr-agent[187]: 00:00:41.135 [I] BorderRouter--: Evaluating routing policy
Jan  6 14:12:58 raspberrypi-3177 otbr-agent[187]: 00:00:41.142 [I] Settings------: Added BrOnLinkPrefixes entry {prefix:::/0,lifetime:0}
Jan  6 14:12:58 raspberrypi-3177 otbr-agent[187]: 00:00:41.143 [I] BorderRouter--: Publishing local on-link prefix fd00:db8:0:0::/64 in netdata
Jan  6 14:12:58 raspberrypi-3177 otbr-agent[187]: 00:00:41.143 [I] BorderRouter--: THCU: PublishExternalRoute: fd00:db8:0:0::/64
Jan  6 14:12:58 raspberrypi-3177 otbr-agent[187]: 00:00:41.143 [C] Platform------: ------------------ BEGINNING OF CRASH -------------
Jan  6 14:12:58 raspberrypi-3177 otbr-agent[187]: 00:00:41.144 [C] Platform------: *** FATAL ERROR: Caught signal: 6 (Aborted)
Jan  6 14:12:58 raspberrypi-3177 otbr-agent[187]: 00:00:41.144 [C] Platform------: ------------------ END OF CRASH ------------------
```
It's the Mesh local prefix that is picked up as the on-link prefix and then the stack tries to publish it as an external route, which fails (the IsValid) check

**To Reproduce**

1. Git commit 9af0bfa60
2. posix - qorvo rcp
3. otbr build
4. BH 1_1 test BR_1 dut

abtink, 2023-01-27:
Thanks @thcu-gp .

How did you form the network? How was the operational dataset created?

Generally, the mesh-local prefix should be generated randomly.
(e.g. `dataset init new` to generate random, `dataset commit active` to set it).

Looks like somehow in your setup the mesh-local prefix was set to same prefix as the local on-link prefix" (which is supposed to be derived from extended pan ID).

jwhui, 2023-03-02:
Closing stale issue.

closed 2023-03-02 by jwhui