openthread/openthread #8641: CoAP block2 responses are missing the token after first block response

tpierson86, 2023-01-10:
CoAP Block2 requests are not working after the first block response is sent. This issue is similar to #7976, only that the token is missing on Block2 responses, therefore the fix from #7980 does not apply to block2 requests. 

To reproduce, create any block2 request greater than the block size (e.g. set block size to 1024 bytes and send 1500 bytes). The Block1 ACK response will have a missing token (Token Length = 0).

[image]

I propose the following fix:

[coap.zip](https://github.com/openthread/openthread/files/10386327/coap.zip)
File name: src\core\coap.cpp
method name: CoapBase::ProcessBlock2Request
```
response->SetMessageId(aMessage.GetMessageId());

++ SuccessOrExit(error = response->SetTokenFromMessage(aMessage));

VerifyOrExit((bufLen = otCoapBlockSizeFromExponent(aMessage.GetBlockWiseBlockSize())) <= kMaxBlockLength,
                 error = kErrorNoBufs);
```

jwhui, 2023-01-26:
@tbitcs , thanks for raising this issue. Would you like to submit a PR?

tbitcs, 2023-01-26:
@jwhui, PR #8699 submitted.

tbitcs, 2023-01-27:
@jwhui I had to create a new fork and PR due to cla issues. Please see #8700.

closed 2023-02-06 by jwhui