openthread/openthread #8663: Client doesn’t resolve a hostname when pinging a hostname

dingdang2012, 2023-01-17:
When pinging a hostname, Error 6: Parse is observed.
Build Steps:
a) git clone https://github.com/openthread/ot-nrf528xx.git --recursive
b) cd ot-nrf528xx
c) ./script/bootstrap
b) script/build nrf52840 USB_trans -DOT_SRP_CLIENT=ON -DOT_ECDSA=ON
c) arm-none-eabi-objcopy -O ihex build/bin/ot-cli-ftd ot-cli-ftd.hex
d) nrfjprog -f nrf52 --chiperase --program ot-cli-ftd.hex --reset
e)
GitHubGitHub
[GitHub - openthread/ot-nrf528xx: OpenThread on Nordic nRF528xx examples.](https://github.com/openthread/ot-nrf528xx.git)
OpenThread on Nordic nRF528xx examples. Contribute to openthread/ot-nrf528xx development by creating an account on GitHub. (53 kB)

jwhui, 2023-01-17:
Can you provide more information?
1) A copy of your console showing command and resulting output.
2) What Thread Border Router are you using? How did you configure it?
3) Packet capture for analysis in Wireshark.
4) DEBG-level OpenThread logs on both end device and border router.

jwhui, 2023-01-17:
Also note that the `ping` CLI command only accepts IP addresses for the destination, and not a host name.

See the [ping CLI documentation](https://github.com/openthread/openthread/issues/new?assignees=&labels=&template=bug_report.md) for more details.