openthread/openthread #8681: Thread v1.3 Child updates RxOnWhenIdle from False to True to a Thread v1.1 Parent

AlanLCollins, 2023-01-23:
**Describe the bug** 
Communication between Parent-Child fails after CHILD update message to change RxOnWhenIdle from False to True.

**To Reproduce** Information to reproduce the behavior, including:
- create a Thread network by thread v1.1 parent device, 
- join thread v1.3 child device as sleepy device
- wait until child device updates rxOnWhenIdle = true

Parent information: 
github/ot-br-posix (79a0893eec7af03ae3bbe51fa9e74f469b88f15a)  &
github/openthread (61d9e72c18588d6ee3a9b9a2720f65e7d39bdb87)

**IEEE 802.15.4 hardware platform**: 
Qorvo GP712
**Build steps**
compiles as Thread v1.1 
-DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1  
**Network topology**: 
2 nodes Parent-Child

**Console/log output** If applicable, add console/log output to help explain your problem.
[parent-child-update-rxonwhenidle.pcap.zip](https://github.com/openthread/openthread/files/10484075/parent-child-update-rxonwhenidle.pcap.zip)
[parent_grep_CHIP_API_otbr-agent.txt](https://github.com/openthread/openthread/files/10484076/parent_grep_CHIP_API_otbr-agent.txt)
Network key = 011f81976e955a380ce6e59740e8af09 

**Additional context** Add any other context about the problem here.
sniffer packet 754 shows the "child update request" that triggers the failure state. 
corresponding to Parent logs: 
`12-13 14:43:42.109 I/otbr-agent( 1813): [I] MeshForwarder-: Received IPv6 UDP msg, len:107, chksum:5206, ecn:no, from:d6528a1678c67da8, sec:no, prio:net, rss:-56.0`
all subsequent "child update responses" are ignored.

abtink, 2023-01-23:
I checked the logs around the timestamp. I don't see anything wrong on parent side. It looks like there may be some issue on child side (it may be rebooting/resetting around the same time). 

On parent, we see the "Child Update Request" from child which now says mode changed to `rx-on:yes`
```
12-13 14:43:42.110 I/otbr-agent( 1813): [I] Mle-----------: Receive Child Update Request from child (fe80:0:0:0:d452:8a16:78c6:7da8)
12-13 14:43:42.110 I/otbr-agent( 1813): [I] Mle-----------: Child 0x8801 IPv6 address[1]=fdde:ad00:beef:0:2500:7d84:659c:6b8
12-13 14:43:42.110 I/otbr-agent( 1813): [I] Mle-----------: Child 0x8801 IPv6 address[2]=fd45:6c32:296e:1:449e:c647:93f1:4d40
12-13 14:43:42.113 I/otbr-agent( 1813): [I] Mle-----------: Child 0x8801 has 2 registered IPv6 addresses, 2 addresses stored
12-13 14:43:42.113 I/otbr-agent( 1813): [N] Mle-----------: Child 0x8801 mode change 0x04 -> 0x0c [rx-on:yes ftd:no full-net:no]
```
Note that the child has rloc16 `0x8801`, ext-address  `d6528a1678c67da` and link-local `fe80:0:0:0:d452:8a16:78c6:7da8` address along with `fdde:ad00:beef:0:2500:7d84:659c:6b8` and `fdde:ad00:beef:0:2500:7d84:659c:6b8`

Parent prepares and sends a response but gets no ack from the child:
```
12-13 14:43:42.119 I/otbr-agent( 1813): [I] Mle-----------: Send Child Update Response to child (fe80:0:0:0:d452:8a16:78c6:7da8,0x8801)
12-13 14:43:42.328 I/otbr-agent( 1813): [I] Mac-----------: Frame tx attempt 16/16 failed, error:NoAck, len:82, seqnum:238, type:Data, src:ee83fb606c45044d, dst:d6528a1678c67da8, sec:no, ac
12-13 14:43:42.329 I/otbr-agent( 1813): [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:98, chksum:7cfe, ecn:no, to:d6528a1678c67da8, sec:no, error:NoAck, prio:net
12-13 14:43:42.329 I/otbr-agent( 1813): [N] MeshForwarder-:     src:[fe80:0:0:0:ec83:fb60:6c45:44d]:19788
12-13 14:43:42.329 I/otbr-agent( 1813): [N] MeshForwarder-:     dst:[fe80:0:0:0:d452:8a16:78c6:7da8]:19788
```

Other msg to the child `0x8801` around same time also get no ack:
```
12-13 14:43:42.557 I/otbr-agent( 1813): [I] SrpServer-----: Send success response with granted lease: 7200 and key lease: 680400
12-13 14:43:42.787 I/otbr-agent( 1813): [I] Mac-----------: Frame tx attempt 16/16 failed, error:NoAck, len:84, seqnum:239, type:Data, src:0x8800, dst:0x8801, sec:yes, ackreq:yes
12-13 14:43:42.788 I/otbr-agent( 1813): [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:83, chksum:fca7, ecn:no, to:0x8801, sec:yes, error:NoAck, prio:low
```

So looks like child may be off. We see in a bit later that child tries to perform "discover scan" (note `fe80:0:0:0:d452:8a16:78c6:7da8` is address of the child)
```
12-13 14:43:44.873 I/otbr-agent( 1813): [I] Mle-----------: Receive Discovery Request (fe80:0:0:0:d452:8a16:78c6:7da8)
12-13 14:43:44.874 I/otbr-agent( 1813): [I] Mle-----------: Delay Discovery Response (fe80:0:0:0:d452:8a16:78c6:7da8)
```

Then tries to establish its link with parent as sleepy which all works:
```
12-13 14:43:47.102 I/otbr-agent( 1813): [I] Mle-----------: Receive Child Update Request from child (fe80:0:0:0:d452:8a16:78c6:7da8)
...
12-13 14:43:47.102 I/otbr-agent( 1813): [N] Mle-----------: Child 0x8801 mode change 0x0c -> 0x04 [rx-on:no ftd:no full-net:no]
...
12-13 14:43:47.107 I/otbr-agent( 1813): [I] Mle-----------: Send Child Update Response to child (fe80:0:0:0:d452:8a16:78c6:7da8,0x8801)
12-13 14:43:47.128 I/otbr-agent( 1813): [I] MeshForwarder-: Prepping indir tx IPv6 UDP msg, len:98, chksum:ac67, ecn:no, to:0x8801, sec:no, prio:net
12-13 14:43:47.128 I/otbr-agent( 1813): [I] DataPollHandlr: Rx data poll, src:0x8801, qed_msgs:1, rss:-56, ack-fp:1
12-13 14:43:47.162 I/otbr-agent( 1813): [I] MeshForwarder-: Sent IPv6 UDP msg, len:98, chksum:73f9, ecn:no, to:d6528a1678c67da8, sec:no, prio:net
```

jwhui, 2023-01-26:
@AlanLCollins , any insight into what might be triggering the device to send an MLE Discovery Request? Is this due to a Matter `ScanNetworks` command? For example here: https://github.com/project-chip/connectedhomeip/pull/21279

AlanLCollins, 2023-01-26:
yes. the discovery requests from the accessory device are related to Matter ScanNetworks command.

abtink, 2023-01-26:
So looks like this may be the issue: Child is scanning and switches to different rx channels, so does not rx and ack the "Child Update Response" from parent. After scan is done things start working again.

jwhui, 2023-01-26:
> I think we need the logs from the accessory device because the Discovery Request may not be related to a ScanNetwork command, but a mechanism to recover from the lost network.

I don't understand the need to perform a network scan if the device detaches from the network. It should simply use MLE Parent Request (and possibly MLE Announce) messages.

> the spec allows going RxOnWhenIdle from false to true, but it prohibits going from True to False. I am guessing the only way to do that is by reconnecting to the parent and send a new Child Id request.

The Thread Spec allows going from non-sleepy to sleepy if the device originally attached as a sleepy.

AlanLCollins, 2023-01-26:
Perhaps I got an old version of the spec. I am referring to 4.6.1 : 
[image]

jwhui, 2023-01-26:
This line allows a device that attached as rx-off-when-idle to change back and forth:

> If it initially attached with the "R" bit value '0' (Rx-off-when-idle) it MAY modify the "R" bit value.

AlanLCollins, 2023-01-26:
I see, I understand now. So if the Child request was done with R = 0, then later we can update it to whichever we want to.

AlanLCollins, 2023-01-28:
offline conversation with @LuDuda: MLE discovery is process allowed only for RxOnWhenIdle=true devices, that's the reason the sensor sends the Child update request. 
@abtink analysis seems correct, the MLE Discovery switches the sensor to other channels so it misses the child update response. 
We have identified a potential improvement to have a 100-200ms delay before going into MLE discovery, so sensor catches the child update response (or something similar). We will run some tests to confirm assumptions.
There are other open questions like: Why the sensor decides to deattach and attach again (Parent request). We will continue investigating. 
All these have nothing to do with v1.3 vs v1.1 so I will modify the title of this ticket.

AlanLCollins, 2023-01-28:
Parent removes child from tables due to failures on Child update response:
```12-13 14:43:42.113 I/otbr-agent( 1813): [N] Mle-----------: Child 0x8801 mode change 0x04 -> 0x0c [rx-on:yes ftd:no full-net:no]
12-13 14:43:42.116 I/otbr-agent( 1813): [I] Settings------: Read ChildInfo {rloc:0x8801, extaddr:d6528a1678c67da8, timeout:240, mode:0x04, version:4}
12-13 14:43:42.117 I/otbr-agent( 1813): [I] Settings------: Removed ChildInfo {rloc:0x8801, extaddr:d6528a1678c67da8, timeout:240, mode:0x04, version:4}
12-13 14:43:42.119 I/otbr-agent( 1813): [I] Settings------: Added ChildInfo {rloc:0x8801, extaddr:d6528a1678c67da8, timeout:240, mode:0x0c, version:4}
12-13 14:43:42.119 I/otbr-agent( 1813): [I] Mle-----------: Send Child Update Response to child (fe80:0:0:0:d452:8a16:78c6:7da8,0x8801)```

abtink, 2023-01-28:
@AlanLCollins the log snippet does not indicate that parent removed the child. It is just updating the `ChildInfo` saved in `Settings` (non-volatile memory). It is removing the previous `ChildInfo` with old mode `mode:0x04` (from `Settings`) and saving/adding it with new mode `mode:0x0c`.

jwhui, 2023-05-09:
Resolved by #8832

closed 2023-05-09 by jwhui