openthread/openthread #8726: Border Routers fail to communicate with leader

pdgendt, 2023-02-06:
**Describe the bug**

In a setup with multiple (6) TREL enabled border routers, after some time (25d in our case), routers are unable to communicate with the leader node. Netdata updates aren't propagated, pings timeout (both directions), ...
Wireshark shows TREL traffic on the leader node, but no application layer communication.

Restarting `otbr-agent` on the leader fixes communication.

**To Reproduce**

The issue appears to be random (uptime was 25d).

1. Git commit id
ot-br-posix `c5011963d0a527d58c9d3651be9fb1a8ef4286e6`
openthread `23b5b2845fe651e52fb372fc0fa2db94b3d2953f`

2. IEEE 802.15.4 hardware platform
RCP nrf52840 connected using UART

3. Build steps
`cmake` arguments:
```
            -DOT_APP_NCP=OFF \
            -DOT_APP_CLI=OFF \
            -DOT_APP_RCP=OFF \
            -DOT_CHANNEL_MANAGER=ON \
            -DOT_CHANNEL_MONITOR=ON \
            -DOT_LINK_METRICS_INITIATOR=ON \
            -DOT_SERVICE=ON \
            -DOT_RCP_RESTORATION_MAX_COUNT=10 \
            -DOT_TREL=ON \
            -DOT_FIREWALL=OFF \
            -DOTBR_TREL=ON \
            -DOTBR_BORDER_ROUTING=ON \
            -DOTBR_INFRA_IF_NAME=eth0 \
            -DOT_CONFIG="$(OT_BR_POSIX_PKGDIR)/ot-config.h" \
            -DBUILD_SHARED_LIBS=OFF \
            -DBUILD_TESTING=OFF
```

```c
// ot-config.h

#define OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS 4096

// TREL: allows OpenThread to forward packets over IP
#define OPENTHREAD_POSIX_CONFIG_TREL_UDP_PORT htons(7531)

// Increase amount of services we can add
#define OPENTHREAD_CONFIG_TMF_NETDATA_SERVICE_MAX_ALOCS 5

// Disable way too verbose meshforwarder logging
#define OPENTHREAD_CONFIG_LOG_SRC_DST_IP_ADDRESSES 0

#include <openthread-core-posix-config.h>

```

4. Network topology
6 Border routers on the same Thread partition, and same LAN

**Expected behavior**

We should be able to ping the leader using IPv6.

**Additional context**

Weirdly enough toggling TREL allowed for a few pings to arrive:
```
# ping -I wpan0 fdde:ad00:beef:0:0:ff:fe00:5800
PING fdde:ad00:beef:0:0:ff:fe00:5800 (fdde:ad00:beef::ff:fe00:5800): 56 data bytes
[ trel disable ]
[ trel enable ]
64 bytes from fdde:ad00:beef::ff:fe00:5800: seq=13 ttl=64 time=29.389 ms
64 bytes from fdde:ad00:beef::ff:fe00:5800: seq=14 ttl=64 time=11.353 ms
64 bytes from fdde:ad00:beef::ff:fe00:5800: seq=15 ttl=64 time=11.231 ms
[ trel disable ]
[ trel enable ]
64 bytes from fdde:ad00:beef::ff:fe00:5800: seq=30 ttl=64 time=12.146 ms
^C
--- fdde:ad00:beef:0:0:ff:fe00:5800 ping statistics ---
36 packets transmitted, 4 packets received, 88% packet loss
round-trip min/avg/max = 11.231/16.029/29.389 ms
```

The router table on the leader has weird `0000000000000000` entries (also after restarting otbr-agent):
```
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
|  3 | 0x0c00 |       12 |         1 |     3 |      3 |  12 | 1ed32909b910f46c |    1 |
| 11 | 0x2c00 |       62 |         1 |     0 |      0 | 152 | 0000000000000000 |    0 |   <-- ???
| 12 | 0x3000 |        3 |         1 |     3 |      3 |  11 | aac263762c9f2348 |    1 |
| 22 | 0x5800 |       63 |         0 |     0 |      0 |   0 | 0000000000000000 |    0 |   <-- leader/self
| 25 | 0x6400 |        3 |         1 |     3 |      3 |   0 | 563ff46e201740cf |    1 |
| 46 | 0xb800 |       12 |         1 |     3 |      3 |  10 | 624414a166b9c75e |    1 |
| 47 | 0xbc00 |       12 |         1 |     3 |      3 |   8 | 6e4fa81098cf140f |    1 |
| 62 | 0xf800 |       12 |         1 |     3 |      3 |  10 | 6e3e42227f3bfc93 |    1 |

Done
```

Other router shows a "correct" table:
```
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
|  3 | 0x0c00 |       12 |         1 |     3 |      3 |  20 | 1ed32909b910f46c |    1 |
| 11 | 0x2c00 |       62 |         1 |     3 |      2 |   4 | 47587f263e104834 |    1 |
| 12 | 0x3000 |       62 |         1 |     3 |      3 |  13 | aac263762c9f2348 |    1 |
| 22 | 0x5800 |        3 |         1 |     3 |      3 |  16 | 92dc22ca2eca9af5 |    1 |   <-- leader
| 25 | 0x6400 |       12 |         1 |     3 |      3 |  19 | 563ff46e201740cf |    1 |
| 46 | 0xb800 |       12 |         1 |     3 |      3 |  14 | 624414a166b9c75e |    1 |
| 47 | 0xbc00 |       63 |         0 |     0 |      0 |   0 | 6e4fa81098cf140f |    0 |
| 62 | 0xf800 |       12 |         1 |     3 |      3 |  21 | 6e3e42227f3bfc93 |    1 |

Done
```

Some more output on the leader:
```
> history netinfo
| Age                  | Role     | Mode | RLOC16 | Partition ID |
+----------------------+----------+------+--------+--------------+
| 25 days 09:01:49.923 | leader   | rdn  | 0x5800 |   1202407051 |
| 25 days 09:01:49.975 | detached | rdn  | 0x5800 |            0 |
Done

> ipaddr
fd64:9a0c:5508:1:7e6c:aa60:19ba:7f96
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:5800
fdde:ad00:beef:0:15f5:75:1551:4923
fe80:0:0:0:90dc:22ca:2eca:9af5
Done

> counters mac
TxTotal: 4512995
    TxUnicast: 2602951
    TxBroadcast: 1910044
    TxAckRequested: 2602950
    TxAcked: 2519924
    TxNoAckRequested: 1910045
    TxData: 3505283
    TxDataPoll: 0
    TxBeacon: 1
    TxBeaconRequest: 16
    TxOther: 0
    TxRetry: 0
    TxErrCca: 436
    TxErrBusyChannel: 436
    TxErrAbort: 72866
    TxDirectMaxRetryExpiry: 30541
    TxIndirectMaxRetryExpiry: 0
RxTotal: 11891272
    RxUnicast: 3943420
    RxBroadcast: 7938944
    RxData: 8869129
    RxDataPoll: 0
    RxBeacon: 15
    RxBeaconRequest: 1
    RxOther: 0
    RxAddressFiltered: 0
    RxDestAddrFiltered: 0
    RxDuplicated: 88994
    RxErrNoFrame: 0
    RxErrNoUnknownNeighbor: 8907
    RxErrInvalidSrcAddr: 0
    RxErrSec: 2924224
    RxErrFcs: 0
    RxErrOther: 2
Done

> counters ip
TxSuccess: 3346205
TxFailed: 30541
RxSuccess: 7703116
RxFailed: 471
Done
```

abtink, 2023-02-06:
One quick point, the `router table` output looks fine (the `00000` mac addresses is fine when `Link` columon is `0`).

- This command gives list of all currently allocated router IDs with Thread mesh. 
    - Some of the routers may also be a neighbor of the device and some can be hops away (not a neighbor).
- The Ext MAC Address is known for routers with which device has established a link (i.e. they are a neighbor as well as router).

pdgendt, 2023-02-06:
> One quick point, the `router table` output looks fine (the `00000` mac addresses is fine when `Link` columon is `0`).

Thanks @abtink for the quick response, is it also normal to show `0000000000000000` for your own device (as is the case with the leader node)?

When we would be able to reproduce the issue again, what information could help narrow down the cause?

abtink, 2023-02-06:
> is it also normal to show 0000000000000000 for your own device (as is the case with the leader node)?

Yes. That's expected. And we should see `Link` column as zero for device itself (we technically don't establish a link to ourself :)).

----

Things I can suggest:
- `history neighbor` to see history of neighbor table changes (would be good to on differnt nodes)
- `history netinfo` on all nodes (if they detach/attach?)
- `router table` is useful (on all devices) - to see the next hop and links between routers.

I am not sure if this may be trel related but these can be helpful:
- `trel peers` to see list of discovered trel peers 
- `multiradio neighbor list` - to list neighbors and tell which radios they support (and we use to talk to them).

LanderN, 2023-02-22:
We encountered the issue again in a setup with two TREL-enabled border routers, 2 MTDs and 1 FTD and got the output you suggested:

**Router 1:**
```shellsession
# ot-ctl extaddr
0ebccca7fd3e7b15
Done
# ot-ctl state
leader
Done
# ot-ctl history neighbor
| Age                  | Type   | Event     | Extended Address | RLOC16 | Mode | Ave RSS |
+----------------------+--------+-----------+------------------+--------+------+---------+
|         00:11:11.236 | Child  | Added     | b2fa06bf4804b6d2 | 0xac02 | rn   |     -75 |
|         00:13:21.869 | Child  | Added     | 12347fa3f8d49383 | 0xac01 | rn   |     -70 |
|         00:16:37.966 | Child  | Removed   | b2fa06bf4804b6d2 | 0xac24 | rn   |     -71 |
|         00:17:56.877 | Child  | Added     | b2fa06bf4804b6d2 | 0xac24 | rn   |     -68 |
|         00:23:19.564 | Router | Added     | c27df1d1e8bd2418 | 0x9800 | rdn  |     -48 |
|         00:25:42.050 | Router | Added     | 9a2f0a4a14746790 | 0xcc00 | rdn  |     -40 |
|         00:25:42.141 | Child  | Restoring | b2fa06bf4804b6d2 | 0xac24 | rn   |     127 |
Done
# ot-ctl history netinfo
| Age                  | Role     | Mode | RLOC16 | Partition ID |
+----------------------+----------+------+--------+--------------+
|         00:25:45.554 | leader   | rdn  | 0xac00 |   1980626451 |
|         00:25:45.600 | detached | rdn  | 0xac00 |            0 |
Done
# ot-ctl router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
| 38 | 0x9800 |       51 |         1 |     3 |      3 |  14 | c27df1d1e8bd2418 |    1 |
| 43 | 0xac00 |       63 |         0 |     0 |      0 |   0 | 0000000000000000 |    0 |
| 51 | 0xcc00 |       38 |         1 |     3 |      3 |   7 | 9a2f0a4a14746790 |    1 |

Done
# ot-ctl trel peers
| No  | Ext MAC Address  | Ext PAN Id       | IPv6 Socket Address                              |
+-----+------------------+------------------+--------------------------------------------------+
|   1 | c27df1d1e8bd2418 | 97ef8474e4248428 | [fd97:ef84:74e4:8428:522d:f4ff:fe08:8f15]:7531   |
Done
# ot-ctl multiradio neighbor list
ExtAddr:12347fa3f8d49383, RLOC16:0xac01, Radios:[15.4(150)]
ExtAddr:b2fa06bf4804b6d2, RLOC16:0xac02, Radios:[15.4(25)]
ExtAddr:c27df1d1e8bd2418, RLOC16:0x9800, Radios:[15.4(255), TREL(0)]
ExtAddr:9a2f0a4a14746790, RLOC16:0xcc00, Radios:[15.4(255)]
Done
# avahi-browse -tr _trel._udp
+  wpan0 IPv6 0ebccca7fd3e7b15                              _trel._udp           local
+   eth0 IPv6 c27df1d1e8bd2418                              _trel._udp           local
+   eth0 IPv6 0ebccca7fd3e7b15                              _trel._udp           local
+   eth0 IPv4 c27df1d1e8bd2418                              _trel._udp           local
+   eth0 IPv4 0ebccca7fd3e7b15                              _trel._udp           local
+     lo IPv4 0ebccca7fd3e7b15                              _trel._udp           local
=  wpan0 IPv6 0ebccca7fd3e7b15                              _trel._udp           local
   hostname = [basalte.local]
   address = [fdca:6ae0:fb95:1:9c30:67e8:9dc8:98a]
   port = [7531]
   txt = ["xp=t$(" "xa=\014̧>{\021"]
=   eth0 IPv6 0ebccca7fd3e7b15                              _trel._udp           local
   hostname = [basalte.local]
   address = [fd97:ef84:74e4:8428:fadc:7aff:fe90:4185]
   port = [7531]
   txt = ["xp=t$(" "xa=\014̧>{\021"]
=   eth0 IPv6 c27df1d1e8bd2418                              _trel._udp           local
   hostname = [basalte-2.local]
   address = [fd97:ef84:74e4:8428:522d:f4ff:fe08:8f15]
   port = [7531]
   txt = ["xp=t$(" "xa=}$\024"]
=   eth0 IPv4 0ebccca7fd3e7b15                              _trel._udp           local
   hostname = [basalte.local]
[... 180 lines cut ...]
       valid_lft forever preferred_lft forever
3: wpan0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1280 qdisc pfifo_fast qlen 500
    link/[65534] 
    inet6 fd97:ef84:74e4::ff:fe00:fc10/64 scope global deprecated flags 02 
       valid_lft forever preferred_lft 0sec
    inet6 fdca:6ae0:fb95:1:5284:2a37:7ae0:fc62/64 scope global flags 02 
       valid_lft forever preferred_lft forever
    inet6 fd97:ef84:74e4::ff:fe00:9800/64 scope global deprecated flags 02 
       valid_lft forever preferred_lft 0sec
    inet6 fd97:ef84:74e4:0:a6ff:e3d4:fdb2:a9d6/64 scope global deprecated flags 02 
       valid_lft forever preferred_lft 0sec
    inet6 fe80::c07d:f1d1:e8bd:2418/64 scope link flags 02 
       valid_lft forever preferred_lft forever
# ping -c5 fd97:ef84:74e4:8428:fadc:7aff:fe90:4185 # pinging over eth0 works
PING fd97:ef84:74e4:8428:fadc:7aff:fe90:4185 (fd97:ef84:74e4:8428:fadc:7aff:fe90:4185): 56 data bytes
64 bytes from fd97:ef84:74e4:8428:fadc:7aff:fe90:4185: seq=0 ttl=64 time=5.249 ms
64 bytes from fd97:ef84:74e4:8428:fadc:7aff:fe90:4185: seq=1 ttl=64 time=1.739 ms
64 bytes from fd97:ef84:74e4:8428:fadc:7aff:fe90:4185: seq=2 ttl=64 time=1.861 ms
64 bytes from fd97:ef84:74e4:8428:fadc:7aff:fe90:4185: seq=3 ttl=64 time=1.740 ms
64 bytes from fd97:ef84:74e4:8428:fadc:7aff:fe90:4185: seq=4 ttl=64 time=1.861 ms

--- fd97:ef84:74e4:8428:fadc:7aff:fe90:4185 ping statistics ---
5 packets transmitted, 5 packets received, 0% packet loss
round-trip min/avg/max = 1.739/2.490/5.249 ms
```

abtink, 2023-02-22:
I don't see anything immediately bad with different properties.

They both have each other in their `router table` and seem to have link established.

One thing I noticed is `ExtAddr:0ebccca7fd3e7b15, RLOC16:0xac00, Radios:[15.4(255), TREL(0)]`  (preference `255` for 15.4 and zero for TREL)
- This indicates it would use 15.4 for this neighbor.
   - Somehow device could not talk over wifi/TREL. This aligns with the fact that `ot-ctl trel peers` is empty on `router 2`
- We see same on `router 1` -> `ExtAddr:c27df1d1e8bd2418, RLOC16:0x9800, Radios:[15.4(255), TREL(0)]`
- One question is why this is happening?

But independent of this, I expect the ping message to go over 15.4 radio.

I can suggest maybe collecting device OT logs when in this state (at `INFO` or `DEBG` levels)? It may provide more info.
And/or maybe sniffing 15.4 to capture messages?

jwhui, 2023-05-09:
Closing stale issue.

closed 2023-05-09 by jwhui