openthread/openthread #8738: CSL sampling window for CSL Receiver (SSED) is too tight

EskoDijk, 2023-02-09:
**Describe the bug** A clear and concise description of what the bug is.
The start of the CSL sampling window is very tightly calculated based on the CSL Accuracy reported parameters by the Parent. This may lead to interoperability issues, especially if the device (CSL Transmitter) sending the CSL Accuracy TLV takes the description in the spec (4.4.33) rather literally.  The description (for CslAccuracy and CslUncertainty, at the bottom) in the spec does not mention these values must be worst-case which could lead to a device sending its typical CSL clock uncertainty, not the worst-case one.

Another issue is that the exact interpretation of what happens at the end of CSL Phase may differ among implementers per SPEC-1154.  This may cause a Parent CSL Transmitter to be 192 us early w.r.t. to the OpenThread receiver.

Consequence: the CSL transmission is not received as it could arrive too early at the CSL Receiver.

**To Reproduce** Information to reproduce the behavior, including:

See Thread CSL test captures.

**Expected behavior** A clear and concise description of what you expected to happen.

CSL transmission is received, even if reported CSL Accuracy values are slightly off or vendor of Parent uses a slightly different interpretation of CSL Phase.

The proposal is to add some additional "slack" time that allows for the variation. E.g.  `SubMac::GetCslWindowEdges()` starts receiving window a fixed 256 us earlier. This is defined separate from the vendor's RF RX ramp up time. 
That is, it is insufficient to just adapt the value kCslReceiveTimeAhead to be higher. In the call `Get<Radio>().ReceiveAt()` in sub_mac.cpp the time kCslReceiveTimeAhead is not included in determining the start of Rx time. So it needs to be a separate constant that's also applied in the `ReceiveAt()` call start time.

Background:
```
static constexpr uint32_t kCslReceiveTimeAhead = OPENTHREAD_CONFIG_CSL_RECEIVE_TIME_AHEAD;
```

**Console/log output** If applicable, add console/log output to help explain your problem.
N/A

**Additional context** Add any other context about the problem here.
SPEC-1154

Irving-cl, 2023-02-10:
I submitted a PR. I'm doing some testing and will then ask for reviews.

edmont, 2023-02-10:
Hi, still investigating on the Nordic's side, but the proposed solution doesn't seem optimal:

- It increases receive window duration (more power consumption).
- Need to verify that the CSL Phase value injected in the CSL IE is matching the expected timing.

I'll keep you updated with our findings.

edmont, 2023-02-10:
Please check this alternative: https://github.com/openthread/openthread/pull/8743

Main differences are:

- Preamble is considered for the transmission as well.
- Receive window duration is not extended.
- Local radio uncertainty is taken into account for receive window duration.

EskoDijk, 2023-02-10:
@edmont I see you want to re-use the existing CSL uncertainty value (`GetCslUncertainty()` / `otPlatRadioGetCslUncertainty()`) for both CSL Transmitter and CSL receiver to express the uncertainty. That avoids having to add more functions for Tx vs Rx uncertainty separately.  But it does force this internal quantity to be defined in "multiples of 10 us" which would in principle not be needed. It could be defined as a value in units of 1 us, as it's just an internal compilation constant that's not sent over the wire.

In case we go for the "single value" approach I think the OT platform developers have to pay attention in case their platform supports compilation to both CSL Transmitter (FTD REED) and Receiver (MTD SSED) that for respective cases the right number is chosen. E.g. using preprocessor directives or by having multiple platform configurations.

Since it may differ what the CSL Tx scheduling uncertainty is, vs the Rx uncertainty.

So if I'm correct a CSL Receiver may have some uncertainty in the timestamp of the received CSL IE frame, e.g. the reception time may only be available in multiples of 16 us. In that case the CSL Rx Uncertainty value is at least 16 us. If defined in units of 1 us then the Rx Uncertainty constant would be `16`.  Using the current definition it has to be `2`.  
This seems close enough to not matter much.

> Receive window duration is not extended

Sounds ok, if I read the code correctly the `Get<Radio>().Sleep()` is not called if the radio is still in the process of receiving some frame , right?

edmont, 2023-02-13:
> @edmont I see you want to re-use the existing CSL uncertainty value (`GetCslUncertainty()` / `otPlatRadioGetCslUncertainty()`) for both CSL Transmitter and CSL receiver to express the uncertainty. 

The general idea is that the radio would have about the same uncertainty when scheduling reception or transmission operations. A granularity of 10 us is good enough in terms or energy consumption. And, likewise the parent uncertainty case, the API leaves the option open for more complex platform implementations which take into account current conditions to calculate varying with time values.

Ideally a separate API should be used for RX or TX uncertainty, but I'd say this is a fair optimization.

> Sounds ok, if I read the code correctly the `Get<Radio>().Sleep()` is not called if the radio is still in the process of receiving some frame , right?

Exactly.

EskoDijk, 2023-02-13:
One further thing to note on the CSL receiver uncertainty in the @edmont  PR is that it should be rounded "upwards". So say the receiver has a 1, 2 or 4, or 5 us uncertainty then its reported CSL uncertainty value must be `1` for 10 us. And if it has 11, 12, 14, 16 then the reported value becomes `2` for 20 us. Maybe we could/should adjust the calculation so that more intuitive rounding can take place (1, 2, 3, 4 -> `0`  and 5, 6, 7 -- 14 -> `1`, etc.)   This is easy to do by just adding +5 us to the calculation of CSLuncertainty value -> us. 

Also in the PR the receiver still has zero "slack" time, that is, if Parent reports its inaccuracy/uncertainty perfectly and Child knows its inaccurayc/uncertainty perfectly then it may occur that the transmission starts exactly at the time the reception is ready.  A slight imperfection here in the estimate of the parameters may cause the frame transmission to fail.

Maybe it should have either a "slack" time or "precaution" time in the receiver of say 4, 8 or 16 us, or else the transmitter or the receiver needs to add +1 to their CSL uncertainty value.  A transmitter currently may report '0' if it's timer has 1 us or 2 us resolution. The receiver should take this into account but currently it doesn't as there is zero slack.

edmont, 2023-02-13:
> So say the receiver has a 1, 2 or 4, or 5 us uncertainty then its reported CSL uncertainty value must be `1` for 10 us.

I don't consider this a big deal. In the worst case, assuming that someone would be able to make so accurate measurements that is certain to confirm that the receiver uncertainty is ±1 us, it would need to use the 10 us value which means the receive window duration is extended by "unneeded" 18 us extra. I'd say this is negligible in the total power consumption.

>  A slight imperfection here in the estimate of the parameters may cause the frame transmission to fail.

Wouldn't this mean some of the sides didn't report their real uncertainty?

> or the receiver needs to add +1 to their CSL uncertainty value

I'd say this would be a good practice to cover for this "slack" time.

>  A transmitter currently may report '0' if it's timer has 1 us or 2 us resolution.

I would expect this transmitter to report the minimum uncertainty value covering this case, as allowed by Thread Specification, which would be 10 us.

EskoDijk, 2023-02-13:
> Wouldn't this mean some of the sides didn't report their real uncertainty?
True, but reporting your uncertainty as '10 us' `1` when your real uncertainty is say 2 us is counter-intuitive for some. Reporting '0 us' `0` would be closer to the truth in fact. So what exactly is the real uncertainty vs what to report is the question here.

But agree here with you that letting both transmitter and receiver "round up" their CSL uncertainty is an easy solution and works. The spec would then also need to clarify this a bit.

Also the PR assumes that the first symbol of the MAC-header will occur at the time "T_phase", the time that can be calculated by (CSL-phase * 160) us. Just wanted to check that this is correct and intended.

EskoDijk, 2023-02-14:
Question on the current code and the PR: We discussed that a CSL receiver wouldn't stop receiving (i.e. go to sleep) when it was still busy receiving a frame.  If so then the kMinCslWindow can be made much shorter -- only long enough to receive the PHY header in principle so that the receiver can detect a frame Rx is ongoing and can report this state to OpenThread.

That would dramatically improve the energy efficiency. Or, is it the idea that OpenThread provides a default value -- e.g. for radios that don't detect that they are receiving something - or a default value that's only meant to be adapted by the platform implementers?

Irving-cl, 2023-02-14:
@EskoDijk 

> Sounds ok, if I read the code correctly the Get<Radio>().Sleep() is not called if the radio is still in the process of receiving some frame , right?

As I understand, Get<Radio>().Sleep() will still be called if the radio is receiving. But the Nordic platform will allow the receiving complete before turning to sleep. I think maybe we can configure the window based on platform. Currently OT provides a default value `OPENTHREAD_CONFIG_CSL_MIN_RECEIVE_ON`. That can be overriden by upper level parameters.

jwhui, 2023-04-03:
Resolved by #8743

closed 2023-04-03 by jwhui