openthread/openthread #8758: Provide more infomations for CLI's `udp bind & `udp connect`

erjiaqing, 2023-02-14:
- In POSIX, we can call bind with 0 as the port number, means bind a OS selected port, then we can call some other function to get the actual port binded. We can implement similar function for `ot-cli`.

This is useful when we don't care about the port used while we still need the actual address infomation (e.g. for testing, we may want to verify the source address / port)

**Proposed solution** Print a line of for address and the port bound, e.g.

```
> udp bind ::
Socket ready: [0:0:0:0:0:0:0:0]:54321
Done

> udp connect fdde:ad00:beef:0:bb1:ebd6:ad10:f33 1234
Socket ready: [0:0:0:0:0:0:0:0]:54321
Done
```

jwhui, 2023-03-14:
Getting more information about the UDP socket is useful.

Another option is to have another command that provides this information on demand. For example:
```
udp info
```

Thoughts?

jwhui, 2025-03-11:
Stale issue.