openthread/openthread #8773: Automatically install routes in Linux networking for on-link prefixes

gabekassel, 2023-02-17:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
[This PR ](https://github.com/openthread/openthread/pull/8315/files) changed OT behavior whereby OT creates an on-link prefix prior to checking whether one already exists. We found a situation where the existing infrastructure prefix was retained, but our system wasn't setting up routes for the OT generated prefix. 

**Describe the solution you'd like** A clear and concise description of what you want to happen.
A solution would be to have OT directly manage this, installing routes in Linux as they're created. This is similar to how OMRs are handled [here](https://github.com/openthread/openthread/blob/c28bb194d41d377d1f42ffe50a3916d1545d135c/src/posix/platform/netif.cpp#L708). 

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
We could manually watch Thread network data and add ip routes, however that feels potentially brittle. 

**Additional context** Add any other context or screenshots about the feature request here.

wgtdkp, 2023-02-17:
> We found a situation where the existing infrastructure prefix was retained, but our system wasn't setting up routes for the OT generated prefix.

If there is a working infra link prefix, OT won't use it's own generated prefix. What's the problem with this?

Can you provide more details of how this is wrong?, For example:
1. is the `existing infrastructure prefix` deprecated or invalidated?
2. Any OTBR log?

jwhui, 2023-02-17:
> If there is a working infra link prefix, OT won't use it's own generated prefix.

Note this addition in https://github.com/openthread/openthread/pull/8315/:

https://github.com/openthread/openthread/blob/2060039bbfec52ac00f22a7cc3213c2ab1df138e/src/core/border_router/routing_manager.cpp#L2202-L2210

I believe the above will cause OT to start publishing its own prefix immediately, before even having a chance to evaluate prefixes advertised by other routers on the infrastructure link.

Depending on the value of the prefix(es) advertised by other routers, OT may choose not to unpublish its own:

https://github.com/openthread/openthread/blob/2060039bbfec52ac00f22a7cc3213c2ab1df138e/src/core/border_router/routing_manager.cpp#L2269-L2274

Thoughts?

abtink, 2023-02-17:
> I believe the above will cause OT to start publishing its own prefix immediately, before even having a chance to evaluate prefixes advertised by other routers on the infrastructure link.

I need to read the code again to recall all details but related to calling `Get<RoutingManager>().EvaluatePublishingPrefix(mLocalPrefix);` from `Start()`, I don't think that this call on its own will cause the local prefix to be published. This should be tracked by `mState` in the `OnLinkPrefixManager` (e.g.,`PublishAndAdvertise()` or `Deprecate()` will change `mState`).

The `EvaluatePublishingPrefix()` will end up calling the `OnLinkPrefixManager::ShouldPublish()` which should check `mState` (when prefix the local on link).

The `PublishAndAdvertise()` call happens in `Evaluate()` which checks `VerifyOrExit(!Get<RoutingManager>().mRsSender.IsInProgress());` at the top (waiting for initial RS tx to be done).

jwhui, 2023-02-17:
@gabekassel provided this [otbr.log](https://github.com/openthread/openthread/files/10770670/otbr.log).

It appears that OTBR discovers the infrastructure prefix, but then does not receive any response to its Neighbor Solicitations. As a result, OTBR will invalidate information for that router and start publishing its own local prefix.

```
Jan 31 20:39:32.373 eero local5.info otbr-agent[2745]: 00:01:43.543 [I] BorderRouter--: Sent Neighbor Solicitation to fe80:0:0:0:6697:14ff:fe23:2bf2 - attempt:1/5
Jan 31 20:39:33.375 eero local5.info otbr-agent[2745]: 00:01:44.543 [I] BorderRouter--: Sent Neighbor Solicitation to fe80:0:0:0:6697:14ff:fe23:2bf2 - attempt:2/5
Jan 31 20:39:34.373 eero local5.info otbr-agent[2745]: 00:01:45.543 [I] BorderRouter--: Sent Neighbor Solicitation to fe80:0:0:0:6697:14ff:fe23:2bf2 - attempt:3/5
Jan 31 20:39:35.374 eero local5.info otbr-agent[2745]: 00:01:46.543 [I] BorderRouter--: Sent Neighbor Solicitation to fe80:0:0:0:6697:14ff:fe23:2bf2 - attempt:4/5
Jan 31 20:39:36.375 eero local5.info otbr-agent[2745]: 00:01:47.543 [I] BorderRouter--: Sent Neighbor Solicitation to fe80:0:0:0:6697:14ff:fe23:2bf2 - attempt:5/5
Jan 31 20:39:38.374 eero local5.info otbr-agent[2745]: 00:01:49.543 [I] BorderRouter--: No response to all Neighbor Solicitations attempts from router fe80:0:0:0:6697:14ff:fe23:2bf2
Jan 31 20:39:38.375 eero local5.debug otbr-agent[2745]: 00:01:49.543 [D] BorderRouter--: Prefix stale timer scheduled in 1727556 ms
Jan 31 20:39:38.375 eero local5.info otbr-agent[2745]: 00:01:49.543 [I] BorderRouter--: Start evaluating routing policy, scheduled in 3094 milliseconds
Jan 31 20:39:41.470 eero local5.info otbr-agent[2745]: 00:01:52.637 [I] BorderRouter--: Evaluating routing policy
Jan 31 20:39:41.470 eero local5.info otbr-agent[2745]: 00:01:52.637 [I] BorderRouter--: Publishing local on-link prefix fd31:de77:89a2:2331::/64 in netdata
Jan 31 20:39:41.471 eero local5.info otbr-agent[2745]: 00:01:52.637 [I] BorderRouter--: Start advertising local on-link prefix fd31:de77:89a2:2331::/64
```

@abtink , you ran into this issue recently as well, thoughts?

abtink, 2023-02-17:
My guess is that this is to do with platform.

We need to see if NA response is received by device and if so check if it being passed to OT stack.
I recall there was some recent PR in posix platform which removed some code which was filtering NA?

jwhui, 2023-02-17:
@gabekassel , just to confirm - does your build include:
- https://github.com/openthread/openthread/pull/8521

gabekassel, 2023-02-17:
@jwhui we observed this using [this](https://github.com/openthread/ot-br-posix/commit/079bbce34ad7868e2879a8e70550c6ae638d6bac) OTBR commit which coincidentally was merged on the same day as the NA change linked above. 

I couldn't pick out the dependabot PR that included it, but given your workflow I'm assuming this merged to OTBR later. 

I can make a new build with top of tree if helpful.

abtink, 2023-02-17:
@gabekassel I did investigate situation I was seeing (missing NA) which based on @jwhui comment is similar to what you see.

I noticed that we do get the NA response, but router seems to use a different link-local address for NA message (different than destination of NS). Here is result of `tcpdump`:
```
14:10:34.498436 IP6 fe80::37af:48f8:5284:9910 > fe80::200:5eff:fe00:265: ICMP6, neighbor solicitation, who has fe80::200:5eff:fe00:265, length 24
14:10:34.507393 IP6 fe80::56e0:320f:1e74:77f0 > fe80::37af:48f8:5284:9910: ICMP6, neighbor advertisement, tgt is fe80::200:5eff:fe00:265, length 24
```
We send NS to `fe80::200:5eff:fe00:265` (for same target) but get the NA from `fe80::56e0:320f:1e74:77f0` (for the correct target).

I submitted https://github.com/openthread/openthread/pull/8777 which changes the code to use target address to look up the router (ignore src address). This helped address the situation I was seeing. It may be that this is the same issue as what yours

My assumption here was that router may have multiple link-local addresses and used a different one for NS. 
Though checking `ip -v6 neighbour` it looks like the two addresses are associated with devices using different MACs.
```
ip -6 neighbour
fe80::56e0:320f:1e74:77f0 dev wlp2s0 lladdr [REDACATED]:77:f0 router STALE 
fe80::200:5eff:fe00:265 dev wlp2s0 lladdr [REDACTED]:02:65 router REACHABLE 
```

gabekassel, 2023-02-20:
@abtink I tested with your change in our system and it appears to work! I can't fully remember how this failed before, but iirc it was as fundamental as not being able to ping from wifi client<>Thread ED. This now works. Could we merge today so I can get this into our next release and generally move up the tree?

cc: @jwhui

abtink, 2023-02-20:
Good to hear. Thanks @gabekassel.

jwhui, 2023-02-21:
@gabekassel , thanks for testing and reporting back!

jwhui, 2023-03-02:
Fixed by #8777.

closed 2023-03-02 by jwhui