openthread/openthread #8782: Error-handling disparity on otPlatFlash implementation across platforms

AlanLCollins, 2023-02-21:
all otPlatFlash functions (otPlatFlashInit, otPlatFlashWrite, otPlatFlashRead, otPlatFlashErase,..) do not return error which leads platform owners to handle errors differently. 

The settings driver is the main use case of the otPlatFlash functions: https://github.com/openthread/openthread/blob/99a615bec2eeccad9ff03d6c5581c1fd51becf68/src/core/common/settings.cpp#L187  
which has verify or exit calls when errors are received, but these are useless when OPENTHREAD_CONFIG_PLATFORM_FLASH_API_ENABLE is enabled.

In most cases, platforms assert the thread service when error happens. This triggers the service to re-start, but without proper handling of certain errors the device enters an endless loop of restarting the service. 

corner cases are most evident in Linux platforms which use system calls like "pread" and "pwrite" - e.g. the simulation Flash.c : https://github.com/openthread/openthread/blob/564982c818666787096e5ad7a81a2d2f036aba36/examples/platforms/simulation/flash.c#L129 
An attempt to read an empty file, would return size != psize
```
rval = pread(sFlashFd, aData, aSize, (off_t)(address + aOffset));
    VerifyOrDie((uint32_t)rval == aSize, OT_EXIT_ERROR_ERRNO);
``` 
which sends the device into an endless re-starting loop. 

This specific corner case can be fixed by erasing the file before doing the assert, so the next re-start the code will recover and find all FFs when reading the file. 

I wanted to give visibility to the community to analyze and decide if it makes sense to modify the otPlatFlash functions to return error values and have a more robust ot solution.

wgtdkp, 2023-02-21:
I raised the same concerns before but didn't get an actual use case to support why we should return the error code rather than `assert` or `VerifyOrDie`. But I come to realize more limitations of not returning the error code:
1. in on-chip or RTOS systems where Thread, Wi-Fi and BLE stacks run in a single thread/process, die in OT stack is going to die the whole system including Wi-Fi and BLE. It's worse if this device is a Wi-Fi + Thread router as it's going to break connectivity of all in-home devices.
2. It stops users/vendors/developers to use OT as a library - An OT API can kill the whole process indeterminately whenever a `VerifyOrDie` is triggered and there is no way the upper layer can catch the root failure and take proper action. Current `otPlatFlash` APIs are forcing the developers to die or ignore the error, but neither is a good handling of a library error. Concrete examples are:
    - It stops integrating OT into OS kernel
    - It stops integrating OT into Android system server
    - It stops integrating OT into an Android App with JNI - `VerifyOrDie` in a JNI library is going to kill the App directly

jwhui, 2023-02-21:
Any platform API that returns errors must come with well-documented behaviors of what OpenThread will do when it encounters such an error. I don't think we should have a blanket rule for having all platform APIs return an error. If there is nothing OpenThread can do other than to die, then there isn't much benefit to returning the error.

wgtdkp, 2023-02-21:
> If there is nothing OpenThread can do other than to die, then there isn't much benefit to returning the error.

For errors which OT doesn't know how to handle, OT should propagate the error to upper layer to let the application  decides whether to die, disable Thread or continue with an alert/logging.

Specific to the otPlatFlash APIs, the Thread stack can actually continue working even the API fails - it simply can't survive a reboot.

jwhui, 2023-02-21:
> For errors which OT doesn't know how to handle, OT should propagate the error to upper layer to let the application decides whether to die, disable Thread or continue with an alert/logging.

This forces the client / application layer to understand the impact of various platform errors, which also requires them to understand Thread and OpenThread as its implementation.

> Specific to the otPlatFlash APIs, the Thread stack can actually continue working even the API fails - it simply can't survive a reboot.

This behavior would not conform to the Thread Specification requirements and thus would not be certifiable.

abtink, 2023-02-21:
> Any platform API that returns errors must come with well-documented behaviors of what OpenThread will do when it encounters such an error. I don't think we should have a blanket rule for having all [...]

I fully agree with @jwhui  on this.  Each platform can choose how it may make sense for it to handle errors (these are really platform internal errors).
- `VerifyOrDie()` (or `DieNow()`) checks in OT are all in `posix` platform code (mechanism added to handle posix issues).
- It OT core we don't really have this notion of "Die". We may `OT_ASSERT()` but this is for things that are clearly unexpected (major failures).

> For errors which OT doesn't know how to handle, OT should propagate the error to upper layer to let the application decides whether to die, disable Thread or continue with an alert/logging.

I think such mechanism can be implemented in platform layer code itself (have callback from platform to report to app layer that something in platform layer failed, we don't need this to go through OT stack and/or its APIs.

wgtdkp, 2023-02-22:
> This forces the client / application layer to understand the impact of various platform errors, which also requires them to understand Thread and OpenThread as its implementation.

The platform errors should be mapped to standard OT errors and the user needs to understand only the OT error. The user of OT public APIs don't need to know whether an API failed because of platform error or OT stack error, they just need to know whether the OT public API failed and with what error category which is represented by `otError`. The user don't even need to know that the OT stack/public APIs are implemented with a platform API - this is the common case that OT is usually integrated to 3P SDK (e.g. nrfconnect SDK) for IoT developers and they don't need to be aware of the platform API at all.

Take `otDatasetSetActive()` as example, we should clarify in the API documentation that this is going to save the dataset to persistent storage so that the dataset can survive a reboot. We need to clarify this because it really makes a difference whether the API persists the data. And I believe it makes sense that writing to whatever persistent storage could fail with an IO error. I don't think a developer is going to expect that `otDatasetSetActive()`  is going to crash their system/app rather than returning an IO error.

> VerifyOrDie() (or DieNow()) checks in OT are all in posix platform code (mechanism added to handle posix issues)

Yes, that's because the ot-posix usually runs in a standalone process as a daemon where killing itself won't affect other components. But the VerifyOrDie there stops us to put OTBR into Android system server which includes hundreds of critical system components and crashing the system server should be avoid if possible.

> I think such mechanism can be implemented in platform layer code itself (have callback from platform to report to app layer that something in platform layer failed, we don't need this to go through OT stack and/or its APIs.

It sounds weird for users of `otDatasetSetActive()` to handle the failure of this API in a separate callback rather than via its return value. The callback mechanism also lose the context of error, for example, if the user wants to tolerate any error of `otDatasetSetPending()` but disable Thread when `otDatasetSetActive()` fails, how do we pass the context to the callback and further to the user?

jwhui, 2023-02-22:
> The user don't even need to know that the OT stack/public APIs are implemented with a platform API - this is the common case that OT is usually integrated to 3P SDK (e.g. nrfconnect SDK) for IoT developers and they don't need to be aware of the platform API at all.

Platform APIs are not always called as a direct result of User API calls. For example, flash reads/writes may be called in response to receiving a network message or timer firing. In these cases, how do we expect the application developer to know what to do?

> I don't think a developer is going to expect that `otDatasetSetActive()` is going to crash their system/app rather than returning an IO error.

No, it is not expected. But that is the point of not allowing the platform to return error, it means that the platform needs to succeed. OpenThread does not have any way to handle the failure properly.

> It sounds weird for users of `otDatasetSetActive()` to handle the failure of this API in a separate callback rather than via its return value.

As mentioned above, platform APIs are not always called as part of a user API call.

I think it would help if someone could provide a concrete proposal, which documents all the scenarios where the platform API may be called, when the platform API may return error, and how OpenThread will handle and/or propagate those errors to the user in every case.

wgtdkp, 2023-02-23:
Sure, let me try to draft a document!

jwhui, 2024-09-27:
Closing stale issue.

closed 2024-09-27 by jwhui