openthread/openthread #8792: can not access to IPv6 Internet

lancersky2017, 2023-02-22:
**Describe the bug** A clear and concise description of what the bug is.
Hi experts,
I formed a Thread network according to the documentation https://openthread.google.cn/codelabs/openthread-border-router-nat64#0
```
~# ot-ctl state
leader
Done
~# ot-ctl ipaddr
fdde:ad00:beef:0:0:ff:fe00:fc11
fdde:ad00:beef:0:0:ff:fe00:fc10
fdde:ad00:beef:0:0:ff:fe00:fc38
fdc9:58d8:8547:1:686:10fc:8fad:c1ca
fdde:ad00:beef:0:0:ff:fe00:fc00
fdde:ad00:beef:0:0:ff:fe00:3c00
fdde:ad00:beef:0:7fef:ca83:4a41:a6bd
fe80:0:0:0:24b5:6dbc:892:4bf3
Done
~# ot-ctl netdata show
Prefixes:
fdc9:58d8:8547:1::/64 paos low 3c00
Routes:
fdc9:58d8:8547:2:0:0::/96 sn low 3c00
fdf5:9feb:1392:231::/64 s med 3c00
Services:
44970 01 2b000500000e10 s 3c00
44970 5d fddead00beef00007fefca834a41a6bdd121 s 3c00
Done
~# ot-ctl nat64 state
PrefixManager: Active
Translator: Active
Done
```
And I successfully tested NAT64/DNS64 with an NRF52840 USB dongle.
```
> version
OPENTHREAD/thread-reference-20200818-2312-g242c7cc7c; NRF52840; Feb 16 2023 10:56:03
Done
> ping 8.8.8.8
Pinging synthesized IPv6 address: fdc9:58d8:8547:2:0:0:808:808
16 bytes from fdc9:58d8:8547:2:0:0:808:808: icmp_seq=33 hlim=115 time=64ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 64/64.0/64 ms.
Done
> dns resolve4 gitlab.com 8.8.8.8
Synthesized IPv6 DNS server address: fdc9:58d8:8547:2:0:0:808:808
DNS response for gitlab.com. - fdc9:58d8:8547:2:0:0:ac41:fb4e TTL:300 
Done
> ping fdc9:58d8:8547:2:0:0:ac41:fb4e
16 bytes from fdc9:58d8:8547:2:0:0:ac41:fb4e: icmp_seq=34 hlim=54 time=201ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 201/201.0/201 ms.
Done
> ping 192.168.18.203
Pinging synthesized IPv6 address: fdc9:58d8:8547:2:0:0:c0a8:12cb
16 bytes from fdc9:58d8:8547:2:0:0:c0a8:12cb: icmp_seq=36 hlim=63 time=15ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 15/15.0/15 ms.
Done
```
But the MTD node can not access to IPv6 Internet. And I grab wpan0 with tcpdump without any packets.
Even if I turn nat64 off it doesn't work.
```
> dns resolve gitlab.com 8.8.8.8
Synthesized IPv6 DNS server address: fdc9:58d8:8547:2:0:0:808:808
DNS response for gitlab.com. - 2606:4700:90:0:f22e:fbec:5bed:a9b9 TTL:300 
Done
> ping 2606:4700:90:0:f22e:fbec:5bed:a9b9
1 packets transmitted, 0 packets received. Packet loss = 100.0%.
Done
```
I checked that OTBR is able to access the IPv6 network.
```
~# ping 2606:4700:90:0:f22e:fbec:5bed:a9b9
PING 2606:4700:90:0:f22e:fbec:5bed:a9b9 (2606:4700:90:0:f22e:fbec:5bed:a9b9): 56 data bytes
64 bytes from 2606:4700:90:0:f22e:fbec:5bed:a9b9: seq=0 ttl=55 time=241.518 ms
64 bytes from 2606:4700:90:0:f22e:fbec:5bed:a9b9: seq=1 ttl=55 time=241.962 ms
64 bytes from 2606:4700:90:0:f22e:fbec:5bed:a9b9: seq=2 ttl=55 time=241.422 ms
^C
--- 2606:4700:90:0:f22e:fbec:5bed:a9b9 ping statistics ---
5 packets transmitted, 3 packets received, 40% packet loss
round-trip min/avg/max = 241.422/241.634/241.962 ms

~# ping -6 gitlab.com
PING gitlab.com (2606:4700:90:0:f22e:fbec:5bed:a9b9): 56 data bytes
64 bytes from 2606:4700:90:0:f22e:fbec:5bed:a9b9: seq=0 ttl=55 time=189.532 ms
64 bytes from 2606:4700:90:0:f22e:fbec:5bed:a9b9: seq=1 ttl=55 time=189.199 ms
64 bytes from 2606:4700:90:0:f22e:fbec:5bed:a9b9: seq=2 ttl=55 time=189.630 ms
^C
--- gitlab.com ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 189.199/189.453/189.630 ms
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id OTBR base on **GSDK4.2.1**
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

wgtdkp, 2023-02-22:
Accessing of global IPv6 internet is not supported by OT/OTBR yet.

kianooshkarami, 2023-02-22:
> Accessing of global IPv6 internet is not supported by OT/OTBR yet.

Could you share where we could track when that functionality is planned for and how we could track it?