openthread/openthread #8854: Stale prefix contexts will not get cleared if leader goes into a crashloop /rebooted frequently

sarveshkumarv3, 2023-03-13:
Currently if there are stale prefix contexts in the network data (with only the 6LoWPAN context ID) and leader goes into a crashloop or rebooted frequently, the stale contexts would not get cleared

Specifically  Leader::HandleNetworkDataRestoredAfterReset method would invoke StartContextReuseTimer for stale prefix on every crash which would reset the mContextLastUsed timer value. 

Thinking about potential solutions, one of the approaches to handle this could be to persist how long the entry has remained stale and restore the timer to run only for the remaining time?

ndyck14, 2023-03-14:
This is related to discussion around the length of this timer: https://threadgroup.atlassian.net/browse/SPEC-1163

Leader weight is also relevant (intermittent powered devices should become less likely to to become leader).

jwhui, 2023-04-06:
Resolved by #8859.

closed 2023-04-06 by jwhui