openthread/openthread #8873: "otbr-agent" received signal SIGILL, Illegal instruction

lancersky, 2023-03-16:
**Describe the bug** A clear and concise description of what the bug is.
Hi, experts, In the following test scenarios, continuous ping tests (PC->Thread device) will cause otbr-agent to crash.
We used remote GDB to debug it and captured the following information.
```
Starting program: /xxxx/openwrt-21.02/openwrt-21.02.2/build_dir/target-mips_24kc_musl/ot-br-posix-2022-11-08-r2/ipkg-mips_24kc/ot-br-posix/usr/sbin/otbr-agent -d 7 -B br-lan -I wpan0 spinel+hdlc+uart:///dev/ttyCH343USB0?uart-baudrate=1000000\\&uart-flow-control
[New Thread 13719.13728]

Thread 1 "otbr-agent" received signal SIGILL, Illegal instruction.
0x004043e9 in otbr::MainloopManager::GetInstance () at /xxxx/openwrt-21.02/openwrt-21.02.2/build_dir/target-mips_24kc_musl/ot-br-posix-2022-11-08-r2/src/common/mainloop_manager.hpp:68
68              static MainloopManager sMainloopManager;
(gdb) 
```
[image]

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id **GSDK4.2.1**
2. IEEE 802.15.4 hardware platform **EFR32MG21F768**
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

lancersky, 2023-03-21:
backtrack,
```
Thread 1 "otbr-agent" received signal SIGILL, Illegal instruction.
0x004043e9 in otbr::MainloopManager::GetInstance () at /xxxx/openwrt-21.02/openwrt-21.02.2/build_dir/target-mips_24kc_musl/ot-br-posix-2022-11-08-r3/src/common/mainloop_manager.hpp:68
68              static MainloopManager sMainloopManager;
(gdb) bt
#0  0x004043e9 in otbr::MainloopManager::GetInstance () at /xxxx/openwrt-21.02/openwrt-21.02.2/build_dir/target-mips_24kc_musl/ot-br-posix-2022-11-08-r3/src/common/mainloop_manager.hpp:68
#1  0x00404179 in otbr::Application::Run (this=0x7fffa930) at /xxxx/openwrt-21.02/openwrt-21.02.2/build_dir/target-mips_24kc_musl/ot-br-posix-2022-11-08-r3/src/agent/application.cpp:162
#2  0x004048f7 in realmain (argc=<optimized out>, argv=<optimized out>) at /xxxx/openwrt-21.02/openwrt-21.02.2/build_dir/target-mips_24kc_musl/ot-br-posix-2022-11-08-r3/src/agent/main.cpp:350
#3  0x00403e55 in main (argc=<optimized out>, argv=<optimized out>) at /xxxx/openwrt-21.02/openwrt-21.02.2/build_dir/target-mips_24kc_musl/ot-br-posix-2022-11-08-r3/src/agent/main.cpp:389
```

superwhd, 2023-03-21:
I haven't met this before. Maybe you can check the assembly code to see what instruction caused this error.

lancersky, 2023-03-22:
The processor chip uses QCA9531.
The GCC compiles by default with `-mips16 -minterlink-mips16` on OpenWrt. If I turn this option off, it works fine for a long time.
I guess it may be related to the encryption library, so I checked the default configuration of mbedtls.
https://github.com/openwrt/openwrt/blob/23c86d44bcfa00255c32196458b0a47dfa6732d5/package/libs/mbedtls/Makefile#L10-L13

lancersky, 2023-03-23:
The issue has been solved. Thanks.

closed 2023-03-23 by lancersky