openthread/openthread #8888: Clarify required for "SFD timestamp" in radio.h

EskoDijk, 2023-03-21:
**Describe the bug** 
There is code in `radio.h` 

```
             /**
             * The timestamp when the frame was received in microseconds.
             *
             * The value SHALL be the time when the SFD was received when TIME_SYNC or CSL is enabled.
             * Otherwise, the time when the MAC frame was fully received is also acceptable.
             *
             */
            uint64_t mTimestamp;

```

It's not clear if mTimestamp shall be start of first symbol of SFD, or end of last symbol of SFD (i.e. after it has been received). Platforms could use different interpretations which leads to a 32 us time difference in CSL timing between platforms, potentially.

**To Reproduce** 

Latest commit.

**Expected behavior**
Comment should indicate for `mTimestamp` if the time is 
1. the start of the first symbol of SFD, or
2. the end of the last symbol of SFD
3. time exactly in between first and last symbol (Maybe :) )

jwhui, 2023-03-22:
I always assumed it would be after receiving the last symbol of the SFD - until the last symbol is received, the SFD is not yet detected. However, I can understand that implementations may calculate a different time based on when the SFD was detected.

@EskoDijk , would you be willing to propose an update?

EskoDijk, 2023-03-22:
@jwhui Yes, added the PR now.

PR #8894 (2023-03-29): [radio] clarify `mTimestamp` meaning in comments
close #8888
closed 2023-03-29 by jwhui