openthread/openthread #8902: HandleRcpTimeout() at radio_spinel_impl.hpp:2297: RadioSpinelNoResponse

dsyx, 2023-03-24:
After the Border Router runs for a period of time (this time will be shortened or extended according to the amount of network traffic), the RCP timeout problem will occur.

I use [GSDK v4.2.2](https://github.com/SiliconLabs/gecko_sdk/commit/11dd105731f1ac4cac424691e09b159a03e76d32) to develop Host and RCP applications. The hardware platform is Raspberry Pi 3B+ and EFR32 (MG12 and MG13)

otbr-agent logs:

```
Mar 24 11:22:33 otbr otbr-agent[423694]: 00:00:08.682 [I] BorderRouter--: Start evaluating routing policy, scheduled in 2233 milliseconds
Mar 24 11:22:33 otbr otbr-agent[423694]: 00:00:08.759 [I] BorderRouter--: Received Router Advertisement from fe80:0:0:0:f22f:74ff:feb6:3e10 on infra netif 2
Mar 24 11:22:33 otbr otbr-agent[423694]: 00:00:08.759 [I] BorderRouter--: Processing PIO (240e:3b6:4a8:abd0::/64, 600 seconds)
Mar 24 11:22:33 otbr otbr-agent[423694]: 00:00:08.759 [I] NetDataPublshr: Publishing ExternalRoute 240e:3b6:4a8:abd0::/64
Mar 24 11:22:33 otbr otbr-agent[423694]: 00:00:08.759 [I] NetDataPublshr: ExternalRoute 240e:3b6:4a8:abd0::/64 (state:Added in netdata - total:1, preferred:0, desired:10
Mar 24 11:22:33 otbr otbr-agent[423694]: 00:00:08.763 [I] BorderRouter--: Start evaluating routing policy, scheduled in 2943 milliseconds
Mar 24 11:22:34 otbr otbr-agent[423694]: 00:00:09.682 [I] BorderRouter--: Start evaluating routing policy, scheduled in 1233 milliseconds
Mar 24 11:22:35 otbr otbr-agent[423694]: 00:00:10.271 [W] Platform------: radio tx timeout
Mar 24 11:22:35 otbr otbr-agent[423694]: 00:00:10.272 [C] Platform------: HandleRcpTimeout() at radio_spinel_impl.hpp:2297: RadioSpinelNoResponse
Mar 24 11:22:35 otbr systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Mar 24 11:22:35 otbr systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Mar 24 11:22:40 otbr systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 187.
Mar 24 11:22:40 otbr systemd[1]: Stopped OpenThread Border Router Agent.
Mar 24 11:22:40 otbr systemd[1]: Starting OpenThread Border Router Agent...
Mar 24 11:22:44 otbr systemd[1]: Started OpenThread Border Router Agent.
Mar 24 11:22:44 otbr otbr-agent[423833]: [NOTE]-AGENT---: Backbone interface: eth0
Mar 24 11:22:44 otbr otbr-agent[423833]: [NOTE]-AGENT---: Running 0.3.0-11dd10573-dirty
Mar 24 11:22:44 otbr otbr-agent[423833]: [NOTE]-AGENT---: Thread version: 1.3.0
Mar 24 11:22:44 otbr otbr-agent[423833]: [NOTE]-AGENT---: Thread interface: wpan0
Mar 24 11:22:44 otbr otbr-agent[423833]: [NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyUSB0?uart-baudrate=460800&uart-flow-control
Mar 24 11:22:44 otbr otbr-agent[423833]: [NOTE]-AGENT---: Radio URL: trel://eth0
Mar 24 11:22:44 otbr otbr-agent[423833]: [NOTE]-ILS-----: Infra link selected: eth0
Mar 24 11:22:44 otbr otbr-agent[423833]: [INFO]-NCP-----: OpenThread log level changed to 4
Mar 24 11:22:46 otbr otbr-agent[423833]: 52d.16:34:53.084 [W] Platform------: Wait for response timeout
Mar 24 11:22:46 otbr otbr-agent[423833]: 52d.16:34:53.084 [C] Platform------: HandleRcpTimeout() at radio_spinel_impl.hpp:2297: RadioSpinelNoResponse
Mar 24 11:22:46 otbr systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Mar 24 11:22:46 otbr systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Mar 24 11:22:51 otbr systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 188.
Mar 24 11:22:52 otbr systemd[1]: Stopped OpenThread Border Router Agent.
Mar 24 11:22:52 otbr systemd[1]: Starting OpenThread Border Router Agent...
Mar 24 11:22:56 otbr systemd[1]: Started OpenThread Border Router Agent.
Mar 24 11:22:56 otbr otbr-agent[423863]: [NOTE]-AGENT---: Backbone interface: eth0
Mar 24 11:22:56 otbr otbr-agent[423863]: [NOTE]-AGENT---: Running 0.3.0-11dd10573-dirty
Mar 24 11:22:56 otbr otbr-agent[423863]: [NOTE]-AGENT---: Thread version: 1.3.0
Mar 24 11:22:56 otbr otbr-agent[423863]: [NOTE]-AGENT---: Thread interface: wpan0
Mar 24 11:22:56 otbr otbr-agent[423863]: [NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyUSB0?uart-baudrate=460800&uart-flow-control
Mar 24 11:22:56 otbr otbr-agent[423863]: [NOTE]-AGENT---: Radio URL: trel://eth0
Mar 24 11:22:56 otbr otbr-agent[423863]: [NOTE]-ILS-----: Infra link selected: eth0
Mar 24 11:22:56 otbr otbr-agent[423863]: [INFO]-NCP-----: OpenThread log level changed to 4
Mar 24 11:22:58 otbr otbr-agent[423863]: 52d.16:35:04.643 [W] Platform------: Wait for response timeout
Mar 24 11:22:58 otbr otbr-agent[423863]: 52d.16:35:04.643 [C] Platform------: HandleRcpTimeout() at radio_spinel_impl.hpp:2297: RadioSpinelNoResponse
Mar 24 11:22:58 otbr systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Mar 24 11:22:58 otbr systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Mar 24 11:23:03 otbr systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 189.
Mar 24 11:23:03 otbr systemd[1]: Stopped OpenThread Border Router Agent.
```

suveshpratapa, 2023-03-24:
@dsyx Can you verify that your RCP is connected at the right `/dev/tty` device that is also configured in your OTBR agent properties at `/etc/default/otbr-agent`?

dsyx, 2023-03-26:
@suveshpratapa 

RCP is always physically connected to HOST. `/etc/default/otbr-agent` is also configured. In the beginning, `otbr-agent` works fine, but after some time it will show `HandleRcpTimeout() at radio_spinel_impl.hpp:2297: RadioSpinelNoResponse`. After several `RadioSpinelNoResponse`, otbr-agent` will return to normal.

jwhui, 2024-03-05:
Closing stale issue.

closed 2024-03-05 by jwhui