openthread/openthread #8905: Next Version: Require any REED to be able to be commissioned or downgraded to FED-only.

ekobres, 2023-03-25:
**Is your feature request related to a problem? Please describe.** 
Retrofit devices connected to power switches should often *not* be eligible as routers.
When power is interrupted to a retrofit light bulb in a lamp or in a fixture connected to a switch, the resulting disruption to the mesh is often significant. This is especially true for fixtures that contain multiple bulbs, like chandeliers. This is also true for certain types of fixtures like lamps and other things that humans are in the habit of turning off manually.

**Describe the solution you'd like** 
An ideal solution would be to require REEDs to be able to be downgraded to FEDs in cases where the installer knows it is a bad idea for the device to become a router. This could either be an option during commissioning, or a special command issued by an OTBR that would cause a persistent, non-permanent downgrade of the device. This process should be reversible, and reconfiguration in either direction (REED->FED FED->REED) should require proof of control of the device by either requiring re-commissioning, or manipulating the power to the device. It would not be ideal to allow this to happen seamlessly during otherwise normal operation or without some security controls.

**Describe alternatives you've considered** 
Alternatives would be for manufacturers to provide alternative firmware for devices for REED and FED operation, but this is a big as of manufacturers, and would be arduous for installers.

**Additional context** 
This concept is well-documented in the Zigbee universe - and in many cases retrofit bulbs are specifically shipped as child-only devices for precisely the reasons mentioned above.

abtink, 2023-03-25:
Related to this OpenThread already provides an API [`otThreadSetRouterEligible()`](https://github.com/openthread/openthread/blob/e8f3ec0fd3356d0c52b1a6f444e63b5c86abbbdb/include/openthread/thread_ftd.h#L154-L178).

ekobres, 2023-03-25:
Yes, and that would be the expected implementation. This request is to require REEDs to persistently honor a request to set their state to not-eligible through a new mechanism during commissioning or at a later time such that the person setting up the device can control whether or not it is router-eligible. I am asking to *require* device manufacturers allow router functionality to be disabled.

"Shall" rather than "may".