openthread/openthread #8970: Test 5 6 6 Network Data Expiration Failed

lancersky2017, 2023-04-18:
**Describe the bug** A clear and concise description of what the bug is.
Hi, Can you check for me to see where the problem is? 
[image]
[Test 5 6 6 Network Data Expiration Failed.zip](https://github.com/openthread/openthread/files/11257250/Test.5.6.6.Network.Data.Expiration.Failed.zip)
[CMakeCache.txt](https://github.com/openthread/openthread/files/11257266/CMakeCache.txt)

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id **GSDK4.2.2**
2. IEEE 802.15.4 hardware platform **EFR32MG21F768**
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

lancersky2017, 2023-04-19:
Hi @jwhui, I also tested this case on Raspberry Pi4 and it also failed on Test No.18. 
```
pi@raspberrypi:~ $ sudo ot-ctl version
OPENTHREAD/c6179c24e; POSIX; Jan 19 2023 13:25:54
Done
pi@raspberrypi:~ $ sudo ot-ctl rcp version
OPENTHREAD/d7de666; Zephyr; Jan 19 2023 18:43:03
Done
```

jwhui, 2023-04-19:
The Border Router is explicitly removing `2003::` prefix in Packet 186. So I would say OpenThread is behaving as expected in this case. I would reach out to the Test Harness team to see if they have any ideas.

[image]