openthread/openthread #8973: DNS-SD browse from Thread device does not return the same IPv6 address as avahi-browse

etiennedm, 2023-04-18:
**Describe the bug** 
When browsing for a service from a Thread device, the resolved IPv6 address does not match the one returned by avahi-browse.

**To Reproduce**

1. Git commit id: 3e138d3f5
2. IEEE 802.15.4 hardware platform: Nordic nrf52840
3. Build steps: Zephyr based firmware
4. Network topology: laptop publishing a _lwm2m._udp service, Raspberry Pi acting as the OTBR, nrf52840 custom board acting as the client Thread device

**Expected behavior**
When publishing the service with avahi-publish from my laptop:
```bash
avahi-publish -s "LwM2M Server" _lwm2m._udp 568
```
Then browsing from the Thread device:
```bash
uart:~$ ot dns browse _lwm2m._udp.default.service.arpa
DNS browse response for _lwm2m._udp.default.service.arpa.
LwM2M Server                                      
    Port:5685, Priority:0, Weight:0, TTL:10        
    Host:Latitude-E7470-341.default.service.arpa.  
    HostAddress:fd11:1111:1122:2222:363:13df:6445:f299 TTL:10
    TXT:[] TTL:10                                 
Done 
```
I am not getting the same address as the one I get from avahi-browse:
```bash
➜  ~ avahi-browse -rt _lwm2m._udp
+ wlp1s0 IPv6 LwM2M Server                                  _lwm2m._udp          local
+ wlp1s0 IPv4 LwM2M Server                                  _lwm2m._udp          local
+     lo IPv4 LwM2M Server                                  _lwm2m._udp          local
= wlp1s0 IPv6 LwM2M Server                                  _lwm2m._udp          local
   hostname = [Latitude-E7470-341.local]
   address = [fd11:1111:1122:2222:be95:c327:f96c:1ba8]
   port = [5685]
   txt = []
= wlp1s0 IPv4 LwM2M Server                                  _lwm2m._udp          local
   hostname = [Latitude-E7470-341.local]
   address = [192.168.1.69]
   port = [5685]
   txt = []
=     lo IPv4 LwM2M Server                                  _lwm2m._udp          local
   hostname = [Latitude-E7470-341.local]
   address = [127.0.0.1]
   port = [5685]
   txt = []
```

This is problematic since it seems like `fd11:1111:1122:2222:be95:c327:f96c:1ba8` is the one I need to use to talk to the LwM2M server running on my laptop. If I use `fd11:1111:1122:2222:363:13df:6445:f299` to connect from the Thread device, the reply I get comes back from `fd11:1111:1122:2222:be95:c327:f96c:1ba8` and is therefore dropped (I'm guessing it's because it's not coming back from the IP the request was sent to).

**Additional context** 
For reference here is the IPv6 situation on the laptop running the LwM2M server:
```bash
➜  ~ ip -6 addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
3: wlp1s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fd11:1111:1122:2222:c064:d962:133b:b863/64 scope global temporary dynamic 
       valid_lft 1704sec preferred_lft 1704sec
    inet6 fd11:1111:1122:2222:363:13df:6445:f299/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 1704sec preferred_lft 1704sec
    inet6 fd11:1111:1122:2222:be95:c327:f96c:1ba8/64 scope global temporary dynamic 
       valid_lft 203sec preferred_lft 203sec
    inet6 fe80::a707:e0d5:ef2a:660a/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```

And the logs on the border router associated with browsing from the Thread device:
```bash
Apr 18 23:34:21 otbr otbr-agent[686]: 5d.05:52:48.531 [I] DnssdServer---: ANSWER: TRANSACTION=0x4b1a, QUESTION=[_lwm2m._udp.default.service.arpa. 1 12], RCODE=3
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-DPROXY--: Subscribe: _lwm2m._udp.default.service.arpa.
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: Subscribe service ._lwm2m._udp (total 1)
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: DNSServiceBrowse _lwm2m._udp
Apr 18 23:34:21 otbr otbr-agent[686]: 5d.05:52:48.808 [I] MeshForwarder-: Received IPv6 UDP msg, len:91, chksum:772b, ecn:no, from:fea6db13e75a3766, sec:no, prio:net, rss:-81.0
Apr 18 23:34:21 otbr otbr-agent[686]: 5d.05:52:48.808 [I] MeshForwarder-:     src:[fe80:0:0:0:fca6:db13:e75a:3766]:19788
Apr 18 23:34:21 otbr otbr-agent[686]: 5d.05:52:48.808 [I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
Apr 18 23:34:21 otbr otbr-agent[686]: 5d.05:52:48.808 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:fca6:db13:e75a:3766,0x0400)
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: DNSServiceBrowse reply: add LwM2M Server._lwm2m._udp. inf 2, flags=2, error=0
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: DNSServiceResolve LwM2M Server _lwm2m._udp. inf 2
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: DNSServiceResolve reply: LwM2M\032Server._lwm2m._udp.local. host Latitude-E7470-341.local.:13590, TXT=1B inf 2, flags=0
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: DNSServiceGetAddrInfo Latitude-E7470-341.local. inf 2
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: DNSServiceGetAddrInfo reply: flags=1073741827, host=Latitude-E7470-341.local., sa_family=10, error=0
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: DNSServiceGetAddrInfo reply: address=fd11:1111:1122:2222:363:13df:6445:f299, ttl=120
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: Service _lwm2m._udp is resolved successfully: add LwM2M\032Server host Latitude-E7470-341.local. addresses 1
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-DPROXY--: Service discovered: _lwm2m._udp, instance LwM2M\032Server hostname Latitude-E7470-341.local. addresses 1 port 5685 priority 0 weight 0
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-DPROXY--: Unsubscribe: _lwm2m._udp.default.service.arpa. 
Apr 18 23:34:21 otbr otbr-agent[686]: [INFO]-MDNS----: Unsubscribe service ._lwm2m._udp (left 0)
Apr 18 23:34:21 otbr otbr-agent[686]: 5d.05:52:48.937 [I] DnssdServer---: send DNS-SD reply: OK, RCODE=0
```

It looks like the OTBR does not forward all the information that is sent back by the laptop (running the LwM2M server). See wireshark screenshot below:
[image]

I would appreciate any help or suggestions! This might not be the ways to do things properly (ultimately what I want is not to hardcode IP addresses in the firmware and have Thread devices discover the IP of the LwM2M server).

wgtdkp, 2023-04-19:
Currently Discovery Proxy supports only a single address I believe.

> If I use fd11:1111:1122:2222:363:13df:6445:f299 to connect from the Thread device, the reply I get comes back from fd11:1111:1122:2222:be95:c327:f96c:1ba8 and is therefore dropped (I'm guessing it's because it's not coming back from the IP the request was sent to).

I assume this is UDP transport, right? It should be a bug of your server program if it's not sending the reply with the client address received in the request (likely it will be using "sendto()" or "sendmsg()" to send the reply, so need to make sure the destination address is set to the source address of the request).

jwhui, 2023-04-19:
@etiennedm , couple comments:

1. The service should only publish the IPv6 addresses that can be used to reach the service. It does not make sense to publish multiple IPv6 addresses, some of which do not work since the client will then need to potentially try a number of different IPv6 addresses.
2. If the application-layer protocol relies on address/port matching to correlate request and response messages, then it should also make sure to copy the Request's Source Address/Port into the Response's Destination Address/Port.

etiennedm, 2023-04-19:
I do realize that the OpenThread stack on the device as well as the BR are more than likely behaving as expected, so I very much appreciate your help!

The `fd11:1111:1122:2222::/64` prefix is advertised by the Raspberry Pi running the OTBR: I checked with Wireshark who is sending the RAs and they come from the Raspberry Pi link local address. It's also sending a route information with prefix `fd11:22::/64` which matches with the prefix advertised on the Thread network. I did follow the codelab you pointed to @wgtdkp although I built it with the following options: `WEB_GUI=1 NAT64=1 INFRA_IF_NAME=eth0 ./script/setup`.

I'm a little wary of sharing a pcap file with potentially private information (and I wouldn't know what to redact anyway!). I can maybe apply appropriate filters if you tell me what you're interested in.

@jwhui regarding your comments:

1. Both addresses are actually valid as far as I know. The packets from the Thread device are properly received and processed, but the reply is not sent from the same IP as the one the query was received on. And I am not aware of a way to force the mDNS responder to only pick one address to be advertised. This would actually go against section 6.2 of the RFC 6762: [https://www.rfc-editor.org/rfc/rfc6762#section-6.2](https://www.rfc-editor.org/rfc/rfc6762#section-6.2).
2. This is something I need to look into. It seems it's not so obvious that the server knows from what IP the UDP packets are coming from. My understanding is that this is somehow more of an OS decision when you bind your server socket to all interfaces? I'm going to try to have the server bind to a specific IP (the one reported by the service discovery from the perspective of the Thread device).

Finally I think I might be reasoning the wrong way: I am not at all used to administrating servers, and it looks like the "server" way of doing things is to have a static IP. Using my laptop, which is really configured in various ways as a client is probably not such a great idea.

jwhui, 2023-04-19:
@etiennedm , when using sockets API, you can use sendto/recvfrom to access the sockaddr struct. The server not using the appropriate IPv6 Source Address in its responses seems like a bug.

etiennedm, 2023-04-30:
Hi, so I've finally had some time to finish testing on this issue.

After reading some more about server/client applications and networking APIs, I completely agree with you @jwhui.

Digging into the configuration of the service I'm using, I found a way to bind to a specific IP address instead of any (https://thingsboard.io/docs/user-guide/install/config/ if anyone stumbles upon this issue later on). This takes care of the problem. The other solution, since it is running in a Docker container, was to map the port to a specific host IP, as documented here https://docs.docker.com/config/containers/container-networking/.

Thanks for your help! I'm closing this issue.

closed 2023-04-30 by etiennedm