openthread/openthread #8985: CCM_8 being not recommanded anymore, would be nice to open the cabalibities to use full size CCM (16)

jvermillard, 2023-04-21:
Following https://www.iana.org/assignments/tls-parameters/tls-parameters.xhtml, ccm_8 is not recommended anymore due to "potential" weakness in ccm_8 (not ccm, which uses 16bits tags)

Here is a good overview of the current concerns of the community:
https://mailarchive.ietf.org/arch/browse/core/?gbt=1&index=WnRInwF-j0uZmLggFh37ySljnwE

I’m not aware of practical PoC, but that would not be surprising to see something appearing a couple of years from now.

Using TLS_PSK_WITH_AES_128_CCM or, even better, TLS_ECDHE_PSK_WITH_AES_128_CCM would make the design safe for longer.
But as an Openthread user, you are stuck with CCM_8 even if MbedTLS support many more ciphersuites.

jwhui, 2023-04-21:
@jvermillard , thanks for raising this issue.

I assume you are referring to [OpenThread's CoAP Security example](https://github.com/openthread/openthread/blob/main/src/cli/README_COAPS.md#configure-dtls-ciphersuite). As noted in the README, this is simply an example of how to implement coaps using OpenThread's CoAP implementation. Note that this ciphersuite is not used anywhere within the Thread Protocol itself.

If you are willing, feel free to submit a PR with a proposed change. Thanks!

jvermillard, 2023-04-24:
Thanks @jwhui for the clarification, I 'll take a deeper look

closed 2023-05-09 by jwhui