openthread/openthread #9063: [mle] test harness issue when joining after #8836

jwhui, 2023-05-17:
We found joining issue with Kirale leader with this PR.  During analysis we suspected aNeighbor.SetLastHeard(TimerMilli::GetNow()) which is now calling from InitNeighbor() from Handle parent request and response. 

Do you think aNeighbor.SetLastHeard(TimerMilli::GetNow()) required to update during Handle parent request and response? 
Before this PR, it was not updated in parent request and response. When we are not calling in parent request and response, we able to join the device. 

can you elaborate please?

_Originally posted by @Ashish-Vara in https://github.com/openthread/openthread/issues/8836#issuecomment-1551166072_

jwhui, 2023-05-17:
@Ashish-Vara, are you able to post the Test Harness output (e.g. test report PDF, pcap, logs, etc.)?

abtink, 2023-05-17:
I responded on the PR -> https://github.com/openthread/openthread/pull/8836#issuecomment-1551891608. Copying it here:

> Hi @Ashish-Vara ,
> 
> > Do you think `aNeighbor.SetLastHeard(TimerMilli::GetNow())` required to update during Handle parent request and response?
> 
> Yes. We should set `LastHeard` time in these cases. Note that `MleRouter::HandleParentRequest()` was setting the "last heard" on `child` before this PR from this `if()` block:
> 
> https://github.com/openthread/openthread/blob/dae3ff2c50210cf2e098caee65c1b30f94c6fb40/src/core/thread/mle_router.cpp#L1437-L1441
> 
> (now `InitNeighbor` is also setting the same from earlier block of code witin same method)
> 
> > When we are not calling in parent request and response, we able to join the device.
> 
> This is weird. Suggest investigating this further to figure out the root cause and what is happening (collect the logs on devices to see what is happening).

Ashish-Vara, 2023-05-19:
[L_5_3_2_issue.zip](https://github.com/openthread/openthread/files/11514249/L_5_3_2_issue.zip)

Please check the logs. It contains few debug logs also. Kirale is leader and DUT acting as a router. We see after sometimes link quality becomes zero.

jwhui, 2023-05-19:
@Ashish-Vara , I reviewed the packet trace and it appears that the Leader is not establishing a link with the Router.

- In pkt13, you can see that the Router has a valid link to the Leader (in/out link quality are both 3).
- In pkt23, you can see that the Leader does not have a valid link to the Router (in/out link quality are both 0).

At no time does the Leader attempt to send an MLE Link Request to establish a link with the neighboring router. The Leader should be doing this when it receives an MLE Advertisement from a router that it does not have an established link with.

I believe this is an issue with the Leader reference device. I suggest raising this issue with the Thread Group directly.

Ashish-Vara, 2023-05-23:
Thanks @jwhui, 
I will raise with thread group.