openthread/openthread #9122: GH action Android NDK build failing

abtink, 2023-06-01:
GH action Android NDK seems to fail:
https://github.com/openthread/openthread/actions/runs/5141290740/jobs/9266852284
```
2023-06-01T16:48:02.1826387Z ##[group]Run actions/checkout@ac593985615ec2ede58e132d2e21d2b1cbd6127c
2023-06-01T16:48:02.1826663Z with:
2023-06-01T16:48:02.1826839Z   submodules: true
2023-06-01T16:48:02.1827066Z   repository: openthread/openthread
2023-06-01T16:48:02.1827445Z   token: ***
2023-06-01T16:48:02.1827639Z   ssh-strict: true
2023-06-01T16:48:02.1827856Z   persist-credentials: true
2023-06-01T16:48:02.1828071Z   clean: true
2023-06-01T16:48:02.1828252Z   fetch-depth: 1
2023-06-01T16:48:02.1828441Z   lfs: false
2023-06-01T16:48:02.1828644Z   set-safe-directory: true
2023-06-01T16:48:02.1828842Z ##[endgroup]
2023-06-01T16:48:02.1832152Z ##[command]/usr/bin/docker exec  a302f54f5e3393ff720c846c9eda75a91372fd4174e190a8a9f4d9dc841b79bb sh -c "cat /etc/*release | grep ^ID"
2023-06-01T16:48:02.4809067Z Syncing repository: openthread/openthread
2023-06-01T16:48:02.4810712Z ##[group]Getting Git version info
2023-06-01T16:48:02.4811151Z Working directory is '/__w/openthread/openthread'
2023-06-01T16:48:02.4811689Z [command]/usr/bin/git version
2023-06-01T16:48:02.4811911Z git version 2.17.1
2023-06-01T16:48:02.4825770Z ##[endgroup]
2023-06-01T16:48:02.4829099Z Deleting the contents of '/__w/openthread/openthread'
2023-06-01T16:48:02.4834427Z The repository will be downloaded using the GitHub REST API
2023-06-01T16:48:02.4834814Z To create a local Git repository instead, add Git 2.18 or higher to the PATH
2023-06-01T16:48:02.4886356Z ##[error]Input 'submodules' not supported when falling back to download using the GitHub REST API. To create a local Git repository instead, add Git 2.18 or higher to the PATH.
2023-06-01T16:48:02.5017774Z Post job cleanup.
```

abtink, 2023-06-01:
Looks like the issue is with the container used for this:
```
    container:
      image: openthread/environment
```
It needs to have git version > 2.18.
(relevant/similar discussion -> https://github.com/actions/checkout/issues/758#issuecomment-1426266741)

To unblock, for now, submitted https://github.com/openthread/openthread/pull/9123 to skip this job.

jwhui, 2023-06-01:
Thanks for looking into this, @abtink !

I triggered a new build of the `openthread/environment` image. The checks seems to be passing now.

closed 2023-06-01 by jwhui