openthread/openthread #9196: Indirect TX abort does not increment attempt counter

Damian-Nordic, 2023-06-21:
**Describe the bug**
I am not sure if this is a bug but I spotted a discrepancy that I cannot explain to myself :).

I was testing the following network topology:
- Leader: ot-daemon on Linux + RCP on nRF52840 DK
- (S)SED: OpenThread CLI on nRF52840 DK

At some point, the RCP entered a bad state (perhaps, due to my code or configuration changes but this is irrelevant for this issue) and started aborting frame transmissions. I realized that the host would never give up and it would try to send a single frame endlessly:
```
./build/posix/src/posix/ot-daemon[8773]: 00:14:36.808 [I] DataPollHandlr: Rx data poll, src:0x7c02, qed_msgs:1, rss:-23, ack-fp:1
./build/posix/src/posix/ot-daemon[8773]: 00:14:36.808 [I] Mac-----------: Frame tx attempt 1/1 failed, error:Abort, len:52, seqnum:89, type:Data, src:b24d1059a182c3ef, dst:4ebb0ec6842ad284, sec:yes, ackreq:yes
./build/posix/src/posix/ot-daemon[8773]: 00:14:37.809 [I] DataPollHandlr: Rx data poll, src:0x7c02, qed_msgs:1, rss:-23, ack-fp:1
./build/posix/src/posix/ot-daemon[8773]: 00:14:37.809 [I] Mac-----------: Frame tx attempt 1/1 failed, error:Abort, len:52, seqnum:90, type:Data, src:b24d1059a182c3ef, dst:4ebb0ec6842ad284, sec:yes, ackreq:yes
./build/posix/src/posix/ot-daemon[8773]: 00:14:37.913 [I] DataPollHandlr: Rx data poll, src:0x7c02, qed_msgs:1, rss:-23, ack-fp:1
./build/posix/src/posix/ot-daemon[8773]: 00:14:37.913 [I] Mac-----------: Frame tx attempt 1/1 failed, error:Abort, len:52, seqnum:91, type:Data, src:b24d1059a182c3ef, dst:4ebb0ec6842ad284, sec:yes, ackreq:yes
...
```

So I did more testing (with extra code for injecting TX aborts) and realized this:
- for direct transmissions, any TX error is treated the same way and the current message is simply removed from the queue
- for indirect transmissions, be it data polls or CSL, only `kErrorNoAck` error results in `aChild.IncrementIndirectTxAttempts()` and for other errors, such as `kErrorChannelAccessFailure` or `kErrorAbort` the attempt counter is not incremented, hence the transmitter continues sending the frame for ever.

Is there a justification for such a behavior? I know aborts are not something that typically happens multiple times in a row but still, shouldn't the stack recover from such conditions?

**To Reproduce**
OT commit: `d9abe3071c0131a4adb5d7e7451319b735e6d855`

**Expected behavior**
The sender eventually gives up after a given frame could not be transmitted despite of the reason.

**Console/log output**
N/A

**Additional context**
N/A

abtink, 2023-06-21:
From what I recall, this was intentional (and I think was changed during very early stages of testings).

The idea being that we don't want to drop queued indirect msgs to a sleepy child too quickly (only after certain number of failed actual over-the-air radio tx attempts and not due to channel being busy or coex or radio issues).

Damian-Nordic, 2023-06-22:
@abtink Thanks for the insights. I am not sure if there shouldn't be some limit for the number of aborts just in case there is some incompatibility between the host and the RCP, but the issue shouldn't normally happen so if there was a reason to implement the current behavior, I will no try to change it. :)

closed 2023-06-22 by Damian-Nordic