openthread/openthread #9248: `Test...PlatfromGeneratedNd()` failure in `test_routing_manager` on GH actions

abtink, 2023-07-02:
Notice that the unit test `test_routing_manager` started failing on most recent runs.
Example -> https://github.com/openthread/openthread/actions/runs/5437152261/jobs/9887818089

Seems like `TestBorderRoutingProcessPlatfromGeneratedNd()` is failing on GH action runs. On local machine it is passing.

abtink, 2023-07-02:
This is run on my fork GH action with more logs enabled to see what is happening:
https://github.com/abtink/openthread/actions/runs/5437851554/jobs/9888677354

Seems like the PD prefix we get does not match what is expected we expect `2001:db8:cafe:0::/64` but get `2001:db8:cafe:1::/64`
```
02:50:53.000 otPlatInfraIfSendIcmp6Nd(aDestAddr: ff02:0:0:0:0:0:0:1, aBufferLength:80)
02:50:53.000   Router Advertisement message
02:50:53.000      RA header - lifetime 0, prf:med
02:50:53.000      PIO - fdde:ad00:beef:cafe::/64, flags:LA, valid:1800, preferred:1800
02:50:53.000      RIO - fd2b:c809:b469:1::/64, prf:med, lifetime:0
02:50:53.000      RIO - 2001:db8:cafe:1::/64, prf:med, lifetime:1800
02:51:00.000 VerifyPdOmrPrefix(2001:db8:cafe:0::/64)
02:51:00.000 In VerifyPdOmrPrefix(), we got 2001:db8:cafe:1::/64
```

This is at this step:
https://github.com/openthread/openthread/blob/8081664d37f7c4ab49da3915b5ac6e73491552cd/tests/unit/test_routing_manager.cpp#L3400-L3413

Looks like we are passing short `/48` prefix but when it gets extended to `/64` extra bits may not be cleared, so we ended up with `2001:db8:cafe:1::/64` instead of `2001:db8:cafe:0::/64`

abtink, 2023-07-02:
This PR should help address it
-  https://github.com/openthread/openthread/pull/9249

jwhui, 2023-07-10:
Resolved by #9249

closed 2023-07-10 by jwhui