openthread/openthread #9298: netlink: setsockopt `SOL_NETLINK` causes `otbr` to hang on startup

jdswensen, 2023-07-19:
**Describe the bug**

`otbr` is hanging on startup.

**To Reproduce** 

1. Git commit id
    - https://github.com/openthread/openthread/pull/9287 
3. IEEE 802.15.4 hardware platform
    - Nordic nRF52840 dongle
5. Build steps
    - Built with Yocto, standard BR setup. I can get the build flags if anyone thinks its related.
7. Network topology
    - None. `ot-br-posix` is failing to start. 😄 

**Expected behavior** 
Ideally, some combination of:
- Handle the error without hanging
- If this is a kernel config thing, guard the behavior around a different `#define`
    - I think `SOL_NETLINK` is always defined as `270` even if the protocol is not available
- Document what configuration is required to support the new behavior (if any)

**Console/log output**
```
[NOTE]-AGENT---: Thread version: 1.3.0
[NOTE]-AGENT---: Thread interface: wpan0
[NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
[NOTE]-ILS-----: Infra link selected: eth0
[INFO]-NCP-----: OpenThread log level changed to 5
49d.17:17:38.797 [I] Platform------: RCP reset: RESET_POWER_ON
49d.17:17:38.799 [I] Platform------: RCP reset: RESET_POWER_ON
49d.17:17:38.800 [I] Platform------: Software reset RCP successfully
49d.17:17:38.807 [I] Platform------: Backbone interface is configured to eth0 (2)
49d.17:17:38.809 [C] Platform------: platformConfigureNetLink() at netif.cpp:2007: Protocol not available
```

**Additional context**
The platform I'm running on has a pretty minimal distro and kernel config. I'm guessing I'm missing a library or a kconfig option.

jdswensen, 2023-07-19:
Let me know if there are any specific logs or other info you may need. It's 100% reproducible on my end if I have that recent `netlink` commit included.

wgtdkp, 2023-07-19:
```c++
        VerifyOrDie(setsockopt(sNetlinkFd, SOL_NETLINK, NETLINK_EXT_ACK, &enable, sizeof(enable)) == 0,
                    OT_EXIT_ERROR_ERRNO);
```

This line is failing. So it looks like we can compile but some how the feature `NETLINK_EXT_ACK` is not enabled your setup.

Let me loose the check to simply log a warning when it fails (We don't really depend on those two options to functioning

wgtdkp, 2023-07-19:
@jdswensen Cloud you help check if https://github.com/openthread/openthread/pull/9299 fixes your issue?

closed 2023-07-19 by jwhui