openthread/openthread #9300: Why pinging rloc address with routerId of zero (fd4e:2b22:1beb:e1b0:0:ff:fe00:0) is dropped

luyou-tuya, 2023-07-19:
There are three FTD devices and one otbr under the thread network. The otbr is powered off for a few minutes and then powered on. The otbr wants to be upgraded to a router device and is discarded when sending Send Address Solicit to the leader FTD device. The rloc of the leader device is "fd4e:2b22 :1beb:e1b0:0:ff:fe00:0”, ping can not be communicated.
 otbr is a sub-device of the leader device, it can communicate with him, and can update srp information to otbr, ping the addresses of ula and eid of the leader device can be communicated
otbr information is as follows：
[image]
[image]
[image]

The leader device information is as follows
[image]

otbr log file：
[MobaXterm_root@SmartGatewaytmptuya_20230719_151728.txt](https://github.com/openthread/openthread/files/12092949/MobaXterm_root%40SmartGatewaytmptuya_20230719_151728.txt)

git commit id
otbr
commit 7a6c826c83579f57fae889a2031580649586478b
Author: Abtin Keshavarzian <abtink@google.com>
Date:   Tue May 23 12:36:18 2023 -0700

Hardware platform：
silicon

jwhui, 2023-07-19:
Thanks for submitting this issue. From the information provided, it's not immediately obvious where the issue might be.

Can you provide a pcap file for analysis in wireshark? more verbose OTBR logs can help as well.

luyou-tuya, 2023-07-24:
> Thanks for submitting this issue. From the information provided, it's not immediately obvious where the issue might be.
> 
> Can you provide a pcap file for analysis in wireshark? more verbose OTBR logs can help as well.

Now the environment where the router id is zero is gone, and I am not reproducing it. If it appears later, I will provide the next data packet

jwhui, 2023-08-30:
Closing stale issue.

closed 2023-08-30 by jwhui