openthread/openthread #9333: A null pointer access may occur in Tasklet::Scheduler::ProcessQueuedTasklets

dsyx, 2023-08-01:
**Describe the bug**

After a few days of running my Thread device normally, it stopped. According to Ozone's report, it has a HardFault exception, which seems to be caused by accessing a null pointer.

**To Reproduce**

1. Git commit id: `91fa1f455` (Gecko SDK Suite v4.2.3)
2. IEEE 802.15.4 hardware platform: `MGM240PA32VNA`

I created the application from the template OpenThread BLE DMP – SoC Free RTOS with Simplicity Studio 5. A Thread-based TCP service is then built on top of this template. A machine on the external network accesses this TCP service for data once per second.

**Additional context**

Below is a screenshot of the Ozone report:

[image]

[image]

Ozone shows that the `tasklet` value at this time is `0x0000 0000`, so `tasklet->mNext` will trigger a null pointer access. Should the nullptr value judgment of tasklet be added after `Tasklet *tasklet = tail->mNext;`?

jwhui, 2023-08-01:
The tasklet queue should be maintained in a way that this should never happen.

OpenThread APIs are designed to be called from a single thread of execution. Can you confirm whether this is the case in your project?

dsyx, 2023-08-01:
Hi @jwhui ,

I know that OpenThread API is not thread safe, so I refer to [ot-rtos](https://github.com/openthread/ot-rtos) to wrap it, namely:

```c
#define OT_API_CALL(...)     \
    do                       \
    {                        \
        otrLock();           \
        __VA_ARGS__;         \
        otrUnlock();         \
        otrTaskNotifyGive(); \
    } while (0)
```

In my application, I call any OpenThread API with OT_API_CALL.

abtink, 2023-08-01:
The `Tasklet`s are stored as a *circular singly linked list*. So any posted `Tasklet` will always have a non-null `mNext` pointer (that is how we check if a `Tasklet` is posted in  `IsPosted()` method).

As @jwhui mentioned, behaviors like this are likely due to OT APIs and/or platform APIs and callback not being called from the same thread of execution. Another (though less likely) possibility is memory corruption.

jwhui, 2023-08-30:
Closing stale issue.

closed 2023-08-30 by jwhui