openthread/openthread #9358: CSL Rx window calculation does not consider CSL phase quantization error

EskoDijk, 2023-08-16:
**Describe the bug** A clear and concise description of what the bug is.

In `sub_mac.cpp` , `GetCslWindowEdges()` calculates, on a CSL receiver, the window by how much the CSL transmission could come early or late. This considers clock drift over time (CSL Accuracy) and the radio's inherent uncertainty (CSL Uncertainty).

The radio code (`radio.hpp`) calls the function `otPlatRadioUpdateCslSampleTime()` of the platform to communicate the CSL sample time (= start of sample window; radio-on time; start of first symbol of SHR). This should be used by the radio platform to derive the value of "CSL phase" for the next time that a frame with CSL IE will be sent out. CSL phase is in units of 160 us.

The problem is that the time parameter of `otPlatRadioUpdateCslSampleTime` could be any time value, not necessarily a multiple of 160 us. This implies that the radio platform's calculation of CSL Phase on the CSL receiver may contain an error w.r.t. the actual CSL sample window: a quantization error.

This error may be, depending on platform implementation, in the range [-80, +79] us, or [0, +159] us, or [-159,0] us.
And the CSL receiver does not take into account this error (additional source of error) that may be present on an OpenThread CSL transmitter.

Not sure if i see this problem correctly, a check would be appreciated.

**To Reproduce** Information to reproduce the behavior, including:

(See source code)  Experiment results could be added later if needed; using CSL logging (-DOT_CSL_DEBUG=ON).

**Expected behavior** A clear and concise description of what you expected to happen.

Ideally the CSL receiver should have its CSL sampling window adapted to match exactly to the value that was sent out in CSL IE.

Or as a (less ideal) alternative, the CSL receiver may indicate a minimum CSL Uncertainty parameter of 9 or 17 (resp for 90 us or 170 us) depending on platform implementation. 

**Console/log output** If applicable, add console/log output to help explain your problem.
N/A

LuDuda, 2023-08-16:
@edmont could you please take a look?

edmont, 2023-08-17:
Thanks @EskoDijk, this is indeed a fact that might have been overlooked by some radio implementations. In any case it's a platform implementation decision whether to improve CSL Receiver uncertainty or just account for it internally in `GetCslWindowEdges()`.

The only missing piece, in my opinion, is a radio API to get the different uncertainty for CSL Receiver and CSL Transmitter roles. Currently we only have `otPlatRadioGetCslUncertainty()`. So, what about splitting it into `otPlatRadioGetCslReceiverUncertainty()` and `otPlatRadioGetCslTransmitterUncertainty()`? Alternatively, a parameter could be added to the existing API: `bool aIsCslReceiver`. Spinel property should be extended as well (`SPINEL_PROP_RCP_CSL_UNCERTAINTY`).

EskoDijk, 2023-08-17:
@edmont Splitting could be a solution, but it would be easier, clearer and more precise to simply take this quantization effect into account already in the window calculation in `GetCslWindowEdges()` . E.g. extend the window end exactly with the needed quantity which is in the [0, 159] us range. And add some comments in the API on how the platform should calculate the CSL Phase integer (to have the rouding/truncation done correctly).  Maybe I can later make a PR for this; if such a solution is ok?

edmont, 2023-08-17:
There is a way to remove this rounding error completely: just re-schedule the receive window around a corrected `mCslSampleTime` based on transmitted CSL IE phase (assuming a known CSL Phase rounding approach). But this adds code complexity, so for now your proposal sounds okay.

But how do you plan to get the exact amount? Also from the sent CSL IE phase?