openthread/openthread #9386: [OT] transmit power parameter is not reset after calling "reset" or "factoryreset" commands from the host

RohithShanbhagh, 2023-08-30:
Describe the bug: 
In src/lib/spinel/radio_spinel_impl.hpp,  otError RadioSpinel<InterfaceType>::Enable(otInstance *aInstance) function does not re-initialize tx power during factoryreset.

Execution Steps:
ot-ctl txpower 3
ot-ctl txpower
ot-ctl factoryreset
ot-ctl txpower

Expected behavior:
transmit power should be set to default value.

jwhui, 2023-08-30:
Transmit Power is directly managed by the platform layer.

https://github.com/openthread/openthread/blob/e4d953c9bfbc52c4fd2871707e044d21a460ee72/src/cli/cli.cpp#L7322-L7340

I would suggest looking into the platform implementation or reaching out to your platform provider.

abtink, 2023-08-30:
I believe this may be the intended behavior (chosen by radio platform developers). These APIs, along with other radio "calibration" APIs (e.g., otPlatRadioAddCalibratedPower()), are intended to configure the radio (perhaps as part of the factory configuration itself). It makes sense for the platform to require the calibration information to persist until it is explicitly overwritten/changed

closed 2023-10-20 by jwhui