openthread/openthread #9394: Channel switching issue in MAC layer occurs when there is no NW formation

hastigondaliya, 2023-09-01:
Description of the bug
We encountered an issue while performing a scan energy without NW formation. The openthread stack follows two different flows for scan energy, with and without providing a time duration.

**1. scan energy without time duration**

- In this case, the MAC layer forces the firmware to change the channel by sending a spinel command (“cmd:PROP_VALUE_SET, key:PHY_CHAN,”).
> if (mChannel != aChannel)
{
error = Set(SPINEL_PROP_PHY_CHAN, SPINEL_DATATYPE_UINT8_S, aChannel);
SuccessOrExit(error);
mChannel = aChannel;
}

- Spinel property set handler updates the same channel to mReceiveChannel in link_raw.cpp

**2. scan energy with time duration**

- In this case, MAC layer calls energy scan to set 3 spinel property
> SuccessOrExit(error = Set(SPINEL_PROP_MAC_SCAN_MASK, SPINEL_DATATYPE_DATA_S, &aScanChannel, sizeof(uint8_t)));
SuccessOrExit(error = Set(SPINEL_PROP_MAC_SCAN_PERIOD, SPINEL_DATATYPE_UINT16_S, aScanDuration));
SuccessOrExit(error = Set(SPINEL_PROP_MAC_SCAN_STATE, SPINEL_DATATYPE_UINT8_S, SPINEL_SCAN_STATE_ENERGY));

- Spinel property set handler register one callback LinkRawEnergyScanDone to otLinkRawEnergyScan API

- LinkRawEnergyScanDone callback trigger otPlatRadioReceive with mReceiveChannel which updates the same channel in firmware.

- So when scan energy performs with time duration, it doesn't update mReceiveChannel for the callback.

Attached flow diagram for scan energy with zero/non-zero duration for reference.
[image]

**To Reproduce**
After execution of the below commands,

1. **scan energy** -> `mChannel:25, aChannel:26, mReceiveChannel:26, fw Channel:26`
2. **scan energy 11** -> `mChannel:26 and aChannel:11, mReceiveChannel: 26, fw Channel: 26`
3. **scan energy** -> `mChannel:11, aChannel:11, mReceiveChannel:26`
In this scenario, mChannel and aChannel will be the same while scanning for channel 11, which means the MAC doesn't send a spinel command to update the channel. However, the firmware channel is still set to 26. As s result, the scan will be conducted for channel 26 instead of channel 11.

**Expected behavior**
The channel should be updated with the given channel in the MAC layer

**Fix**
To set the channel while scan energy with time duration, need to add the below lines in `PerformEnergyScan` (mac.cpp). 
Attached snap for reference.
>  if(!GetRxOnWhenIdle())
    {
        SuccessOrAssert(Get<Radio>().Receive(mScanChannel));
    }
[image]

abtink, 2023-09-05:
Thanks @hastigondaliya for reporting this and the nice and detailed analysis.

I agree that there may be an issue here. The root cause seems to be ambiguity in how `otLinkRawEnergyScan()` (and in turn `otPlatRadioEnergyScan()`) are expected to work after the scan is done: Should the radio go back to its previous channel, or 
should it stay on the channel where the scan was performed? 

- The `radio_spinel` layer seems to assume that radio will stay on the new scan channel as we see that it sets its local copy of `mChannel` from `RadioSpinel::EnergyScan()` to the given scan channel.
- While `ncp_base` and `LinkRaw` assume that channel should switch back after the scan is done:
https://github.com/openthread/openthread/blob/c6bece0729f0c962766ccbf01d2178e984c9f8d4/src/ncp/ncp_base_radio.cpp#L198-L207
- And `LinkRaw` does not change its copy of `mReceiveChannel` when `EnergyScan()` is requested. So it goes back to previous channel
- This can then leads to inconsistency between what `RadioSpinel` is tracking vs channel RCP and radio are on.

I can see two ways of fixing this:
1. We clarify and document one behavior as expected and update `RadioSpinel` and/or `LinkRaw`.
1. Or the fix you proposed which always issue `Radio::Receive()` on scan channel before actually calling the radio energy scan API (remove any ambiguity by forcing channel change)

While I feel option 1 may be more correct, I think it can be tricky as it may create incompatibility between RCP and host (specially if there are running different versions of software). So I think option 2 may be overall better.

Thoughts?

BTW, can you submit a PR with your proposed fix? (if not let me know, I can do so on your behalf).

hastigondaliya, 2023-09-06:
Hi @abtink,

Thank you for your input. 
I'll submit PR with our proposed fix.

PR #9405 (2023-09-13): [mac] fix channel switching issue during energy scan
- This commit resolves inconsistency between Host and RCP channel on scan energy when network not created.
- Added fix in MAC layer to update the channel on scan energy

Fixes #9394
closed 2023-09-13 by jwhui