openthread/openthread #9409: Router can seem to hold onto invalid eidcache for node

ndyck14, 2023-09-09:
**Describe the bug** I'm seeing a case where a BR is holding on to a stale eidcache 
```
> eidcache
fd2f:2067:77a8:1:1058:fc0e:1593:be16 a000 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:f22f:fba6:1516:7836
fd2f:2067:77a8:1:c26f:97e:4b45:38b8 6c00 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:2c41:66e6:2550:2852
fd2f:2067:77a8:1:6280:ce57:a63f:46a 2800 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:3bed:708c:6450:8cfd
fd2f:2067:77a8:1:3d32:5604:767d:aae3 0c00 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:5de9:f10e:2f2:e038
fd2f:2067:77a8:1:b856:fcfa:4589:2041 1400 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:1e12:7b6d:8fb7:8e23
fd2f:2067:77a8:1:afa6:c0dd:9b4:7de8 8405 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:d14e:bb38:b45c:3534
fd2f:2067:77a8:1:65da:8596:83ac:b473 b407 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:b24:8d26:b41e:8c7e
fd2f:2067:77a8:1:d95b:4002:5e97:5977 e400 cache canEvict=1
fd2f:2067:77a8:1:8ab3:9c5c:da27:d2c2 b800 cache canEvict=1
fd2f:2067:77a8:1:b551:fd41:be70:c402 8402 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:cc3c:21ad:e1d6:e50d
fd2f:2067:77a8:1:89ea:9674:c15a:104d 8404 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:35dc:3a10:9444:4735
fd2f:2067:77a8:1:3556:8c33:2bf8:e1f5 8403 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:332d:2213:bee4:891c
fd2f:2067:77a8:1:4cd2:9fca:97f:123 ac00 cache canEvict=1
fd2f:2067:77a8:1:aee3:de98:cefb:2e7e b400 cache canEvict=1
fd2f:2067:77a8:1:f9aa:9b28:50ab:79d3 e800 cache canEvict=1
fd2f:2067:77a8:1:428e:9739:356f:fce4 4c00 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:6cf2:8b1b:862:7158
fd2f:2067:77a8:1:b3f0:3f3c:add3:84ba 8400 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:a337:8500:af27:6248
fd2f:2067:77a8:1:2dcd:812b:cf3d:1094 8400 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:d14c:296d:dc47:1fe1
fd2f:2067:77a8:1:d50c:81d1:c8d5:169b a001 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:5ed9:5c3d:e094:8249
fdcf:8d62:8570:0:5b47:c1f6:474a:9a70 8401 cache canEvict=1
fd2f:2067:77a8:1:9c52:ea6f:df6d:1cb bc00 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:869c:644d:1efb:5728
fd2f:2067:77a8:1:4c72:3c36:3f42:3cc7 1800 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:bf89:3582:da9e:1381
fd2f:2067:77a8:1:f872:74a4:fa8d:4682 8401 cache canEvict=1 transTime=0 eid=fdcf:8d62:8570:0:5b47:c1f6:474a:9a70
fd2f:2067:77a8:1:2b03:5960:d39a:6583 fffe retry canEvict=1 timeout=0 retryDelay=30
Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
|  3 | 0x0c00 |       33 |         1 |     1 |      1 |   7 | e62bf306b0f70c4f |    1 |
|  5 | 0x1400 |       33 |         2 |     2 |      3 |   7 | c66e1574d2735c85 |    1 |
|  6 | 0x1800 |       33 |         1 |     2 |      2 |  15 | 3ae75f8b8a91c239 |    1 |
| 10 | 0x2800 |       33 |         1 |     1 |      2 |  26 | dea9d8ec0225a0ff |    1 |
| 20 | 0x5000 |       63 |         0 |     0 |      0 |   0 | 0000000000000000 |    0 |
| 27 | 0x6c00 |       46 |         3 |     1 |      1 |  25 | 06d838f00155ca5d |    1 |
| 33 | 0x8400 |       51 |         1 |     3 |      3 |   3 | d2483ee530c6b41f |    1 |
| 36 | 0x9000 |       51 |         4 |     0 |      0 |  19 | 8af4955487e9f116 |    0 |
| 40 | 0xa000 |       51 |         1 |     1 |      1 |  24 | 727e27068c33dd2a |    1 |
| 43 | 0xac00 |       33 |         2 |     1 |      2 |  86 | 06eb0dc305d53eca |    1 |
| 45 | 0xb400 |       33 |         2 |     1 |      2 |  12 | 7299cf87474fffcc |    1 |
| 46 | 0xb800 |       33 |         2 |     2 |      2 |  15 | 16847a5afb1782f9 |    1 |
| 47 | 0xbc00 |       33 |         2 |     1 |      2 |  27 | 1af3a011a1df3784 |    1 |
| 51 | 0xcc00 |       33 |         1 |     3 |      3 |  11 | baf737a9e2c4ee08 |    1 |
| 57 | 0xe400 |       33 |         2 |     1 |      2 |  31 | 8ef6d4853cef4748 |    1 |
| 58 | 0xe800 |       33 |         2 |     1 |      2 |  23 | ae5014e64a8276b1 |    1 |
```
Address in question is `fd2f:2067:77a8:1:428e:9739:356f:fce4` and any packets coming through this node are dropped due to noroute, which i presume is because the rloc in the cache, `0x4c00` is not in the router table. It seems to me, this should remove the entry from the cache and force an address query, but that does not seem to be happening:
```
> ping fd2f:2067:77a8:1:428e:9739:356f:fce4
1 packets transmitted, 0 packets received. Packet loss = 100.0%.
Done
...
Sat Sep  9 11:47:19 2023 user.notice otbr-agent[1468]: 20:48:42.931 [N] MeshForwarder-: Dropping IPv6 ICMP6 msg, len:56, chksum:4cd5, ecn:no, sec:yes, error:NoRoute, prio:normal
Sat Sep  9 11:47:19 2023 user.notice otbr-agent[1468]: 20:48:42.931 [N] MeshForwarder-:     src:[fd2f:2067:77a8:1:18d3:f1f8:f5b4:b0d]
Sat Sep  9 11:47:19 2023 user.notice otbr-agent[1468]: 20:48:42.931 [N] MeshForwarder-:     dst:[fd2f:2067:77a8:1:428e:9739:356f:fce4]
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: checking (last ~3 months)
2. IEEE 802.15.4 hardware platform silabs mg21 rcp
3. Build steps
4. Network topology: ~20 devices.

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2023-09-09:
@ndyck14 , thanks for raising this issue. I agree what you are observing should not happen.

After inspecting the code, it looks like we may need to add some logic for:
1. Invalidating cache entries for router IDs that are removed from the Router ID Set.
2. Avoid creating cache entries for router IDs that are not in the active Router ID Set. I think this could happen due to snooping packets that may be delayed.

abtink, 2023-09-09:
Thanks @ndyck14. This is intresting issue. I agree with Jonthan.

I think (1) logic may already exist. When Router ID set is released we would have:
- On a router, `UpdateRouterIdSet()` will call `RemoveRouterLink()` for any previously allocated ID that is not longer valid in the new set, which in turn, will call `Get<AddressResolver>().RemoveEntriesForRouterId()`.
- On a leader, we do the same from `RouterTable::Release()` when a Router ID is released.

I think this cache entry is likely created using snoop mechanism?

jwhui, 2023-09-09:
> On a router, UpdateRouterIdSet() will call RemoveRouterLink() for any previously allocated ID that is not longer valid in the new Router ID set, which in turn, will call Get<AddressResolver>().RemoveEntriesForRouterId() for it.

Does it remove cache entries in all cases? `RemoveRouterLink()` removes the link to the neighboring router and occurs before the router entry is actually removed from the locally-stored Router ID Set. In particular, when the following condition is `false`:

https://github.com/openthread/openthread/blob/0b269e12b9c313d9453ad5017a7588cec3eea75b/src/core/thread/router_table.cpp#L253-L259

abtink, 2023-09-09:
Unless I am mistaken, I think it should be removed (at least when a previously allocated Router ID is removed from the set). We have the following in `UpdateRouterIdSet()` which sets the next hop to invalid `SetNextHopToInvalid()` on `router` and calls `RemoveRouterLink(*router)`:

https://github.com/openthread/openthread/blob/0b269e12b9c313d9453ad5017a7588cec3eea75b/src/core/thread/router_table.cpp#L521-L544

abtink, 2023-09-09:
Submitted https://github.com/openthread/openthread/pull/9410 as a proposed solution. It should safeguard against this situation. Thoughts? @jwhui ?

jwhui, 2023-09-09:
> Submitted #9410 as a proposed solution. It should safeguard against this situation. Thoughts? @jwhui ?

I think this can help avoid break out of the issue described above. However, I think it still means that the given packet is dropped. Maybe we should consider one or both of the following?
1. Ensure that we trigger an Address Query whenever this occurs.
2. Whenever adding a cache entry, validate that there is a valid route.

ndyck14, 2023-09-10:
I should have mentioned, although it may be evident from the router table, that this BR does not have a direct link with the node in question.

abtink, 2023-09-10:
Submitted #9411. 

> Whenever adding a cache entry, validate that there is a valid route.

On this, I feel we should not check when adding a new entry and instead leave the check to when we want to use the entry to resolve. The situation I have in mind is we create an entry through snoop mechanism from a message from a newly promoted router and we have not yet learned next hop for this new router but will learn soon when advertisements are sent.

ndyck14, 2023-09-11:
agreed that trying to check up front when adding cache entry shouldnt be necessary, and further could be detrimental.

in terms of deployment, this seems obviously urgent for BRs. in terms of end devices (FTD/MTD), what's your take on severity of impact here?

abtink, 2023-09-11:
@ndyck14 as I mentioned, there seems to be already existing mechanism in the code that should remove all stale cache entries once the related Router ID is removed. I expect this to kick in and clear stale cache entries in almost all situations when a router is removed from the network..

The question is, how did the example you provided end up in this state of keeping the stale entry?
- The guess/theory is that this may have happened due to a snoop optimization that allocated an entry for a new router that somehow got offline quickly, before the device had a chance to discover it and add it to its router table. And then the router  later attached with a new/different router ID with the same IPv6 address.
- While this is technically possible, I feel it is not a likely situation to happen.
- So if this is the case and the assumption above about how we get to this state is correct, then I would say this issue is not critical.
- Still, I would recommend fixing it and adding it to all devices.

jwhui, 2023-09-12:
> On this, I feel we should not check when adding a new entry and instead leave the check to when we want to use the entry to resolve. The situation I have in mind is we create an entry through snoop [...]

Yeah, I was wondering about this delay as well. In many cases, the snoop entry will be created via a request initiated by the Thread device. In this case, the response should come pretty quickly afterward, which would invalidate the cache entry anyway. Should we implement some time delay for not evicting the cache entry?

abtink, 2023-09-12:
> Should we implement some time delay for not evicting the cache entry?

Yes. I think this would be a good idea. But I suggest it for a later/separate PR. I am hoping to keep the fix for this (in https://github.com/openthread/openthread/pull/9411) relatively small (so easier to cherry-pick if needed).

jwhui, 2023-09-13:
@ndyck14 , I just merged #9411. Can you help test whether #9411 addresses your issue?

ndyck14, 2023-09-13:
I will check if the state still exists - i believe it evaporated unfortunately.

ndyck14, 2023-09-13:
relating to a/aq i was debugging a device that had actually gone offline (unreliable PSU smh). but i noticed that a/aq seems to have some timeout in eidcache? how does that work? if something is actually unreachable, it seems like a/aq will not fire again until after the timeout (120s iirc)? that does seem potentially too high? can someone explain what actually happens (or doesnt) during that time?

jwhui, 2023-09-13:
> relating to a/aq i was debugging a device that had actually gone offline (unreliable PSU smh). but i noticed that a/aq seems to have some timeout in eidcache? how does that work? if something is [...]

See [SPEC-1164](https://threadgroup.atlassian.net/browse/SPEC-1164) regarding the current implementation.

closed 2023-10-20 by jwhui