openthread/openthread #9415: Off mesh routing for children does not work in some cases

ndyck14, 2023-09-11:
**Describe the bug** A clear and concise description of what the bug is.
Off mesh routing for children does not work in some cases

**To Reproduce** Information to reproduce the behavior, including:
Unclear. Using an MG21 based FTD, off mesh pings are unable to return from child device. Pings from same address to parent OMR IPv6 address work as expected.

message me directly for networkkey if you need, here's snippet:

from pcap:
```
5873	Sep 11, 2023 09:16:37.905825000 EDT	575.076170	fda8:29b7:197d:1:18ff:de02:8708:7ee1	::65da:8596:83ac:b473	ICMPv6	75	Echo (ping) request id=0xec87, seq=15, hop limit=63 (reply in 5875)	
5875	Sep 11, 2023 09:16:37.923126000 EDT	575.093471	::65da:8596:83ac:b473	fda8:29b7:197d:1:18ff:de02:8708:7ee1	ICMPv6	71	Echo (ping) reply id=0xec87, seq=15, hop limit=64 (request in 5873)	
5877	Sep 11, 2023 09:16:37.929036000 EDT	575.099381	fd00:db8::ff:fe00:e800	fd00:db8::ff:fe00:e816	ICMPv6	72	Destination Unreachable (no route to destination)	
```
[image]

From other BR node:
```
> netdata show
Prefixes:
fd2f:2067:77a8:1::/64 paros low 5000
Routes:
::/0 s med 9800
::/0 s med cc00
::/0 s med 5000
::/0 s med 8400
Services:
44970 5d fdcf8d62857000009576444e75689eb4d120 s cc00
44970 5d fdcf8d628570000079306d87f36d3acbd128 s 5000
Contexts:
fd2f:2067:77a8:1::/64 1 c
Done
```

1. Git commit id 8f92d2dc8
2. IEEE 802.15.4 hardware platform mg21
3. Build steps
4. Network topology ~20 thread devices

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.
[ota cap child no routing.pcapng.zip](https://github.com/openthread/openthread/files/12577051/ota.cap.child.no.routing.pcapng.zip)

abtink, 2023-09-11:
BRs are using "::/0" for default route.

Does the child include the commit from PR https://github.com/openthread/openthread/pull/6784 (this is July 2021). Without this the zero length route prefix is treated as invalid. 

The commit id of https://github.com/openthread/openthread/commit/7f3013cb06fd3f61aed2d7820fb7c08e2fbc6be2 seems rather old from 2020.

jwhui, 2023-09-11:
The network data output shows using both the "default route flag" and "zero-length default route prefix".

@abtink , shouldn't the default route flag still allow routing the packet out (via 0x5000)?

abtink, 2023-09-11:
@jwhui yes, you are correct. I missed that. The "default route flag" should work for this.

----

Another thing I noticed is the destination of ping is shown as `fda8:29b7:197d:1:18ff:de02:8708:7ee1` which seems to be mesh-lcoal address. The OMR prefix is `fd2f:2067:77a8:1::`?

jwhui, 2023-09-11:
> Another thing I noticed is the destination of ping is shown as `fda8:29b7:197d:1:18ff:de02:8708:7ee1` which seems to be mesh-lcoal address. The OMR prefix is `fd2f:2067:77a8:1::`?

I think `fda8:29b7:197d:1::/64` is the AIL on-link prefix and `fd2f:2067:77a8:1::` is the Thread on-mesh prefix.

ndyck14, 2023-09-12:
yes, `fda8` is AIL.

abtink, 2023-09-12:
Is `34ecac8` the BR's commit? Do you know what is the commit of the child `0x812`

From the logs seems like the child `0x8412`, (which should be FTD) send a mesh frame with an dest RLCO16 in mesh header that is not reachable?

ndyck14, 2023-09-12:
in the newer BR parent example, child is at commit 8f92d2dc8 + some cherry picks. The example is analogous. `0x8412` responds to ping by addressing its parent `0x8400` (BR). The BR for some reason thinks the off mesh destination is unreachable, so drops the packet and responds to its child with the ICMPv6 type 1 (destination unreachable)

abtink, 2023-09-12:
> The BR for some reason thinks the off mesh destination is unreachable, so drops the packet and responds to its child with the ICMPv6 type 1 (destination unreachable)

I think, in this case the BR is not doing any IPv6 destination check, since what is receives is "lowpan mesh header " frame. In such a frame the sender includes the mesh source and mesh destination RLOC16s (in lowpan mesh header) and then the intermediate node will just forward it onward towards the destination RLOC (they won't check the IPv6 header of message). Unfortunately the log line below does not show the mesh header src/dest RLOCs (just previous sender's src address).

```
[I] MeshForwarder-: Dropping rx mesh frame, error:NoRoute, len:44, src:0x8412, sec:yes
```
So if my theory is correct, then the issue is with the child `0x8412` selecting a bad destination RLOC in mesh header frame it sent?

BTW, the child `0x8412` should be FTD since it is sending/receiving mesh header frames (if it was MTD it would delegate it to its parent).

ndyck14, 2023-09-12:
child is FTD, yes. Here's a pcap: 
[ota cap child no routing from br.pcapng.zip](https://github.com/openthread/openthread/files/12588763/ota.cap.child.no.routing.from.br.pcapng.zip)
```
1910	Sep 12, 2023 12:27:07.410743000 EDT	167.308137	fda8:29b7:197d:1:18ff:de02:8708:7ee1	::4c72:3c36:3f42:3cc7	ICMPv6	75	Echo (ping) request id=0x4760, seq=8, hop limit=63 (reply in 1912)	
1912	Sep 12, 2023 12:27:07.429046000 EDT	167.326440	::4c72:3c36:3f42:3cc7	fda8:29b7:197d:1:18ff:de02:8708:7ee1	ICMPv6	71	Echo (ping) reply id=0x4760, seq=8, hop limit=64 (request in 1910)	
1914	Sep 12, 2023 12:27:07.446354000 EDT	167.343748	fd00:db8::ff:fe00:8400	fd00:db8::ff:fe00:8412	ICMPv6	72	Destination Unreachable (no route to destination)	
```

ndyck14, 2023-09-12:
OK, so update, we've traced this to a phantom routing entry in the netdata from `0x9800`. This does not exist in the leader neighbor table, the routing table, or in `nexthop` data, so its unclear how it got into this state. Rebooting the leader also does not clear the phantom entry, which is at least 1 area of possible improvement/mitigation. @abtink will look further into next steps.

jwhui, 2023-09-13:
Resetting the Leader does not clear the phantom entry because the Leader will restore the network data from a neighboring device after reboot.

But I think this raises one possibility. The Router ID Set and Network Data are versioned separately. It might be possible for a new Router ID Set and Network Data to be communicated in different messages. If the Leader happens to reboot during this period, then the old entry can remain in the Network Data and not be properly cleaned up.

I think we should add a check in `SetNetworkData()` when the Leader is restoring Network Data and clean out any Router IDs that do not exist in the Router ID Set. Only the Leader should do this, as the Leader is authoritative.

@abtink , thoughts?

ndyck14, 2023-09-13:
+1 for policing. might be better to break off into separate issue. i believe we already have leader takeover?

abtink, 2023-09-13:
- Submitted https://github.com/openthread/openthread/pull/9421

ndyck14, 2023-11-08:
@jwhui we just observed another case of this. @abtink re-reading your suggestion, it seems unclear how another node can check this and help old leaders clear things out? An alternative approach could be to have policing device force leader takeover which i think should populate a new partition? Although now i'm unusure if thats possible since i read the spec more deeply around singleton partition

ndyck14, 2023-11-16:
This is with existing leaders that do not have the fix. We don't have a solid reproducer so can't verify #9421

wimleers, 2024-02-21:
👋 Experienced software engineer here, but merely an end user wrt Thread/Matter.

Stumbled upon this issue because ever since Nanoleaf Essentials GU10 (Thread+Matter) bulbs were added to my Apple Home-only Thread network, my Eve Motion (Thread+HomeKit) devices frequently stop working, after >1 year of working with zero issues.

I have 34 Thread devices: 2 HomePod Mini, 1 Apple TV, 2 Eve Motion (HomeKit), 9 Eve Energy (HomeKit), 3 Nanoleaf Lightstrip (HomeKit), 6 Nanoleaf Essentials E27 (HomeKit), 1 Qingping motion sensor (HomeKit), 1 Eve Weather (HomeKit), 9 Nanoleaf GU10 (Matter).
(All on the latest firmware, except for Eve devices where the firmware upgrade would imply moving to Matter, with a feature set reduction.)

Nanoleaf GU10s are on firmware 3.5.41 (the latest since Sep 2023). They're all on the same Thread network. Oddly, the GU10s are all ["Endpoints" according the Eve app](https://www.evehome.com/en/blog/discover-your-thread-network) (which AFAICT is Eve speak for a [_router-eligible end device_](https://openthread.io/guides/thread-primer/node-roles-and-types)), but in the Nanoleaf app they show up as either "Child" or "Mesh Extender".

If you can point me to documentation on how to gather necessary debugging info to 1) verify I can reproduce this problem, 2) perform whichever analysis you'd like, let me know. 🙏

Shadoxity, 2024-02-21:
> 👋 Experienced software engineer here, but merely an end user wrt Thread/Matter.
>
> Stumbled upon this issue because ever since Nanoleaf Essentials GU10 (Thread+Matter) bulbs were added to my Apple [...]

Get the beta, nanoleaf have a kB on how to get access. The latest beta firmware has been way more stable and consistent

wimleers, 2024-02-21:
@Shadoxity Which version are you on? Are there release notes? I already signed up for a developer account but their developer forum is a graveyard 😳

Shadoxity, 2024-02-21:
@wimleers the latest which is 3.6.136

Join the discord, all the chat is happening there and in the last month nanoleaf has committed some resource in there and they are active with us.
The pinned notes will have a link and password to the beta info with list of current identified issues being worked on and a daily survey you can do to say how stable etc things have been.

In case you saw something different this is how to get the beta firmware

https://helpdesk.nanoleaf.me/en-US/how-do-i-sign-up-for-beta-software-updates-for-my-nanoleaf-lights-341522

wimleers, 2024-02-23:
@Shadoxity I've requested _multiple times now_ over the past few days to join the beta program (filled out the form). Still nothing, sadly. So no beta firmware for me yet.

I just got an additional Aqara FP2. Replaced the bathroom Eve Motion with that: bathroom Nanoleaf GU10s working again. The Eve Motion moved to a different room with fewer Nanoleaf GU10s: works fine again.

Still have the one Nanoleaf GU10 + Eve Motion that aren't working out, so still happy to test. (Also fine with moving the Eve Motion back to reproduce the problem again. It's consistently reproducible, so … if it helps sort this out, I'll do it!)

Shadoxity, 2024-02-23:
@wimleers there shouldn't be anything to fill out and wait for. It's sign up to the forum and then go into your services and you can opt then into beta, then the app will show the update.

wimleers, 2024-02-23:
> @wimleers there shouldn't be anything to fill out and wait for. It's sign up to the forum and then go into your services and you can opt then into beta, then the app will show the update.

OMG talk about a _terrible_ UX 🤯🤣 That worked, thanks! 🙏

jwhui, 2024-03-12:
Closing stale issue.

closed 2024-03-12 by jwhui