openthread/openthread #9434: [core] SRP adv-proxy and discovery-proxy in OT core

abtink, 2023-09-19:
This issue is used to track to all the PRs related to DNS/SRP and mainly the new implementation of SRP Advertising Proxy and Discovery Proxy in OT core. It includes list of PRs in `openthread` repo and PRs in `ot-br-posix`:

In `openthread` repo:
- https://github.com/openthread/openthread/pull/9054
- https://github.com/openthread/openthread/pull/9281
- https://github.com/openthread/openthread/pull/9331
- https://github.com/openthread/openthread/pull/9334
- https://github.com/openthread/openthread/pull/9339
- https://github.com/openthread/openthread/pull/9349
- https://github.com/openthread/openthread/pull/9353
- https://github.com/openthread/openthread/pull/9268
- https://github.com/openthread/openthread/pull/9362
- https://github.com/openthread/openthread/pull/9373

In `ot-br-posix` repo:
- https://github.com/openthread/ot-br-posix/pull/1986
- https://github.com/openthread/ot-br-posix/pull/1992
- https://github.com/openthread/ot-br-posix/pull/1996
- https://github.com/openthread/ot-br-posix/pull/1999
- https://github.com/openthread/ot-br-posix/pull/2000
- https://github.com/openthread/ot-br-posix/pull/2004
- https://github.com/openthread/ot-br-posix/pull/2006
- https://github.com/openthread/ot-br-posix/pull/2012
- https://github.com/openthread/ot-br-posix/pull/2013
- https://github.com/openthread/ot-br-posix/pull/2008
- https://github.com/openthread/ot-br-posix/pull/2019
- https://github.com/openthread/ot-br-posix/pull/2022

abtink, 2025-01-30:
PRs related to standalone mDNS library:

- https://github.com/openthread/openthread/pull/11182
- https://github.com/openthread/openthread/pull/11190
- https://github.com/openthread/openthread/pull/11193
- https://github.com/openthread/openthread/pull/11183
- https://github.com/openthread/openthread/pull/11191

EskoDijk, 2025-07-17:
@abtink It looks like this issue can be closed, with PRs merged, and new work moved to 11471 ?