openthread/openthread #9446: otbr-agent crashes if there is an RCP recovery triggered before otbr is initialized

spalas1996, 2023-09-20:
**Describe the bug** otbr-agent crashes after reboot in the scenario where there is an rcp failure detected even before otbr-agent initializes.
The rcp recovery code kicks in which changes mstate of RadioSpinel to kStateSleep from kStateDisabled.
Hence after recovery when otbr-agent initializes and calls RadioSpinel::Enable(), the following check fails: VerifyOrExit(!IsEnabled()) in Enable() and hence RadioSpinel is not enabled properly which leads to  aInstance being set as Null pointer that leads to a crash later on.

**To Reproduce** 
The issue is reproduced only sometimes when there is an RCP failure right before otbr is initialized.

**Expected behavior**
otbr should initialize successfuly even if there is an rcp failure before otbr initialization. 

**Console/log output** 
Crash log
235 08-29 20:35:40.420  2936  2936 W otbr-agent: 00:00:00.214 [W] Platform------: RadioSpinel Transmit 0x0
236 08-29 20:35:40.420  2936  2936 W otbr-agent: 00:00:00.215 [W] Platform------: otPlatRadioTxStarted 0x0
237 08-29 20:35:40.423  2936  2936 F libc    : Fatal signal 11 (SIGSEGV), code 1, fault addr 0x1632c in tid 2936 (otbr-agent)
238 08-29 20:35:40.495  2957  2957 F DEBUG   : pid: 2936, tid: 2936, oom: -1000, name: otbr-agent  >>> /system/bin/otbr-agent <<<

Additional debug logging added when issue happens
we can see in below log when RadioSpinel::Enable() is called mState is 1
09-01 20:49:14.860  3012  3012 I otbr-agent: 49d.17:03:58.681 [N] Platform------: RCP recovery is done
09-01 20:49:14.861  3012  3012 D otbr-agent: 49d.17:03:58.682 [D] Platform------: Wait response: tid=5 key=1
09-01 20:49:14.863  3012  3012 D otbr-agent: 49d.17:03:58.684 [D] Platform------: Wait response: tid=6 key=2
09-01 20:49:14.872  3012  3012 D otbr-agent: 49d.17:03:58.693 [D] Platform------: Wait response: tid=7 key=8
09-01 20:49:14.875  3012  3012 D otbr-agent: 49d.17:03:58.696 [D] Platform------: Wait response: tid=8 key=5
09-01 20:49:14.879  3012  3012 D otbr-agent: 49d.17:03:58.700 [D] Platform------: Wait response: tid=9 key=176
09-01 20:49:14.885  3012  3012 I otbr-agent: 49d.17:03:58.706 [N] Platform------: RCP API Version: 5
09-01 20:49:14.885  3012  3012 D otbr-agent: 49d.17:03:58.707 [D] Platform------: Wait response: tid=10 key=4619
09-01 20:49:14.904  3012  3012 W otbr-agent: 49d.17:03:58.725 [W] Platform------: otSysInit 2
09-01 20:49:14.904  3012  3012 W otbr-agent: 49d.17:03:58.725 [W] Platform------: otSysInit 3
09-01 20:49:14.904  3012  3012 W otbr-agent: 49d.17:03:58.726 [W] Instance------: InitSingle 1
09-01 20:49:14.905  3012  3012 W otbr-agent: 49d.17:03:58.726 [W] Instance------: InitSingle 0xad7eb288
09-01 20:49:14.905  3012  3012 W otbr-agent: 49d.17:03:58.726 [W] Instance------: InitSingle new instance 0xad7eb288
09-01 20:49:14.911  3012  3012 D otbr-agent: 00:00:00.002 [D] Platform------: Wait response: tid=11 key=34
09-01 20:49:14.919  3012  3012 W otbr-agent: 00:00:00.010 [W] SubMac--------: SubMac::Enable start
09-01 20:49:14.919  3012  3012 W otbr-agent: 00:00:00.010 [W] SubMac--------: SubMac::Enable 1
09-01 20:49:14.919  3012  3012 W otbr-agent: 00:00:00.010 [W] Platform------: RadioSpinel before isenabled 0xad7eb288 mState 1
09-01 20:49:14.919  3012  3012 W otbr-agent: 00:00:00.010 [W] SubMac--------: SubMac::Enable 2
09-01 20:49:14.919  3012  3012 W otbr-agent: 00:00:00.010 [W] SubMac--------: SubMac::Enable 3
09-01 20:49:14.919  3012  3012 D otbr-agent: 00:00:00.011 [D] SubMac--------: RadioState: Disabled -> Sleep
09-01 20:49:14.920  3012  3012 W otbr-agent: 00:00:00.011 [W] SubMac--------: SubMac::Enable 4

**Additional context** Add any other context about the problem here.

jwhui, 2023-09-20:
What git commit are you using?

A similar issue was addressed by:
- https://github.com/openthread/openthread/pull/9327

spalas1996, 2023-09-21:
@jwhui The issue mentioned in https://github.com/openthread/openthread/pull/9327 is not the same as this one

jwhui, 2023-09-21:
> @jwhui The issue mentioned in #9327 is not the same as this one

Yes, similar != same :)

spalas1996, 2023-09-21:
@jwhui I have opened up a PR fixing this issue. Please help review the PR. https://github.com/openthread/openthread/pull/9447

jwhui, 2023-09-21:
> @jwhui I have opened up a PR fixing this issue. Please help review the PR. #9447

Can you resolve the Google CLA check failure? https://github.com/openthread/openthread/pull/9447#issuecomment-1728557188

spalas1996, 2023-09-21:
@jwhui Sure, Let me check that. I am trying to find if Amazonlab126 has a CLA already I can use.

jwhui, 2023-09-21:
> @jwhui Sure, Let me check that. I am trying to find if Amazonlab126 has a CLA already I can use.

Maybe @gabekassel @AlanLCollins can help.

gabekassel, 2023-09-21:
I can. @spalas1996 I'll find you.

gabekassel, 2023-09-21:
@spalas1996 @jwhui i filed https://github.com/openthread/openthread/pull/9450 - we can close the other PR

spalas1996, 2023-09-21:
Thanks @gabekassel

closed 2023-09-22 by jwhui