openthread/openthread #9489: Potential incorrect handling of outgoing multicast packets

konradderda, 2023-10-04:
**Describe the bug**
While playing with OpenThread stack integrated in Zephyr I have found a potential bug in the code. I have an idea how this can be handled, however I would like to discuss the details to confirm my findings first.

Briefly - In Zephyr writing data to a socket at the end call OT's API function `otIp6Send()` which calls `Ip6::SendRaw()` under the hood 
https://github.com/openthread/openthread/blob/main/src/core/net/ip6.cpp#L1095

For multicast destination addresses this method always adds MPL-tunneled option to the header for mentioned FF04+ scopes. Right after that `Ip6::HandleDatagram()` decides what to do with the given datagram (receive, passing to host, forwarding to Thread interface). Unfortunately, even if the destination address is subscribed by the node itself it will never receive the datagram:
https://github.com/openthread/openthread/blob/main/src/core/net/ip6.cpp#L1185

1. For the packets from the "host" (in our case the Zephyr network stack) it requires the multicast-loop option to be enabled in otMessageInfo structure and there is no way to set it with the current API.
2. It checks if the destination address is subscribed. It's the second issue as on the entry the function parser the tunneling header which will always have a destination address set to `FF03::FC` (forwarding address). The same problem can be seen few lines above when the stack decides whether to pass the datagram to a host: https://github.com/openthread/openthread/blob/main/src/core/net/ip6.cpp#L1183
3. When the `PassToHost()` method is about to be called (https://github.com/openthread/openthread/blob/main/src/core/net/ip6.cpp#L1241) code does not check if the `aOrigin` is not set to `kFromHostDisallowLoopBack` so it calls the 'received' callback when it should not.

**To Reproduce**

1. Git commit id - observed on d9abe3071c0131a4adb5d7e7451319b735e6d855 but by reviewing the code I can see that behavior should be the same on the current HEAD of the main branch
5. IEEE 802.15.4 hardware - should be reproducible with any Zephyr based OT platform

To reproduce:
1. Subscribe to any multicast address with a scope higher than Realm-Local, e.g. FF05::abcd
2. Open the socket for the OT interface, then send any multicast traffic to subscribed address

**Expected behavior**
1. If the node subscribes to the multicast address with a scope higher than Realm-Local it should receive the datagram even if it's already tunneled.
2. When `kFromHostDisallowLoopBack` is passed as an argument the datagram must not be passed back to the host

**Additional context**
I would like to propose a PR but I would like to discuss the following ideas:
1. Add a way to set multicast-loop option to a message (a separate API function or a change of the current API)
2. While checking the multicast destination address use the original destination address
3. Never call `PassToHost()` method for datagrams with origin set to `kFromHostDisallowLoopBack`

abtink, 2023-10-05:
Thanks @konradderda.

> 1. Add a way to set multicast-loop option to a message (a separate API function or a change of the current API)

I like your proposed idea 1.
- The `mMulticastLoop` flag in `otMessageInfo` is intended to handle this but as you mentioned we cannot use it with `SendRaw()`. 
- We can add a new API to set this flag directly on an `otMessage`, say `otMessageSetMulticastLoopAllowed()`.
- I think this would address this?

BTW, just to let you know, there are some upcoming related changes from this PR https://github.com/openthread/openthread/pull/9437, e.g., it adds APIs to set origin on `otMessage` (`otMessageSetOrigin`) and other flags (`otMessageSetLoopbackToHostAllowed`). This gives more control to users.

konradderda, 2023-10-05:
Hello @abtink,
I believe that in order to handle all mentioned scenarios we need to apply all three proposed changes. I will open a PR soon.

abtink, 2023-10-05:
One thought on this:
- Does it make sense to require the next layer IP stack to receive/process the multicast message (since it should be aware that it is subscribed to this multicast address) and then pass it to Thread stack (to be forwarded on Thread mesh)
   - Basically assuming that `otIp6Send()` will always have `mMulticastLoop` as `false`.
 
- I am wondering what is the behavior if the message destination happens to be unicast and an address of the Thread netif itself.

EskoDijk, 2023-10-05:
For the Linux socket() API there is the option `IPV6_MULTICAST_LOOP` that is by default on and can be changed using `setsockopt`. The default documentation is a bit unclear ("Control whether the socket sees multicast packets that it has send itself" - a client socket obviously can't receive data on multicast groups, it has to be a server socket that's registered to the group(s)? ).

Anyhow for embedded/Thread applications, most use cases don't require looping back own multicast transmissions I guess, but it should be nice to follow the conventions and allow loopback control setting and having it default 'on'. And the bugs should be fixed, ideally.

abtink, 2023-10-05:
> Never call` PassToHost()` method for datagrams with origin set to `kFromHostDisallowLoopBack`

Regarding this, the `PassToHost()` method itself include this check:
```
    VerifyOrExit(aOrigin != kFromHostDisallowLoopBack, error = kErrorNoRoute);
```
So this seems to be already covered? Maybe I am missing something here?

abtink, 2023-10-06:
> For the Linux socket() API there is the option IPV6_MULTICAST_LOOP that is by default on and can be changed using setsockopt. The default documentation is a bit unclear ("Control whether the socket [...]

OT socket-like APIs also provide the `mMulticastLoop` as part of `otMessageInfo` struct which behaves similarly. And I think that it works correctly if used with `otUdpSend`, `otUdpSendDatagram` and `otIcmp6SendEchoRequest()`, ect APIs.

I think the main question here is whether `Ip6::SendRaw()` (which is sort of a lower-level IPv6 API and intended to send an already prepared (raw) IPv6 message to the Thread stack) should provide mechanism to allow caller to choose the `mMulticastLoop` behavior. Current `SendRaw()` API does not let user select this and behaves as if multicast-loop is not allowed.

EskoDijk, 2023-10-06:
@abtink The issue description mentions that `otIp6Send()` is being used by Zephyr. This internally calls `SendRaw()` with parameter `aAllowLoopBackToHost` set to `true`, so the expectation would be surely that the loopback mechanism works for any IPv6 packet that's being sent.  The parameter isn't mentioning "unicast-only loopback".

So I think `SendRaw()` already allows the called to choose loopback yes or no. And how the loopback is realized depending on whether the frame is unicast or multicast, is the responsibility of `SendRaw`.
E.g. a caller of SendRaw() who doesn't want loopback for his multicast packet, can just supply a parameter `false` in the call.
Because the caller knows the packet is multicast, it's logical that the parameter is assumed by the caller to apply to a multicast packet in that case.

abtink, 2023-10-06:
> So I think SendRaw() already allows the called to choose loopback yes or no. 

There is some history behind the `aAllowLoopBackToHost` (which was renamed over the years)

Way back (early years), OT would pass any message that it did not know where to send to host, regardless of whether it was received over the radio or originated from the host itself.  This caused problems with some network stacks, such as some lwip stacks, which would pass the message back to Thread, creating a repeating ping-pong loop. The `SendRaw()` was updated to track messages from host (marked them with boolean flag) and disallow them to be passed back to host and this was the default behavior. Later we needed to change this to address some situation related to bbr. So an boolean was added to control this but the behavior itself was controlled by a config `OPENTHREAD_CONFIG_IP6_ALLOW_LOOP_BACK_HOST_DATAGRAMS` . The `otIp6Send()` and other OT public APIs did not expose this parameter to user, so users could not select this per message.

This is changed very recently from https://github.com/openthread/openthread/pull/9437 which allows users to specify the origin of a message and set this flag on an `otMessage` directly.

----

Regarding multicast behavior, I suggest a different interpretation of the two flags:

- `AllowLoopBackToHost`: Whether or not the OT stack is allowed to pass this message to the host, regardless of the reason (e.g., no route to destination, destination matches this device, etc.). This acts as a general filter for interaction with host.
- `MulticastLoop`: Whether we are allowed to "receive" this multicast message if the device itself is subscribed to the same address, regardless of whether the message originates from the host (from `otIp6Send()`, `otUdpSend()` or other public APIs) or is generated by the Thread stack itself (e.g., a message generated by the Network Diagnostic or MLE modules). Blocking multicast loop "receive" would prevent the message from being passed to the host or processed by any internal OT module listening for such message.

konradderda, 2023-10-06:
Hello @abtink @EskoDijk ,
The main issue is that when I a packet is passed to OT stack with `otIp6Send()` it is not processed by the node itself even if it subscribes to the destination address. If I subscribe for e.g. `ff05::abcd`, and then send a packet it is forwarded to the Thread network but it's not processed because its encapsulated with `ff03::fc` destination address - `Ip6::HandleDatagram()` parses only the first header and will not set `receive = true`. Hence, the packet is not processed by the node itself.

Other comments:
> * Does it make sense to require the next layer IP stack to receive/process the multicast message (since it should be aware that it is subscribed to this multicast address) and then pass it to Thread stack (to be forwarded on Thread mesh)

I think that with a current implementation if I write a multicast message to OT socket with Zephyr I will get responses from all devices within the group except the originator itself. I think the OT stack should at least try to handle the message (I would need to prepare a test case to confirm the suspicion).

> Regarding this, the `PassToHost()` method itself include this check:

Oh, I have missed that (It was my assumption from a code review but I did not check the method itself).

> I think the main question here is whether `Ip6::SendRaw()` (which is sort of a lower-level IPv6 API and intended to send an already prepared (raw) IPv6 message to the Thread stack) should provide [...]

> So I think `SendRaw()` already allows the called to choose loopback yes or no. And how the loopback is realized depending on whether the frame is unicast or multicast, is the responsibility of `SendRaw`.

I can see that we have two "loop" contexts:
1. Metioned by @EskoDijk - `aAllowLoopBackToHost` or `kFromHostDisallowLoopBack` and this controls whether `PassToHost()` method can be called. As @abtink pointed out it works and I am able to set it with a `#define` directive
2. Mentioned by @abtink - `mMulticastLoop` which is a part of `otMessageInfo` it controls if the message from the Host will be processed by the OT stack itself. There is currently no way to set this with the current API.

I think the Zephyr should be able to "ready to be sent" IPv6 packets so it would be nice to be able to control the mMulticastLoop. Moreover, it looks like we should check subscription of a multicast destination address which is encapsulated after the tunneling header instead of checking for the `ff03::fc` address.

abtink, 2023-10-07:
Hi @konradderda 

I did go into the code in more details and did a bunch of experiments.

> If I subscribe for e.g. `ff05::abcd`, and then send a packet it is forwarded to the Thread network but it's not processed because its encapsulated with `ff03::fc` destination address - [...]

> Moreover, it looks like we should check subscription of a multicast destination address which is encapsulated after the tunneling header instead of checking for the ff03::fc address.

I don't think this is an issue. Suppose the message is encapsulated with `ff03::fc` destination address:
- In `HandleDatagram()` we have:
https://github.com/openthread/openthread/blob/50e20c89587b647bd732809a7e931917d448482a/src/core/net/ip6.cpp#L1183-L1187
- All nodes in Thread mesh subscribe to `ff03::fc`, so if `GetMultciastLoop()` is set on `aMessage`, the `if` conditions are met, and `receive` will be set to `true`.
-  A bit later in `HandleDatagram()` we have:
https://github.com/openthread/openthread/blob/50e20c89587b647bd732809a7e931917d448482a/src/core/net/ip6.cpp#L1231-L1237
- Which will remove the encapsulation header from the `aMessage` and jumps back to `start` of the `HandleDatagram()` method to process the `aMessage` again (now without the encapsulation header).

abtink, 2023-10-08:
Submitted PR #9494 which adds new test cases covering pings to multicast destination addresses with and without the "multicast loop" flag, and to realm-local and larger than realm-local scopes (where the `ot::Ip6` module would use a tunnel header for forwarding the message). All tests pass.

So I think for Zephyr use-case, the only thing is to add a mechanism to allow caller to specify "multicast loop" flag when using `otIp6Send()` API which currently does not allow user to specify this flag and assumes  "multicast-loop" flag not set. The `mMulticastFlag` in `otMessageInfo` can be used for all socket-based OT APIs.

konradderda, 2023-10-13:
Hello @abtink, 
Thanks a lot for so valuable input. I see that the PR I have submitted https://github.com/openthread/openthread/pull/9524 (after applying your remarks) alongside disabling `OPENTHREAD_CONFIG_IP6_ALLOW_LOOP_BACK_HOST_DATAGRAMS` should do the trick for the multicast messages in Zephyr.

I am closing this issue as it looks that the missing puzzle piece was very little and can be tracked in the PR itself.

bukepo, 2024-02-02:
@konradderda Is Zephyr expecting a packet delivered to the OT interface being looped back to Zephyr's IP stack? If so, it looks not to be a design that works with different underlying link layer technology. I believe the loopback to host should be implemented in Zephyr's own IP stack, so that all link layer technology would have this feature. Could you help clarify this part?

konradderda, 2024-02-05:
Hello @bukepo, this image represents how sending packets through sockets works: 
[image]

In Zephyr, we can perform OT transmission and reception through socket API. We can list three use cases for multicast packets:
1. Generated by socket and submitted to OT stack (e.g. through mentioned socket)
2. Generated by OT stack (e.g. by `udp` command of OT CLI)
3. Received by OT stack from the network (e.g. End Device sending ping to `ff05:` scope)

I cannot recall it exactly but I think there was an issue with 1st point - we can have a scenario with different software modules listening to traffic from the Thread Network. At the moment of opening this PR I was unable to set multicast loop flag for the packet fed to OT stack through `otIp6Send()`. I agree with you, it would be better if the host would handle this scenario differently but it would probably require much more complicated solution than just setting the flag.

Is there a problem with this approach right now?

bukepo, 2024-02-06:
@konradderda The problem is looping back a packet from host back to host may cause the packet being processed multiple times, and even forwarded to adjacent networks.

I checked Zephyr's [code](https://github.com/zephyrproject-rtos/zephyr), I don't see `otMessageSetLoopbackToHostAllowed()` is called. Does it indicate currently zephyr doesn't rely on OpenThread to loop back packets?

bukepo, 2024-02-21:
@konradderda could you confirm whether loopback to host is really used by Zephyr? If not, probably we can safely disable the behavior looping back packets to host network stack.

konradderda, 2024-03-14:
@bukepo, I am sorry for the delay. When I tested a scenario I have described previously I used a version of OT that did not have `otMessageSetLoopbackToHostAllowed()` implemented.

I agree that such loops should be handled by the kernel itself but it looks that it does not work this way. However, I would like to not block any changes because of potential issues with multicasts in Zephyr so if community would prefer to disable the loopback as whole let's then do this. I will then test the newer version and if I find any issues I will either approach them from the Zephyr's side or create a new ticket here.

closed 2024-03-14 by konradderda