openthread/openthread #9491: MeshForwarder: Dropping (dir queue full)

caipiblack, 2023-10-05:
Hello,

We are using OpenThread with the Matter protocol. We use the Openthread version that comes from GeckoSDK 4.3.1 (SiliconLabs) because we are in a multipan configuration (Zigbee+Thread+BLE) with SiliconLabs chip.

Sometimes (1 time per week?), for unknown reason we loose the communication with all the thread devices. When it appear we have to reboot otbr-agent to get it working again.

When it appear we have this kind of messages:

```
MeshForwarder-: Dropping (dir queue full) IPv6 UDP msg, len:90, chksum:a1db, ecn:no, sec:no, prio:net
```
I see some discussion about this here: https://github.com/openthread/openthread/pull/7908 but I have no access to [SPEC-1041](https://threadgroup.atlassian.net/browse/SPEC-1041).

In order to understand the problem, can someone explain what is this mecanism ? 

We also have a lot of these messages:

```
Mac-----------: Frame rx failed, error:UnknownNeighbor, len:28, seqnum:103, type:Beacon, src:0x0000, dst:None, sec:no, ackreq:no
```

But these messages also appear when everything is working. So we suppose that it is acceptable, but probably not ?

What about this message:

```
Mle-----------: Send Child Update Request to child (fe80:0:0:0:b035:cfd3:dd23:7e74,0xe403)
```

Did someone know what these messages mean ?

Here you will find some logs where otbr-agent is working and then start to fail with `dir queue full`:

[logs_thread.txt](https://github.com/openthread/openthread/files/12818523/logs_thread.txt)

Any help will be highly appreciated.

EskoDijk, 2023-10-05:
It looks like once the 'dropping' begins, every MLE message to be sent is always dropped (Advertisement and Announce). Which is strange because one would expect the queue to automatically empty again - either by dropping too-old frames, or because frames are sent out over Thread. Maybe there's heavy data traffic at this point?

As for the 'beacon' messages, it looks like some device is sending out Beacon Requests to which many devices respond with a Beacon message. This could be from a Zigbee device trying to join? I think it uses Beacon Request for this.

caipiblack, 2023-10-05:
About the `Send Child Update Request to child` messages, I found this: https://openthread.io/guides/build/features/child-supervision?hl=en

In my border router, `OT_CHILD_SUPERVISION=ON`.

But for me it's not clear if I have to set it ON or OFF, they say:

 `Note: This feature is specific to SEDs. It should not be enabled for any other type of end device.`

The borderrouter is not a end device so it's not clear to me if I have to enable it or not.

caipiblack, 2023-10-05:
About the heavy trafic, I don't really know if there is heavy trafic or not, maybe not.

We are not manipulating the zigbee devices but I try to trig the zigbee device and he successfully transmit his temperature

abtink, 2023-10-05:
@caipiblack such pattern are often caused by issues in radio driver/platform, basically radio driver misbehaving and getting stuck (never invoking "tx done" callback for a frame tx). This then causes messages to pile up to in tx queues within OT stack.

From the logs you posted, this seems to be the root cause.

OT `MeshForwarder` logs whenever an IPv6 message is sent (whether successfully or if there was any failure).

Last successful tx we see is here:
```
2023-10-05T09:19:34.438032+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:11.501 [I] MeshForwarder-: Sent IPv6 UDP msg, len:90, chksum:3d68, ecn:no, to:0xf000, sec:yes, prio:low
2023-10-05T09:19:34.438143+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:11.501 [I] MeshForwarder-:     src:[fd64:6325:664d:1:de70:bafa:6f7e:248b]:60685
2023-10-05T09:19:34.438178+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:11.501 [I] MeshForwarder-:     dst:[fd64:6325:664d:1:a483:537:4882:befc]:5540
```
And last failed tx
```
2023-10-05T09:19:45.356714+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:22.419 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:90, chksum:a0cd, ecn:no, to:0xe403, sec:yes, error:NoAck, prio:low
2023-10-05T09:19:45.356757+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:22.420 [N] MeshForwarder-:     src:[fd64:6325:664d:1:de70:bafa:6f7e:248b]:60685
2023-10-05T09:19:45.356865+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:22.420 [N] MeshForwarder-:     dst:[fd64:6325:664d:1:73d2:76a7:45b4:a9c9]:5540
```

And no other tx related logs after this. Which points to radio being stuck.

```
$ cat ~/Downloads/logs_thread.txt | grep "MeshForwarder-" | grep IPv6 | less
...
2023-10-05T09:19:41.929030+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:18.992 [I] MeshForwarder-: Sent IPv6 UDP msg, len:82, chksum:2477, ecn:no, to:0xe403, sec:yes, prio:low
2023-10-05T09:19:41.929255+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:18.992 [I] MeshForwarder-: Prepping indir tx IPv6 UDP msg, len:90, chksum:a0cd, ecn:no, to:0xe403, sec:yes, prio:low
2023-10-05T09:19:42.289080+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:19.352 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:90, chksum:a0cd, ecn:no, to:0xe403, sec:yes, error:NoAck, prio:low
2023-10-05T09:19:45.119721+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:22.182 [I] MeshForwarder-: Prepping indir tx IPv6 UDP msg, len:90, chksum:a0cd, ecn:no, to:0xe403, sec:yes, prio:low
2023-10-05T09:19:45.356714+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:22.419 [N] MeshForwarder-: Failed to send IPv6 UDP msg, len:90, chksum:a0cd, ecn:no, to:0xe403, sec:yes, error:NoAck, prio:low
2023-10-05T09:19:49.855729+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:26.918 [I] MeshForwarder-: Received IPv6 UDP msg, len:83, chksum:b4f0, ecn:no, from:a66fe30638d7b454, sec:no, prio:net, rss:-49.0
2023-10-05T09:19:54.805561+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:31.867 [I] MeshForwarder-: Received IPv6 UDP msg, len:83, chksum:61f1, ecn:no, from:a66fe30638d7b454, sec:no, prio:net, rss:-49.0
2023-10-05T09:19:54.838273+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:31.901 [I] MeshForwarder-: Received IPv6 UDP msg, len:83, chksum:e713, ecn:no, from:76dcbca57bc0b252, sec:no, prio:net, rss:-33.0
2023-10-05T09:20:04.486377+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:41.549 [I] MeshForwarder-: Received IPv6 UDP msg, len:89, chksum:2495, ecn:no, from:0xf000, sec:yes, prio:normal, rss:-61.0
2023-10-05T09:20:05.322110+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:42.384 [I] MeshForwarder-: Received IPv6 UDP msg, len:89, chksum:2495, ecn:no, from:0xf000, sec:yes, prio:normal, rss:-61.0
2023-10-05T09:20:06.173424+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:43.236 [I] MeshForwarder-: Received IPv6 UDP msg, len:89, chksum:2495, ecn:no, from:0xf000, sec:yes, prio:normal, rss:-61.0
2023-10-05T09:20:07.271980+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:44.335 [I] MeshForwarder-: Received IPv6 UDP msg, len:89, chksum:2495, ecn:no, from:0xf000, sec:yes, prio:normal, rss:-61.0
2023-10-05T09:20:08.706858+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:45:45.768 [I] MeshForwarder-: Received IPv6 UDP msg, len:89, chksum:2495, ecn:no, from:0xf000, sec:yes, prio:normal, rss:-62.0
2023-10-05T09:20:27.263572+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:46:04.325 [I] MeshForwarder-: Received IPv6 UDP msg, len:96, chksum:829f, ecn:no, from:7e396fa42c6b9f3a, sec:no, prio:net, rss:-63.0
2023-10-05T09:20:28.259082+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:46:05.322 [I] MeshForwarder-: Received IPv6 UDP msg, len:96, chksum:3f41, ecn:no, from:7e396fa42c6b9f3a, sec:no, prio:net, rss:-63.0
2023-10-05T09:20:29.258592+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:46:06.320 [I] MeshForwarder-: Received IPv6 UDP msg, len:96, chksum:bfcd, ecn:no, from:7e396fa42c6b9f3a, sec:no, prio:net, rss:-63.0
2023-10-05T09:20:31.188733+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:46:08.252 [I] MeshForwarder-: Received IPv6 UDP msg, len:91, chksum:ab1c, ecn:no, from:7e396fa42c6b9f3a, sec:no, prio:net, rss:-63.0
2023-10-05T09:20:32.627599+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:46:09.690 [I] MeshForwarder-: Received IPv6 UDP msg, len:91, chksum:79d9, ecn:no, from:7e396fa42c6b9f3a, sec:no, prio:net, rss:-60.0
[... cut ...]
2023-10-05T09:21:27.837988+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:47:04.900 [I] MeshForwarder-: Prepping indir tx IPv6 UDP msg, len:252, chksum:ab10, ecn:no, to:0xe403, sec:yes, prio:low
2023-10-05T09:21:55.520994+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:47:32.584 [I] MeshForwarder-: Received IPv6 UDP msg, len:91, chksum:144f, ecn:no, from:7e396fa42c6b9f3a, sec:no, prio:net, rss:-62.0
2023-10-05T09:22:00.279354+00:00 hub-mz-maquette-poseidom otbr-agent[691]: 06:47:37.341 [I] MeshForwarder-: Received IPv6 UDP msg, len:112, chksum:fdec, ecn:no, from:7e396fa42c6b9f3a, sec:no, prio:net, rss:-62.0
```

caipiblack, 2023-10-06:
I have new full logs from tonight.

otbr-agent starts at 04:34:22 and It start failing at 07:53:31

[logs_thread_failure.txt](https://github.com/openthread/openthread/files/12828221/logs_thread_failure.txt)

**Our devices (on Zigbee/Thread)**

Matter (thread):

- 1x Aqara P2
- 1x EVE socket
- 1x EVE motion detector
 
Zigbee:

- 1x Aqara Weather Sensor WSDCGQ11LM

I wanted to try with the silab multipan integration in official Openthread repository (so here) (because it's more recent), but it is not yet available: 

https://github.com/openthread/openthread/pull/8914

EskoDijk, 2023-10-06:
@caipiblack You had some questions on how the queue mechanism works also:  it is a max queue size of 100 outgoing frames. Each frame stays at most 1 second in the queue, if not started to send by then, the frame is dropped. That's indeed defined by section 5.12. So normally a queue should automatically empty again. As it looks in your logs, the queue stays full.

@abtink you mention the queue could fill up if the radio driver malfunctions and doesn't send the "tx done" callback. But, that should be only for 1 frame that is currently being sent right? So the other 99 frames should still be dropped after 1 second which should free up the queue again. I assume there's only 1 frame at a time being in state "in progress of being transmitted".

caipiblack, 2023-10-06:
Thanks @EskoDijk for explanations

Just in case if it can help, here is my Yocto recipe to build openthread, inside you can see the CMake options.

As we are using SiliconLabs Multipan stuff (from gecko SDK), we have some "silab" specific defines, as explained in [AN1333](https://www.silabs.com/documents/public/application-notes/an1333-concurrent-protocols-with-802-15-4-rcp.pdf)

```
PN = "openthread-sl"
SUMMARY = "OTBR on i.MX boards"
DESCRIPTION = "OTBR applications"
LICENSE = "BSD-3-Clause"
LIC_FILES_CHKSUM = "file://LICENSE;md5=87109e44b2fda96a8991f27684a7349c"

DEPENDS        += "jsoncpp avahi boost cpc-daemon dbus ncurses"
RDEPENDS_${PN} += "jsoncpp libavahi-client avahi-daemon bash radvd iproute2 dbus libnetfilter-queue ipset iptables"

# Switch the sources depending on the gecko_sdk being used. 
SRC_URI = "git://github.com/SiliconLabs/gecko_sdk.git;protocol=https;branch=gsdk_4.3;rev=2ba59bf969e94a0544e73e501d876722790f7a89"
SRC_URI += "file://vendor.cmake"
SRC_URI += "file://otbr-agent.service"
SRC_URI += "file://otbr-firewall.service"
SRC_URI += "file://otbr-firewall"
SRC_URI += "file://otbr-utils"
SRC_URI += "file://create-thread-network-with-scan"

S = "${WORKDIR}/git/util/third_party/ot-br-posix"

SYSTEMD_AUTO_ENABLE = "enable"
SYSTEMD_SERVICE_${PN} = "otbr-agent.service otbr-web.service otbr-firewall.service"

FILES_${PN} += "${systemd_unitdir}/system/otbr-agent.service"
FILES_${PN} += "${systemd_unitdir}/system/otbr-web.service"
FILES_${PN} += "${systemd_unitdir}/system/otbr-firewall.service"
FILES_${PN} += "${datadir}/otbr-web/*"

TARGET_CFLAGS += " -Wno-error -Wno-unused-but-set-variable"

inherit cmake systemd

EXTRA_OECMAKE = " \
	-DBUILD_TESTING=OFF \
	-DCMAKE_BUILD_TYPE=Release \
	-DCMAKE_EXPORT_COMPILE_COMMANDS=ON \
	-DCMAKE_INSTALL_PREFIX=/usr \
	-DOT_MULTIPAN_RCP=ON \
	-DOT_POSIX_CONFIG_RCP_BUS=VENDOR \
	-DCMAKE_MODULE_PATH=${WORKDIR}/git/protocol/openthread/platform-abstraction/posix \
	-DOT_POSIX_CONFIG_RCP_VENDOR_INTERFACE=${WORKDIR}/git/protocol/openthread/platform-abstraction/posix/cpc_interface.cpp \
	-DOT_CONFIG=${WORKDIR}/git/protocol/openthread/platform-abstraction/posix/openthread-core-silabs-posix-config.h \
	-DOT_BORDER_AGENT=ON \
	-DOT_BORDER_ROUTER=ON \
	-DOT_CHILD_SUPERVISION=ON \
	-DOT_COAP_BLOCK=ON \
	-DOT_COAP_OBSERVE=ON \
	-DOT_COAP=ON \
	-DOT_COAPS=ON \
	-DOT_COMMISSIONER=ON \
	-DOT_COVERAGE=OFF \
	-DOT_DAEMON=ON \
	-DOT_DATASET_UPDATER=ON \
	-DOT_DHCP6_CLIENT=ON \
	-DOT_DHCP6_SERVER=ON \
	-DOT_DIAGNOSTIC=ON \
	-DOT_DNS_CLIENT=ON \
	-DOT_DUA=OFF \
	-DOT_ECDSA=ON \
	-DOT_IP6_FRAGM=ON \
	-DOT_JAM_DETECTION=ON \
	-DOT_JOINER=ON \
	-DOT_LEGACY=ON \
	-DOT_LOG_LEVEL_DYNAMIC=ON \
	-DOT_LOG_OUTPUT=PLATFORM_DEFINED \
	-DOT_MAC_FILTER=ON \
	-DOT_MLR=OFF \
	-DOT_NETDIAG_CLIENT=ON \
	-DOT_NEIGHBOR_DISCOVERY_AGENT=ON \
	-DOT_PING_SENDER=ON \
	-DOT_PLATFORM=posix \
	-DOT_POSIX_MAX_POWER_TABLE=ON \
	-DOT_POSIX_SETTINGS_PATH=\"/etc/openthread\" \
	-DOT_RCP_RESTORATION_MAX_COUNT=10 \
	-DOT_REFERENCE_DEVICE=ON \
	-DOT_SERVICE=ON \
	-DOT_SLAAC=ON \
	-DOT_SNTP_CLIENT=ON \
	-DOT_SRP_CLIENT=ON \
	-DOT_THREAD_VERSION=1.3 \
	-DOTBR_BACKBONE_ROUTER=OFF \
	-DOTBR_BORDER_AGENT=ON \
	-DOTBR_BORDER_ROUTING=ON \
	-DOTBR_COVERAGE=OFF \
	-DOTBR_DBUS=OFF \
	-DOTBR_DNSSD_DISCOVERY_PROXY=ON \
	-DOTBR_MDNS=avahi \
	-DOTBR_MESHCOP_SERVICE_INSTANCE_NAME="OpenThread Border Router" \
	-DOTBR_PRODUCT_NAME="Border Router" \
	-DOTBR_REST=OFF \
	-DOTBR_SRP_ADVERTISING_PROXY=ON \
	-DOTBR_VENDOR_NAME="OpenThread" \
	-DOTBR_WEB=ON \
	-DOT_FIREWALL=ON \
"

do_configure_prepend() {
    rm -f ${WORKDIR}/git/util/third_party/ot-br-posix/third_party/openthread/repo
    ln -s ${WORKDIR}/git/util/third_party/openthread ${WORKDIR}/git/util/third_party/ot-br-posix/third_party/openthread/repo
    
    install -m 0644 ${WORKDIR}/vendor.cmake ${WORKDIR}/git/util/third_party/openthread/src/posix/platform
}

do_install_append() {
    install -d ${D}/${systemd_unitdir}/system
    install -d ${D}/${bindir}
    
    install -m 0644 ${WORKDIR}/otbr-agent.service ${D}/${systemd_unitdir}/system
    install -m 0644 ${WORKDIR}/otbr-firewall.service ${D}/${systemd_unitdir}/system
    install -m 0755 ${WORKDIR}/otbr-firewall ${D}/${bindir}
    install -m 0755 ${WORKDIR}/otbr-utils ${D}/${bindir}
    install -m 0755 ${WORKDIR}/create-thread-network-with-scan ${D}/${bindir}
}
```

jwhui, 2023-10-06:
> @abtink you mention the queue could fill up if the radio driver malfunctions and doesn't send the "tx done" callback. But, that should be only for 1 frame that is currently being sent right? So the [...]

@EskoDijk , the 1 second delay-aware queuing only applies to the direct queue. The indirect queue can still fill up.

@caipiblack , you can inspect [`bufferinfo`](https://github.com/openthread/openthread/tree/main/src/cli#bufferinfo) to see where the message buffers are being held.

abtink, 2023-10-06:
@caipiblack the root cause  seems to be the same in new logs you posted. That is the radio platform seems to be misbehaving and getting stuck after some point (no more sends after some point).
```
Oct  6 07:49:24 otbr-agent[671]: 03:15:01.730 [I] MeshForwarder-: Sent IPv6 UDP msg, len:91, chksum:8345, ecn:no
, to:0xffff, sec:no, prio:net
Oct  6 07:49:24 otbr-agent[671]: 03:15:01.730 [I] MeshForwarder-:     src:[fe80:0:0:0:e81a:908:a060:7343]:19788
Oct  6 07:49:24 otbr-agent[671]: 03:15:01.730 [I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788

If you can enable `OT_LOG_LEVEL_DEBG` on otbr and collect logs, the `Mac` layer logs can indicate this.

This seems to be the root cause, the other aspects are 

---------------

abtink, 2023-10-06:
I could reproduce similar behavior and logs by emulating radio getting stuck (not invoking "txdone" callback after some tx). Check the code/test in this branch if interested: https://github.com/abtink/openthread/commit/eef335e59c5c8b7f144086a1b1539f1148ff4241

@caipiblack the `Dropping (dir queue full) IPv6 UDP msg` is expected when radio gets stuck and nothing is sent out and the messages pile up in the queue. This is not itself an issue. It is mechanism to protect and bound the message queue size.

The root-cause seems to be the radio getting stuck. 

@EskoDijk regarding your question on queue-mgmt., please check this issue https://github.com/openthread/openthread/issues/9493 that I just posted (don't want to discuss it here to avoid confusion/mixing topcis).

caipiblack, 2023-10-12:
Hello,

I did not respond because the problem does not happens during 7 days, but today it appears three times.

@jwhui Here is the result of `ot-ctl bufferinfo` when the problem appear:

```
root@hub-mz-maquette-poseidom:~# ot-ctl bufferinfo
total: 256
free: 130
max-used: 233
6lo send: 121 126 14118
6lo reas: 0 0 0
ip6: 0 0 0
mpl: 0 0 0
mle: 0 0 0
coap: 0 0 0
coap secure: 0 0 0
application coap: 0 0 0
Done

Somes seconds later:

root@hub-mz-maquette-poseidom:~# ot-ctl bufferinfo
total: 256
free: 110
max-used: 233
6lo send: 131 146 16638
6lo reas: 0 0 0
ip6: 0 0 0
mpl: 0 0 0
mle: 0 0 0
coap: 0 0 0
coap secure: 0 0 0
application coap: 0 0 0
Done
```

And here the new logs:

[dropping_queue_light.txt](https://github.com/openthread/openthread/files/12880291/dropping_queue_light.txt)

Note about the logs: I take the logs before executing the `ot-ctl bufferinfo` command, that's why it does not appear in otbr-agent logs.

I think for the issue, the best now is to see with silabs. But currently I loose the ability to create support ticket on the silab website, so I have to fix it first...

jwhui, 2023-10-12:
@caipiblack , the buffer info indicates that messages are getting waiting int he `6lo` queue, which indicates that the radio is not servicing the queue properly or quickly enough.

caipiblack, 2023-12-18:
This is a simple update just to say we are still investigating with silabs.

We loose a lot of time because currently the step is to check if we have the problem using Gecko SDK 4.3.2 (instead of gecko 4.3.1), but we was blocked during a lot of weeks on another problem that appear with gecko 4.3.2:

With Gecko 4.3.2, when you create a firmware for the RCP using Silabs IDE, by default there is an experimental functionality enabled (`SL_RAIL_UTIL_IEEE802154_FAST_CHANNEL_SWITCHING_ENABLED`) and this functionality is possibly broken.

With this functionality we had another kind of crashes in the case where you have a zigbee device and try to create a thread network. Even if we use the same "channel" for zigbee and matter, it was triggering a channel switching that was causing troubles.

At this point, we start the real test: running the device with two thread devices during two weeks to see if this error comes again or not. 

If the error does not comes we can suppose that the problem has been fixed (by chance ?) with Gecko 4.3.2. 
If not we will have to investigate more with silabs.

jwhui, 2024-03-05:
Closing stale issue.