openthread/openthread #9493: Applying queue mgmt rules to messages originating on device

abtink, 2023-10-06:
Background:
Question by @EskoDijk here -> https://github.com/openthread/openthread/issues/9491#issuecomment-1750271966

> @abtink you mention the queue could fill up if the radio driver malfunctions and doesn't send the "tx done" callback. But, that should be only for 1 frame that is currently being sent right? So the other 99 frames should still be dropped after 1 second which should free up the queue again. I assume there's only 1 frame at a time being in state "in progress of being transmitted".

I investigated this further. We seem to have the following check in the code:
https://github.com/openthread/openthread/blob/50e20c89587b647bd732809a7e931917d448482a/src/core/thread/mesh_forwarder.cpp#L270-L283

This is in `UpdateEcnOrDrop()` and in particular the 
```
VerifyOrExit(!Get<ThreadNetif>().HasUnicastAddress(ip6Header.GetSource()));
```` 
will skip appling the check if the message source address is this device itself.

I cannot recall if there was a reason to exclude messages from this device itself. 

Reading the spec, my understanding is that we should apply this to all message (independent of source address).

jwhui, 2023-10-06:
Not applying delay-aware queuing at source is intentional - see the discussion in this [SPEC-1041 comment](https://threadgroup.atlassian.net/browse/SPEC-1041?focusedCommentId=51677).

abtink, 2023-10-06:
Ah! Perfect. Thanks @jwhui 😃 

---

@EskoDijk , so the messages in queue are a variety of MLE msgs (MLE Advertisements, Parent Requests, etc) with source address being this device's link-local address so the queue-management rule is not applied to them.

EskoDijk, 2023-10-10:
@jwhui @abtink Great, thanks for providing the context. I should have remembered this discussion :-) 
The idea was that also, locally, the stack can provide "pushback" on the application(s) generating the packets to generate less traffic in case of buffer congestion. For MLE messages this isn't implemented of course, as the assumption is that "by design" the MLE traffic level is low enough to always fit the radio's outgoing capacity.

Just wondering if this state of "Thread-internal traffic gets dropped" is somehow also detectable by the application layer code? E.g. it could then take action in some form like show something on a display or schedule a reboot.

jwhui, 2023-10-11:
> Just wondering if this state of "Thread-internal traffic gets dropped" is somehow also detectable by the application layer code? E.g. it could then take action in some form like show something on a display or schedule a reboot.

Maybe not directly, but OpenThread does provide the following informative APIs:
- [otMessageGetBufferInfo](https://openthread.io/reference/group/api-message#otmessagegetbufferinfo)
- [otThreadGetIp6Counters](https://openthread.io/reference/group/api-thread-general#otthreadgetip6counters)

abtink, 2023-11-06:
I think we can close this issue (no change is required).

closed 2023-11-06 by abtink