openthread/openthread #9508: OTBR docker makes ot-daemon to fail

Pawlicho, 2023-10-10:
**Description**
Running the OTBR docker kills ot-daemon process. Standalone ot-daemon works properly and allow to communicate with the Thread Device, but combined with docker it immidietely goes down.

**Git commits id to reproduce**
ot-br-posix: 87f01e4257b39f9fe86ff40f496eeee346de485f
openthread: 652e3f20ca7fef2fc3bb9e3e7da4eefc53f25d89
ot-nrf528xx: 95bb3cc0d99792dda33b427d99aa77f0ffb3b2ac

**Platform**
 nRF52833

**Build steps**
RCP is flashed according to https://openthread.io/codelabs/openthread-hardware#2 instruction. Docker is built and run following steps from https://openthread.io/guides/border-router/docker .

**Network topology**
nRF RCP connected to Linux Machine

**Expected behavior** 
Starting fully-functional OTBR application using OTBR Docke.

I attach logs from otbr docker and ot-daemon. I also add configured network interfaces if it helps in any way.
[daemon-log.txt](https://github.com/openthread/openthread/files/12858534/daemon-log.txt)
[otbr-docker-log.txt](https://github.com/openthread/openthread/files/12858535/otbr-docker-log.txt)
[configured-interfaces.txt](https://github.com/openthread/openthread/files/12858536/configured-interfaces.txt)

jwhui, 2023-10-10:
`otbr-agent` and `ot-daemon` are separate executables that communicate with an RCP and manage a standard network interface. Only one can be communicating with a given RCP at a time. The OTBR docker build includes `otbr-agent`, which runs on startup.

Pawlicho, 2023-10-11:
I get it now! Am I able to configure RCP device runnig with docker via terminal, like ot-ctl commands when ot-daemon is active? Or is it only possible to use web GUI?

closed 2023-10-11 by Pawlicho