openthread/openthread #9535: networkdiagnostic run into infinite loop

lancersky2017, 2023-10-16:
**Describe the bug** A clear and concise description of what the bug is.
Hi Experts, I have the following two Thread devices for networking.
ESP32S3+H2: https://github.com/espressif/esp-idf/commit/9a1cc59338aa151bc6333f85c8169d36a478fe2a
```
> state

leader
Done
> child table

| ID  | RLOC16 | Timeout    | Age        | LQ In | C_VN |R|D|N|Ver|CSL|QMsgCnt|Suprvsn| Extended MAC     |
+-----+--------+------------+------------+-------+------+-+-+-+---+---+-------+-------+------------------+

Done
> router table

| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
| 36 | 0x9000 |       63 |         0 |     0 |      0 |   0 | 7a45104dc2ed2059 |    0 |
| 52 | 0xd000 |       63 |         0 |     3 |      3 |  12 | eef803882e99d8bb |    1 |

Done
> version

openthread-esp32/9a1cc59338-5be432ede; esp32s3;  2023-09-14 02:01:34 UTC
Done
```

NRF52840Dongle: https://github.com/openthread/ot-nrf528xx/commit/1bf1946958913a543e5773c7f260440587817f24
```
> child table
| ID  | RLOC16 | Timeout    | Age        | LQ In | C_VN |R|D|N|Ver|CSL|QMsgCnt| Extended MAC     |
+-----+--------+------------+------------+-------+------+-+-+-+---+---+-------+------------------+

Done
> router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     | Link |
+----+--------+----------+-----------+-------+--------+-----+------------------+------+
| 36 | 0x9000 |       63 |         0 |     3 |      3 |  17 | 7a45104dc2ed2059 |    1 |
| 52 | 0xd000 |       63 |         0 |     0 |      0 |   0 | eef803882e99d8bb |    0 |

Done
> version
OPENTHREAD/thread-reference-20230119-20-g3d4f617a0; NRF52840; Oct 16 2023 15:43:59
Done
```

When I execute the `networkdiagnostic get ff03::2 0 1 8` command on NRF52840Dongle, the following information will always be printed. 
```
    - fdeb:ae52:5ad4:517a:0:ff:fe00:9000
    - fdeb:ae52:5ad4:517a:5f80:82b5:9601:6437
    - fe80:0:0:0:7845:104d:c2ed:2059
IP6 Address List:
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc11
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc10
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc38
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc00
    - fdeb:ae52:5ad4:517a:0:ff:fe00:9000
    - fdeb:ae52:5ad4:517a:5f80:82b5:9601:6437
    - fe80:0:0:0:7845:104d:c2ed:2059
IP6 Address List:
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc11
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc10
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc38
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc00
    - fdeb:ae52:5ad4:517a:0:ff:fe00:9000
    - fdeb:ae52:5ad4:517a:5f80:82b5:9601:6437
    - fe80:0:0:0:7845:104d:c2ed:2059
IP6 Address List:
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc11
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc10
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc38
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc00
    - fdeb:ae52:5ad4:517a:0:ff:fe00:9000
    - fdeb:ae52:5ad4:517a:5f80:82b5:9601:6437
    - fe80:0:0:0:7845:104d:c2ed:2059
IP6 Address List:
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc11
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc10
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc38
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc00
    - fdeb:ae52:5ad4:517a:0:ff:fe00:9000
    - fdeb:ae52:5ad4:517a:5f80:82b5:9601:6437
    - fe80:0:0:0:7845:104d:c2ed:2059
IP6 Address List:
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc11
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc10
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc38
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc00
    - fdeb:ae52:5ad4:517a:0:ff:fe00:9000
    - fdeb:ae52:5ad4:517a:5f80:82b5:9601:6437
    - fe80:0:0:0:7845:104d:c2ed:2059
IP6 Address List:
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc11
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc10
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc38
    - fdeb:ae52:5ad4:517a:0:ff:fe00:fc00
    - fdeb:ae52:5ad4:517a:0:ff:fe00:900
```

abtink, 2023-10-16:
@lancersky2017  Inspecting the code at the commit on Nordic  https://github.com/openthread/ot-nrf528xx/commit/1bf1946958913a543e5773c7f260440587817f24 looks all ok (`NetworkDiagnostic::GetNextDiagTlv()` will alway move offset to next TLV):
https://github.com/openthread/openthread/blob/3d4f617a00e63d09bedbe57cb042ebad74c023f0/src/core/thread/network_diagnostic.cpp#L818

I tried reproducing this with same nordic commit (with logs enabled) which all works ok.
```
> networkdiagnostic get ff03::2 0 1 8
00:17:17.382 [I] NetDiag-------: Sent diagnostic get
00:17:17.382 [I] NetDiag-------: Received diagnostic get query
00:17:17.382 [I] NetDiag-------: Sent diagnostic get answer
00:17:17.382 [I] NetDiag-------: Diagnostic get answer received
DIAG_GET.rsp/ans from fdde:ad00:beef:0:0:ff:fe00:6c00: 0008e6ed0e130a780a1d01026c000840fddead00beef0000000000fffe00fc00fddead00beef0000000000fffe006c00fddead00beef0000ac2460c91f5980a3fe80000000000000e4ed0e130a780a1d
Ext Address: 'e6ed0e130a780a1d
Rloc16: 0x6c00
IP6 Address List:
    - fdde:ad00:beef:0:0:ff:fe00:fc00
    - fdde:ad00:beef:0:0:ff:fe00:6c00
    - fdde:ad00:beef:0:ac24:60c9:1f59:80a3
    - fe80:0:0:0:e4ed:e13:a78:a1d
00:17:17.382 [I] NetDiag-------: Sent diagnostic answer acknowledgment
00:17:17.385 [I] MeshForwarder-: Sent IPv6 HopOpts msg, len:73, chksum:0000, ecn:no, to:0xffff, sec:yes, prio:normal
00:17:17.385 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:0:ff:fe00:6c00]
00:17:17.385 [I] MeshForwarder-:     dst:[ff03:0:0:0:0:0:0:2]
00:17:17.385 [I] MeshForwarder-: Received IPv6 UDP msg, len:236, chksum:9585, ecn:no, from:0x6800, sec:yes, prio:net, rss:-20.0
00:17:17.385 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:0:ff:fe00:6800]:61631
00:17:17.385 [I] MeshForwarder-:     dst:[fdde:ad00:beef:0:0:ff:fe00:6c00]:61631
00:17:17.385 [I] NetDiag-------: Diagnostic get answer received
DIAG_GET.rsp/ans from fdde:ad00:beef:0:0:ff:fe00:6800: 0008aa662d46d5fe32760102680008a0fd000000000000000000000000000006fd000000000000000000000000000004fd000000000000000000000000000005fd000000000000000000000000000003fd000000000000000000000000000002fd000000000000000000000000000001fddead00beef0000000000fffe00fc10fddead00beef0000000000fffe006800fddead00beef0000bcc40144bb7951f6fe80000000000000a8662d46d5fe3276
Ext Address: 'aa662d46d5fe3276
Rloc16: 0x6800
IP6 Address List:
    - fd00:0:0:0:0:0:0:6
    - fd00:0:0:0:0:0:0:4
    - fd00:0:0:0:0:0:0:5
    - fd00:0:0:0:0:0:0:3
    - fd00:0:0:0:0:0:0:2
    - fd00:0:0:0:0:0:0:1
    - fdde:ad00:beef:0:0:ff:fe00:fc10
    - fdde:ad00:beef:0:0:ff:fe00:6800
    - fdde:ad00:beef:0:bcc4:144:bb79:51f6
    - fe80:0:0:0:a866:2d46:d5fe:3276
00:17:17.386 [I] NetDiag-------: Sent diagnostic answer acknowledgment
00:17:17.386 [I] MeshForwarder-: Sent IPv6 UDP msg, len:54, chksum:13fa, ecn:no, to:0x6800, sec:yes, prio:normal
00:17:17.386 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:0:ff:fe00:6c00]:61631
00:17:17.386 [I] MeshForwarder-:     dst:[fdde:ad00:beef:0:0:ff:fe00:6800]:61631
00:17:17.401 [I] MeshForwarder-: Sent IPv6 HopOpts msg, len:73, chksum:0000, ecn:no, to:0xffff, sec:yes, prio:normal
00:17:17.401 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:0:ff:fe00:6c00]
00:17:17.401 [I] MeshForwarder-:     dst:[ff03:0:0:0:0:0:0:2]
00:17:17.414 [I] MeshForwarder-: Received IPv6 HopOpts msg, len:73, chksum:0000, ecn:no, from:0x6800, sec:yes, prio:normal, rss:-20.0
00:17:17.414 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:0:ff:fe00:6c00]
00:17:17.414 [I] MeshForwarder-:     dst:[ff03:0:0:0:0:0:0:2]
00:17:17.499 [I] MeshForwarder-: Received IPv6 HopOpts msg, len:73, chksum:0000, ecn:no, from:0x6800, sec:yes, prio:normal, rss:-20.0
00:17:17.499 [I] MeshForwarder-:     src:[fdde:ad00:beef:0:0:ff:fe00:6c00]
00:17:17.499 [I] MeshForwarder-:     dst:[ff03:0:0:0:0:0:0:2]
Done
> 
```

lancersky2017, 2023-10-17:
Hi @abtink 
You need to use two different versions of firmware, one is https://github.com/openthread/ot-nrf528xx/commit/1bf1946958913a543e5773c7f260440587817f24 and the other is https://github.com/openthread/ot-nrf528xx/commit/9ff8f13.
```
> version
OPENTHREAD/thread-reference-20230119-20-g3d4f617a0; NRF52840; Oct 17 2023 09:26:32
Done
```
```
> version
OPENTHREAD/thread-reference-20230706-33-g5be432ede; NRF52840; Oct 17 2023 09:23:33
Done
```
And run the command `networkdiagnostic get ff03::2 0 1 8` on the lower version side. [networkdiagnostic.log](https://github.com/openthread/openthread/files/12923414/networkdiagnostic.log)

lancersky2017, 2023-10-17:
Here is the sniffer data [networkdiagnostic-d8436902fb55d760ea4aa91c0b8353ee.zip](https://github.com/openthread/openthread/files/12923444/networkdiagnostic-d8436902fb55d760ea4aa91c0b8353ee.zip)

lancersky2017, 2023-10-19:
Hi @abtink , It’s related to this commit https://github.com/openthread/openthread/commit/f9cd4b418

abtink, 2023-10-19:
Yes. Thanks for finding it. I forgot about fixing this.

The PR https://github.com/openthread/openthread/pull/8957 did fix this existing issue in the code. 

> This commit fixes `GetNextDiagTlv()` to correctly return error `kErrorNotFound` if the message happens to end with an unknown TLV. We ensure to set the `error` after the end of `while()` loop. With the previous code we could return `kErrorNone` (since we would successfully read and skip over the unknown TLV which would set `error`).

This fix/commit need to be used on (new) devices if we want to use `GetNextDiagTlv()`.

closed 2023-10-20 by lancersky2017