openthread/openthread #9564: [core][crypto] No checks for NULL pointer in aes_ccm.cpp

George-Stefan, 2023-10-26:
There are no checks for NULL pointer in aes_ccm.cpp
https://github.com/openthread/openthread/blob/main/src/core/crypto/aes_ccm.cpp#L195

When an malformed Enh-ACK is received the stack crashes with NULL pointer.

[image]

abtink, 2023-10-26:
Thanks @George-Stefan .

This seems to be an incorrect behavior by the radio platform implementation which seems to pass a non-`null` pointer to an instance of `otRadioFrame *aAckFrame` in `TxDone()` callback  with a `nullptr` as `mPsdu`.

- If no ack is received, the `TxDone` should be called with a `null` for `aAckFrame` instance itself
- The `ProcessEnhAckSecurity()` won't be called if `aAckFrame` is null (checked in `Mac::HandleTransmitDone()`).
- A `otRadioFrame` instance is  always expcted to have a non-null `mPsdu`.

George-Stefan, 2023-10-26:
@abtink, the malformed Enh-ACK has FCS valid but could have the ASH or the HIE corrupted making Payload() to return NULL

abtink, 2023-10-26:
I see.  We should add a check to validate PSDU of an ehc ack frame.

I can add it or do you want to submit a PR?

abtink, 2023-10-26:
Just submitted https://github.com/openthread/openthread/pull/9565

George-Stefan, 2023-10-26:
> Just submitted #9565

Isn't a problem that the Enh-ACK is processed in Mac::RecordFrameTransmitStatus() => ProcessCsl() before ProcessEnhAckSecurity() called from Mac::HandleTransmitDone() ?

An attacker could just disrupt the CSL Tx by sending forged Enh-ACKs with CSL IE

abtink, 2023-10-26:
> Isn't a problem that the Enh-ACK is processed in Mac::RecordFrameTransmitStatus() => ProcessCsl() before ProcessEnhAckSecurity() called from Mac::HandleTransmitDone() ?

From what I recall, the `RecordFrameTransmitStatus()` is mainly intended for updating counters.  I don't think it is calling `ProcessCsl()` (I could not find it in the code?).

George-Stefan, 2023-10-26:
Yes, on latest stack code, I don't see any of the above problems for CSL

closed 2023-10-31 by jwhui