openthread/openthread #9588: MLE/CoAP packet in the middle of a large data packet causes data packet to be dropped

sarveshkumarv3, 2023-11-07:
**Describe the bug** A clear and concise description of what the bug is.

When a MLE packet comes in between fragments of a large data packet, the reassembly of the large data packet fails and it gets dropped. This seems due to reception of new tag causing old tag to be dropped as duplicate. This seems to happen when OPENTHREAD_CONFIG_MULTI_RADIO is enabled 

void MeshForwarder::HandleFragment
...
#if OPENTHREAD_CONFIG_MULTI_RADIO
if (aLinkInfo.mLinkSecurity)
{
...
VerifyOrExit(!neighbor->IsLastRxFragmentTagAfter(tag), error = kErrorDuplicated);
...

https://github.com/openthread/openthread/commit/ffb3fdf9bdc0e1aff3b10a6811eb385afec5c3ee#diff-57fa1438f2cab87c2060b0367576e589d359ce02104fb0ef62fcc466c92381d2R1113

**To Reproduce** Information to reproduce the behavior, including:

Run 1000 byte ping test between 2 nodes and observe packet loss 

1. Git commit id 
- latest OT code  with additional custom prints to debug this issue 

2. IEEE 802.15.4 hardware platform 
 
 - nrf52840 DK board 

3. Build steps 
4. Network topology 

- 2 devices build as FTD and operating as Thread routers 

**Expected behavior** 

Pings should not fail 

**Console/log output** If applicable, add console/log output to help explain your problem.

**Tag 41221 is ICMP packet** 

 01:06:46.427 [N] MeshForwarder-: rx frag frame, error:OK, len:102, src:0x8400, dst:0x1400, tag:41221, offset:0, dglen:1048, sec:yes
 01:06:46.440 [N] MeshForwarder-: rx frag frame, error:OK, len:96, src:0x8400, dst:0x1400, tag:41221, offset:120, dglen:1048, sec:yes
 01:06:46.453 [N] MeshForwarder-: rx frag frame, error:OK, len:96, src:0x8400, dst:0x1400, tag:41221, offset:216, dglen:1048, sec:yes
 01:06:46.465 [N] MeshForwarder-: rx frag frame, error:OK, len:96, src:0x8400, dst:0x1400, tag:41221, offset:312, dglen:1048, sec:yes
 01:06:46.479 [N] MeshForwarder-: rx frag frame, error:OK, len:96, src:0x8400, dst:0x1400, tag:41221, offset:408, dglen:1048, sec:yes
 01:06:46.496 [N] MeshForwarder-: rx frag frame, error:OK, len:96, src:0x8400, dst:0x1400, tag:41221, offset:504, dglen:1048, sec:yes
 01:06:46.518 [N] MeshForwarder-: rx frag frame, error:OK, len:96, src:0x8400, dst:0x1400, tag:41221, offset:600, dglen:1048, sec:yes

**Tag 41222 is MLE advertisement from the same source** 

 01:06:46.536 [N] MeshForwarder-: rx frag frame, error:OK, len:90, src:fe0e29044d75ddfb, dst:0xffff, tag:41222, offset:0, dglen:153, sec:yes
 01:06:46.545 [N] MeshForwarder-: rx frag frame, error:OK, len:25, src:fe0e29044d75ddfb, dst:0xffff, tag:41222, offset:128, dglen:153, sec:yes

**Tag 41223 is CoAP packet from same source** 

 01:06:46.556 [N] MeshForwarder-: rx frag frame, error:OK, len:15, src:0x8400, dst:0x1400, tag:41223, offset:0, dglen:54, sec:yes

**Tag 41221 is ICMP packet continuation which gets dropped** 

 01:06:46.574 [N] MeshForwarder-: Dropping rx frag frame, error:Duplicated, len:96, src:0x8400, dst:0x1400, tag:41221, offset:696, dglen:1048, sec:yes
 01:06:46.636 [N] MeshForwarder-: Dropping rx frag frame, error:Duplicated, len:96, src:0x8400, dst:0x1400, tag:41221, offset:792, dglen:1048, sec:yes
 01:06:46.648 [N] MeshForwarder-: Dropping rx frag frame, error:Duplicated, len:96, src:0x8400, dst:0x1400, tag:41221, offset:888, dglen:1048, sec:yes
 01:06:46.661 [N] MeshForwarder-: Dropping rx frag frame, error:Duplicated, len:64, src:0x8400, dst:0x1400, tag:41221, offset:984, dglen:1048, sec:yes

**Additional context** Add any other context about the problem here.

NA

abtink, 2023-11-08:
Thanks @sarveshkumarv3 for reporting this.

I submitted https://github.com/openthread/openthread/pull/9590 as a possible fix to address this.

jwhui, 2023-11-09:
@sarveshkumarv3 , can you confirm whether #9590 addresses this issue?

sarveshkumarv3, 2023-11-09:
@abtink thanks for the PR. @jwhui sure, will test and get back

sarveshkumarv3, 2023-11-10:
@abtink @jwhui the shared fix looks good, without the fix I hit 4 instances of failures in 100 pings of size (1500 bytes). With the fix, did not hit this issue. Have one code review comment regarding the approach taken in the fix, can you please check?

jwhui, 2023-11-10:
> @abtink @jwhui the shared fix looks good, without the fix I hit 4 instances of failures in 100 pings of size (1500 bytes). With the fix, did not hit this issue. Have one code review comment regarding the approach taken in the fix, can you please check?

@sarveshkumarv3 , great to hear, thanks for validating. I don't see your review comment in the PR, can you double check?

sarveshkumarv3, 2023-11-10:
@jwhui sorry, can you please check again, it did not post the first time, added now.

abtink, 2023-11-10:
Thanks @sarveshkumarv3  for validating the fix and also thanks for finding this issue (does not look an easy one to debug/find) 👍

closed 2023-11-10 by jwhui