openthread/openthread #9597: Segfault related to netlink

gabekassel, 2023-11-10:
**Describe the bug** On a newer commit while testing with NAT64 translator, I regularly saw segfaults. I was able to capture the below stacktrace and see it's somehow related to netlink

**To Reproduce** Information to reproduce the behavior, including:
It's unclear how to reproduce. This seems to happen around OTBR start 

1. Git commit id OTBR e6f497c349060df6375b8d6466381a20f1f9bbf4
2. IEEE 802.15.4 hardware platform Qorvo 7015M
3. Build steps - Yocto build 
4. Network topology - Single BR (this device) with nRF52840 USB dongle as ED

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

```
Reading symbols from .gnu_debugdata for /mnt/hgfs/workspace/gabe-debug/rootfs/usr/sbin/otbr-agent...
(No debugging symbols found in .gnu_debugdata for /mnt/hgfs/workspace/gabe-debug/rootfs/usr/sbin/otbr-agent)
[New LWP 2108]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `otbr-agent -d5 -I wpan0 -B br-lan spinel+hdlc+uart:///usr/sbin/qpg7015m-ot-rcp'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x54c06f04 in processNetlinkEvent(otInstance*) ()
The target architecture is set to "auto" (currently "armv7").
(gdb) bt
#0  0x54c06f04 in processNetlinkEvent(otInstance*) ()
#1  0x54c080ac in platformNetifProcess ()
#2  0x54cb090c in otbr::Ncp::ControllerOpenThread::Process(otSysMainloopContext const&) ()
#3  0x54cb38d4 in otbr::MainloopManager::Process(otSysMainloopContext const&) ()
#4  0x54bdcca4 in otbr::Application::Run() ()
#5  0x54bdda04 in realmain(int, char**) ()
#6  0x54bdc078 in main ()
(gdb)
```

wgtdkp, 2023-11-16:
It's unclear to me where it crashes at, but looking at the code, I think it must be related to how the receive buffer is interpreted. I submitted https://github.com/openthread/openthread/pull/9613 to make the receiver more robust. It may or may not help as I don't have a Yocto device to verify.

wgtdkp, 2023-11-16:
Can you CP this CL and append also the otbr log the next time it crashes?

closed 2023-11-17 by jwhui