openthread/openthread #9605: OTBR agent shows an error "io.openthread.Error.Detached" when the example application enables the Thread feature

MonicaisHer, 2023-11-14:
**Describe the bug** A clear and concise description of what the bug is.
The following "io.openthread.Error.Detached" error appears in OTBR's logs when the [Matter lighting app](https://github.com/project-chip/connectedhomeip/tree/v1.1.0.1/examples/lighting-app/linux) enables the Thread feature:
```
2023-11-14T13:43:55+01:00 otbr-agent[23023]: [INFO]-DBUS----: Handling method org.freedesktop.DBus.Properties.GetAll
2023-11-14T13:43:55+01:00 otbr-agent[23023]: [ERR ]-UTILS---: Replied to org.freedesktop.DBus.Properties.GetAll with result io.openthread.Error.Detached
2023-11-14T14:02:40+01:00 openthread-border-router.otbr-agent[23023]: otbr-agent[23023]: [INFO]-DBUS----: Handling method org.freedesktop.DBus.Properties.GetAll
2023-11-14T14:02:40+01:00 otbr-agent[23023]: [INFO]-DBUS----: Handling method org.freedesktop.DBus.Properties.GetAll
2023-11-14T14:02:40+01:00 otbr-agent[23023]: [ERR ]-UTILS---: Replied to org.freedesktop.DBus.Properties.GetAll with result io.openthread.Error.Detached
2023-11-14T14:02:40+01:00 openthread-border-router.otbr-agent[23023]: otbr-agent[23023]: [ERR ]-UTILS---: Replied to org.freedesktop.DBus.Properties.GetAll with result io.openthread.Error.Detached
```

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: thread-reference-20230119 (22d4f4e)
2. IEEE 802.15.4 hardware platform: amd64 x86 
3. Build steps:
OTBR agent has been set up as required by the Thread feature to communicate with dbus. It's been built from the source (https://github.com/openthread/ot-br-posix.git) with branch (thread-reference-20230119).

The lighting example app is used to enable the Thread feature following these steps:
```
git clone https://github.com/project-chip/connectedhomeip.git --depth=1
cd connectedhomeip
git checkout v1.1.0.1
scripts/checkout_submodules.py --shallow --platform linux
source scripts/activate.sh

cd examples/lighting-app/linux
gn gen out/debug

ninja -C out/debug
./out/debug/chip-lighting-app --thread
```
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.
The lighting app with the Thread feature enabled should be able to be commissioned into the Thread network.

**Console/log output** If applicable, add console/log output to help explain your problem.
Full logs of OTBR: https://pastebin.ubuntu.com/p/btG8tFx3ng/

**Additional context** Add any other context about the problem here.

- Matter SDK Version: 1.2.0.1
- Thread Version: 1.3
- RCP API Version: 6

With the above setup, I used to successfully commission and control the lighting app within the Thread network. I'm not sure where this error is coming from; it might be due to my environment setup. Could you help me with this? Thank you!

jwhui, 2023-11-20:
Some of the properties (e.g. `OTBR_DBUS_PROPERTY_RLOC16`) will return error when the device is detached:

https://github.com/openthread/openthread/blob/6f3b4317d61cb46bda4a4c1c398ff3369c34a778/src/core/thread/mle.cpp#L1014

Looking at the git history, it has been this way for quite some time.

One thing to check is that it may take longer than before for the device to form its own partition, due to changes in the Thread Specification. You may try adding some delay between when Thread is started and when `GetAll` is called.

jwhui, 2024-01-12:
Closing stale issue.

closed 2024-01-12 by jwhui