openthread/openthread #9629: -Wundef - no default for OPENTHREAD_ENABLE_VENDOR_EXTENSION

tima-q, 2023-11-21:
**Describe the bug** 
When using the -Wundef flag (as error) the following flag has no default:
'OPENTHREAD_ENABLE_VENDOR_EXTENSION`

**To Reproduce** Information to reproduce the behavior, including:

1. Was added a very long time ago
2. IEEE 802.15.4 hardware platform - any
3. Build steps - normal build with no `OPENTHREAD_ENABLE_VENDOR_EXTENSION` in configuration header
4. Network topology - NA

**Expected behavior**
Build does not throw 'undefined' warning

**Additional context**
There is no default OPENTHREAD_ENABLE_VENDOR_EXTENSION defined in code. The define is only generated in CMake options - also no fallback there to 0

thcu-gp, 2023-12-12:
Is it an option to enable this -Wundef by default with _warnings as errors_ so these get fixed immediately when new flags are added?

jwhui, 2025-05-21:
Resolved by #11418

closed 2025-05-21 by jwhui