openthread/openthread #9636: Using deprecated address after prefix removal from netdata

abtink, 2023-11-21:
This is related to  [SPEC-1222](https://threadgroup.atlassian.net/browse/SPEC-1222) and the following PRs:
- https://github.com/openthread/openthread/pull/9585
- https://github.com/openthread/openthread/pull/9599

Background:
Investigation failure of `test_zero_len_external_route.py` with https://github.com/openthread/openthread/pull/9585 (see https://github.com/openthread/openthread/pull/9585#discussion_r1400279570) noticed the following situation:

- Suppose we have `::/0` or `fc00::/7` off-mesh route added in netdata, indicating that we have a BR routing to AIL.
- Suppose we have an OMR prefix A in netdata
- The BR that added OMR prefix A removes it from netdata (being turned off or reset, or ...)
- From SPEC-1222, we would like the SLAAC address derived from the removed OMR prefix A to be marked as deprecated rather than immediately removed. 
- This allows devices on the AIL to maintain communication with devices on the mesh using the address based on prefix A for a temporary period before a new OMR prefix is added and SRP services are registered and advertised on AIL.
-  With the changes introduced in the above PRs, we allow messages sent from devices on the AIL to be delivered to Thread devices using the old prefix A address.
- However, if the Thread device attempts to send a response using its deprecated prefix A address as the source address, the message will be dropped and not routed to the BRs publishing `::/0` or `fc00::/7` off-mesh routes.
- The reason for this is how the `RouteLookup()` works:
https://github.com/openthread/openthread/blob/dc6bf363ddf692ff4e2661ee1a5e46d5373a5c35/src/core/thread/network_data_leader.cpp#L259-L284
   - `RouteLookup()` will go through netdata prefix entries matching the source address first.
   -  If a match is found, (off-mesh) external routes associated with the matching prefix's Domain ID are checked.
- In this scenario, the address based on prefix A does not match any prefix in netdata since prefix A has been removed. Even if a new OMR prefix B is added, it won't be matched, resulting in external routes being disregarded.

abtink, 2023-12-07:
> Some ideas on how we may address the situation from the last comment (related to Domain ID use):
> 
> 1. We can relax the check for domain ID in case the  source address is not preferred (but the question become how to select if there happens to be multiple Domain IDs each with their own set of off-mesh routes published in netdata)?
> 
> 2. We can check netdata entries and only if all entries belong to the same  domain ID (which is the common/main case today), then allow the off-mesh route prefixes to be applied/used to determine next hop for the non-preferred source address (that does not match any prefixes)
> 
> 3. We can add new mechanism to remember the domain ID the deprecating prefix/address belonged to and use the routes in netdata associated with the same ID (this seems like the best option to me).
> 
> 4. Change the model and have the deprecating prefix still published in netdata (with its original domain ID). The downside is prefixes may end up piling up in the netdata (I am not personally in in favor of this approach).

From [SPEC-1222](https://threadgroup.atlassian.net/browse/SPEC-1222) disucssion, we are leaning toward option 3.

jwhui, 2024-03-05:
Resolved by #9585

closed 2024-03-05 by jwhui