openthread/openthread #9737: DNS client: default query/browse server should be the local Thread Service Registry

EskoDijk, 2023-12-21:
**Describe the bug** A clear and concise description of what the bug is.
When doing a DNS query or browse, by default the public Google DNS server will be selected.
`#define OPENTHREAD_CONFIG_DNS_CLIENT_DEFAULT_SERVER_IP6_ADDRESS "2001:4860:4860::8888"`

The Thread spec (see Ch14) defines that by default, unicast DNS queries go to the selected Thread Service Registry. Which is typically a BR providing the winning Unicast / Anycast Dataset.  That BR needs to either answer the query itself (in case domain is `default.service.arpa`), or relay the query to its upstream DNS server. What's upstream is determined by the AIL network configuration (like ND RDNSS Option, or fallback to IPv4 DHCP).

This way, a client can use DNS also for local/ULA deployments, and can use DNS also to look up local services or local SRP-registered hostnames.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: bf9944887df4efb329b16e44004298e212b246f8
2. IEEE 802.15.4 hardware platform: any / simulation was used
3. Build steps: nothing special
4. Network topology: one router (DNS client) and one BR publishing Unicast Dataset

BR also publishes a ULA OMR prefix automatically.

Commands at DNS client:
`dns resolve anynamehere.com` or
`dns browse _thread._udp.default.service.arpa`

**Expected behavior** A clear and concise description of what you expected to happen.
Unicast DNS request is sent to BR's ML-EID IPv6 destination address, which is encoded in the Unicast Dataset.

Using Public Google address may be used as a final fallback option perhaps, but not useful if client has only a ULA address.

**Console/log output** If applicable, add console/log output to help explain your problem.
can be added if needed

**Additional context** Add any other context about the problem here.
It would be error-prone if users of the DNS client library would have to determine for themselves the right target address of the DNS server, based on present datasets. Rather, the Thread stack should determine the default here. Making the right selection is also needed to pass certification.

abtink, 2023-12-21:
Thanks @EskoDijk . This is already supported:
- Please check this feature `OPENTHREAD_CONFIG_DNS_CLIENT_DEFAULT_SERVER_ADDRESS_AUTO_SET_ENABLE`:
https://github.com/openthread/openthread/blob/1e82c4e9620e6ab708e94263641674052af9ec30/src/core/config/dns_client.h#L93-L108
- And the related API `otSrpClientIsAutoStartModeEnabled()` to enable this mode.

abtink, 2023-12-21:
Also if a project integrator wants "auto start" mode to be enabled by default can use:
`OPENTHREAD_CONFIG_SRP_CLIENT_AUTO_START_DEFAULT_MODE`
https://github.com/openthread/openthread/blob/1e82c4e9620e6ab708e94263641674052af9ec30/src/core/config/srp_client.h#L73-L83

jwhui, 2023-12-21:
> Also if a project integrator wants "auto start" mode to be enabled by default within OT stack (not wanting to call the API to enable/disable it) they can use `OPENTHREAD_CONFIG_SRP_CLIENT_AUTO_START_DEFAULT_MODE`

Given that the Thread Specification requires sending to BR by default, should we change the default to 1?

abtink, 2023-12-21:
We can change it to `1`. But I expect there may be some test failures (and require changes).

EskoDijk, 2023-12-22:
@abtink Thanks, that works! I didn't realize the DNS client was using the SRP client's server-address (I was looking in CLI docs, not in the source code docs....) But it's good to have both things change together because the DNS/SRP datasets communicate both the servers for DNS and SRP. Having the ability to temporarily change to another resolver (e.g. public one) and back again using `dns config ::0 0` also is good to have for application/vendor-specific reasons. 

Agree with Jonathan that this mode should be the default.  In general, if features are cert-tested already then the default setting won't matter so much because a DUT will fail the test if the setting is not right. But if features are not yet cert-tested (and it can take a while before tests are included in cert), then there's a risk that vendors will leave the default settings and the cert-testing won't catch this mistake.

abtink, 2023-12-22:
I submitted https://github.com/openthread/openthread/pull/9738 to change the config,.
As I expected a bunch of tests did fail with it. I fixed some obvious ones already but there are more to chase.

abtink, 2023-12-26:
All tests are passing now in the new PR (enabling auto-start mode by default).

EskoDijk, 2024-01-18:
Verified that the PR resolves the issue; closing. Thanks @abtink !