openthread/openthread #9749: The SRP client registers a new host name with the same key pair, and the SRP server responds the Name exists.

jinpeng1989, 2024-01-03:
**Describe the bug** A clear and concise description of what the bug is.
We have a product for wireless room sensors (WRS) that acts as a sleep end device in the Thread network. For ease of description, WRS is used instead. For WRS 1.1.1, the host name in the SRP registration information is <SN>.knx. For WRS 1.2.1, the host name is changed to knx-<SN>. When WRS upgrades from version 1.1.1 to version 1.2.1 over the air, WRS as an SRP client sends a DNS update, and the SRP server responds the Name exists. The public/private key pair used by WRS does not change after it is generated for the first time and saved to the flash. In other words, the same key pair is used before and after the WRS firmware upgrade. My understanding of FCFS is that the keys are the same, WRS is the ownership of these names and has the permission to modify these names. I don't understand why SRP Server responds the response code of name exist? Can you help analyze and explain? 

[image]

[image]

[image]

NetWorkKey: fd62cc41dffd04e4a663f9c1402cdeed
[wireshark_name_exists.zip](https://github.com/openthread/openthread/files/13817189/wireshark_name_exists.zip)

[image]
[name_exists.log](https://github.com/openthread/openthread/files/13817369/name_exists.log)

wgtdkp, 2024-01-03:
The error is caused by an existing service name on the infra link:

```
Tue Jan  2 22:42:54 2024 user.info otbr-agent[2724]: 1d.00:58:47.122 [I] SrpServer-----: SRP update handler is notified (updatedId = 1024092267)
Tue Jan  2 22:42:54 2024 user.info otbr-agent[2724]: [INFO]-ADPROXY-: Advertise SRP service updates: host=knx-99fd11f18a00.default.service.arpa.
Tue Jan  2 22:42:54 2024 user.debug otbr-agent[2724]: [DEBG]-ADPROXY-: Publish SRP service '99fd11f18a00._knx._udp.default.service.arpa.'
Tue Jan  2 22:42:54 2024 user.info otbr-agent[2724]: [INFO]-MDNS----: Registering new service 99fd11f18a00._knx._udp,_99fd11f18a00.local, serviceRef = 0
Tue Jan  2 22:42:54 2024 daemon.err mdnsd[1940]: Default: register_service_instance 99fd11f18a00._knx._udp.local. error -65548
Tue Jan  2 22:42:54 2024 user.err otbr-agent[2724]: [ERR ]-MDNS----: Failed to publish service 99fd11f18a00._knx._udp for mdnssd error: Name Conflict!
```

The log shows it have name conflict with advertising the `99fd11f18a00._knx._udp.local` service on the Wi-Fi or Ethernet link via mDNS.

What's the topology of your setup? For example, how many border routers in the network?

It will be helpful if you can capture the mDNS traffic on the infra (Wi-Fi or Ethernet) link of the Border Router.

jinpeng1989, 2024-01-04:
After 86400 seconds, the new SRP information can be successfully registered. The lease time is 86400. therefore, it seems that the SRP registration of WRS 1.1.1 takes up the service name, causing the SRP registration of WRS 1.2.1 to fail.
[image]

jinpeng1989, 2024-01-04:
There is only one TBR, one WRS, one PC, and one router in the network.
[image]

jinpeng1989, 2024-01-04:
To rule out the effect of the infra (Wi-Fi or Ethernet) link, I tested it with topology 2. Topology 2 has only one TBR, one WRS, and one PC. The PC and TBR interact through serial port connections. The test results in network topology 2 are the same as those in network Topology 1, with dnssd still returning 65548. 
[image]

```
BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.7, r11306-c4a6851c72
 -----------------------------------------------------
root@KNBA5F000026:~# 
root@KNBA5F000026:~# 
root@KNBA5F000026:~# 
root@KNBA5F000026:~# ot-ctl srp server service
Done
root@KNBA5F000026:~# ot-ctl child table
| ID  | RLOC16 | Timeout    | Age        | LQ In | C_VN |R|D|N|Ver|CSL|QMsgCnt|Suprvsn| Extended MAC     |
+-----+--------+------------+------------+-------+------+-+-+-+---+---+-------+-------+------------------+
|   6 | 0xac06 |       3300 |          4 |     3 |    9 |1|0|0|  2| 0 |     0 |   129 | fa0f5be3e3574e94 |

Done
root@KNBA5F000026:~# ot-ctl srp server service
99FD11F18A00._knx._udp.default.service.arpa.
    deleted: false
    subtypes: _99FD11F18A00
    port: 5683
    priority: 0
    weight: 0
    ttl: 3600
    lease: 86400
    key-lease: 604800
    TXT: [dt=515041323839322f5749, sw=312e312e31, eui=30303830653132363031363736623862, rl=307861633036, pm=30]
    host: 99FD11F18A00.knx.default.service.arpa.
    addresses: [fd11:22:0:0:4b1e:91d6:c3f2:aa56]
Done
root@KNBA5F000026:~# date
Thu Jan  4 02:22:21 UTC 2024
root@KNBA5F000026:~# ot-ctl srp server service
99FD11F18A00._knx._udp.default.service.arpa.
    deleted: false
    subtypes: _99FD11F18A00
    port: 5683
    priority: 0
    weight: 0
    ttl: 3600
    lease: 86400
    key-lease: 604800
    TXT: [dt=515041323839322f5749, sw=312e312e31, eui=30303830653132363031363736623862, rl=307861633036, pm=30]
    host: 99FD11F18A00.knx.default.service.arpa.
    addresses: [fd11:22:0:0:4b1e:91d6:c3f2:aa56]
Done
root@KNBA5F000026:~# tail /tmp/log_file 
Thu Jan  4 02:41:21 2024 user.notice otbr-agent[2044]: 00:20:21.017 [N] Mle-----------: Child 0xac06 mode change 0x0d -> 0x05 [rx-on:no ftd:no full-net:yes]
Thu Jan  4 02:41:29 2024 daemon.info otbr-mgmt: (ot_ubus.c:318) neighbor_number = 0
Thu Jan  4 02:41:35 2024 daemon.info otbr-mgmt: (ot_ubus.c:438) get event id = 3
Thu Jan  4 02:41:35 2024 user.notice otbr-agent[2044]: 00:20:35.442 [N] Mle-----------: Child 0xac06 mode change 0x05 -> 0x0d [rx-on:yes ftd:no full-net:yes]
Thu Jan  4 02:41:35 2024 user.err otbr-agent[2044]: [ERR ]-MDNS----: Failed to publish service 99fd11f18a00._knx._udp for mdnssd error: Name Conflict!
Thu Jan  4 02:41:35 2024 daemon.err mdnsd[1561]: Default: register_service_instance 99fd11f18a00._knx._udp.local. error -65548
Thu Jan  4 02:41:35 2024 user.warn otbr-agent[2044]: [WARN]-ADPROXY-: Handle publish SRP service '99fd11f18a00._knx._udp.default.service.arpa.': Duplicated
Thu Jan  4 02:41:35 2024 user.warn otbr-agent[2044]: 00:20:35.587 [W] SrpServer-----: Send fail response: 6
Thu Jan  4 02:41:35 2024 daemon.info otbr-mgmt: (ot_ubus.c:438) get event id = 3
Thu Jan  4 02:41:35 2024 user.notice otbr-agent[2044]: 00:20:35.647 [N] Mle-----------: Child 0xac06 mode change 0x0d -> 0x05 [rx-on:no ftd:no full-net:yes]
root@KNBA5F000026:~# 
```

martinzi, 2024-01-04:
The existing `SRP client` implementation prevents changes of `srp client host name` while already registered, see ["can be set when the host is removed or not yet registered with server"](https://github.com/openthread/openthread/blob/ee745130c7035c5ea3268b5869cca90a5e434eee/src/cli/README_SRP_CLIENT.md#host-name). Therefore, your case may not be supported by `SRP server` today.

jinpeng1989, 2024-01-05:
I tested dnssd with dns-sd and found that dnssd allows two hosts with different names to register the same service name. The dnssd assigns and appends numbers to services with the same name. So I don't understand why publishing to dnssd by SRP Server returns Name exist?
[image]

jinpeng1989, 2024-01-05:
I am confused that my test of mdnsd using the dns-sd cli results in support for services with the same name, but otbr-agent used the dns_sd library to publish the service and mdnsd returned 65548.

abtink, 2024-01-05:
`dns-sd` did not actually register two services with the same name. It auto-renamed the second service due to detecting the conflict. Note that it is named `"56FD10F18ADB (2)"`. 

A Thread Border Router acting as an advertising proxy cannot rename host and service name. These names are owned and selected by the Thread devices themselves during SRP registration.

OpenThread (OT) SRP stack does not provide auto-renaming. This design decision prioritizes predictability and avoids potential unexpected behaviors. SRP API user (next layer protocol) is responsible to select names . The SRP API requires `const` strings for service and host names, emphasizing their immutability. OT SRP code offers conflict detection through callbacks to notify users of conflicts, allowing them to implement custom renaming strategies if desired.

jwhui, 2024-03-12:
Closing stale issue.

closed 2024-03-12 by jwhui