openthread/openthread #9787: Child ID request from recently restored SED dropped due to Security error

sarveshkumarv3, 2024-01-21:
**Describe the bug** A clear and concise description of what the bug is.

After the Parent reboots and restores the child from the settings file, Child Update request is sent to the SED child when network data changes. If such a child is in the middle of attaching back (Parent is waiting for Child ID request), the Child ID request from the child fails security. This is because Child Update Request carries new Challenge TLV which is different from the Challenge TLV sent in Parent Response. 

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

Observed in the field logs, parent runs latest OT commit.

**Expected behavior** A clear and concise description of what you expected to happen.

Child ID Request from restored SED child should not be rejected due to security 

**Console/log output** If applicable, add console/log output to help explain your problem.

//Parent Becomes leader 

00:00:27.483 [I] Mle-----------: Send Parent Request to routers (ff02:0:0:0:0:0:0:2)
00:00:28.238 [I] Mle-----------: Send Parent Request to routers (ff02:0:0:0:0:0:0:2)
00:00:28.993 [I] Mle-----------: Send Parent Request to routers and REEDs (ff02:0:0:0:0:0:0:2)
00:00:30.247 [I] Mle-----------: Send Parent Request to routers and REEDs (ff02:0:0:0:0:0:0:2)
00:00:31.502 [I] Mle-----------: Send Parent Request to routers and REEDs (ff02:0:0:0:0:0:0:2)
00:00:32.757 [I] Mle-----------: Send Parent Request to routers and REEDs (ff02:0:0:0:0:0:0:2)
00:00:34.015 [N] Mle-----------: Role detached -> leader

//Received Parent request from the restored SED child 

00:00:34.121 [I] Mle-----------: Receive Parent Request (fe80:0:0:0:cf9:21b:e8d:916b)
00:00:34.124 [I] Mle-----------: Delay Parent Response (fe80:0:0:0:cf9:21b:e8d:916b)

// Network data changes causing a child update request 
00:00:34.179 [I] Mle-----------: Send Data Response (ff02:0:0:0:0:0:0:1)
00:00:34.474 [I] Mle-----------: Send Child Update Request to child (fe80:0:0:0:cf9:21b:e8d:916b,0xa001)

// Child ID request rejected due to Security 

00:00:34.878 [I] Mle-----------: Receive Child ID Request (fe80:0:0:0:cf9:21b:e8d:916b)
00:00:34.878 [W] Mle-----------: Failed to process Child ID Request: Security

**Additional context** Add any other context about the problem here.

NA

abtink, 2024-01-23:
Thanks @sarveshkumarv3.

This is an interesting situation. I think/expect this would be a transient issue and the child will be able to attach on next attempt?

We cannot assume receiving a "Parent Request" from an existing child that the child is detached and is trying to attach back. It may be doing "Parent search" while attached.

sarveshkumarv3, 2024-01-23:
Thanks @abtink.  In scenarios with 1 parent and 1 SED and parent goes through a SW update, the SED performs exponential backoff if parent request fails. If it happens such that the above behavior happens on the nth attempt, n+1 attempt could be 20mins later? 

One approach could be the parent can choose not to restore the child entry from settings file into the child table, if it comes back up after the child timeout duration, but I think it will fall into the same discussion as in https://github.com/openthread/openthread/issues/9700  (where maintaining this time would be platform dependent)? 

Also I am not sure why Child Update got sent to this Child in this scenario?  This Child should be in state kStateRestored and not  kStateValid ? In this scenario, below check should have failed and not sent the Child Update Request right? 

https://github.com/openthread/openthread/blob/7a2983b01c81009d4be26b3db8a6c7d2257aaa6e/src/core/thread/mle_router.cpp#L2629

abtink, 2024-01-26:
> Also I am not sure why Child Update got sent to this Child in this scenario? This Child should be in state kStateRestored and not kStateValid ? In this scenario, below check should have failed and not sent the Child Update Request right?

I agree with you. The `SynchronizeChildNetworkData()` has `Get<ChildTable>().Iterate(Child::kInStateValid))`, so should not be done for children in `kStateRestored`.

Are we sure the NetData change triggers this "Child Update Request" from parent?

sarveshkumarv3, 2024-01-26:
Yes, I

> I agree with you. The `SynchronizeChildNetworkData()` has `Get<ChildTable>().Iterate(Child::kInStateValid))`, so should not be done for children in `kStateRestored`.
>
> Are we sure the NetData change [...]

Yes, I can confirm from the logs, that network data change triggers Child Update Request . The sequence is as below 

00:00:34.209 [I] Notifier------: StateChanged (0x200) [NetData]
00:00:34.209 [I] Mle-----------: Send Data Response (ff02:0:0:0:0:0:0:1)
00:00:34.311 [I] Mle-----------: Receive Parent Request (fe80:0:0:0:cf9:21b:e8d:916b)
00:00:34.314 [I] Mle-----------: Delay Parent Response (fe80:0:0:0:cf9:21b:e8d:916b)
00:00:34.960 [I] Mle-----------: Send Child Update Request to child (fe80:0:0:0:cf9:21b:e8d:916b,0xa001)
00:00:35.567 [I] Mle-----------: Receive Child ID Request (fe80:0:0:0:cf9:21b:e8d:916b)
00:00:35.567 [W] Mle-----------: Failed to process Child ID Request: Security

I think there are 2 enums for kStateValid one is under [StateFilter](https://github.com/openthread/openthread/blob/dacdc7d36f0ad5b8b04b69c1e06efe22cd1993b8/src/core/thread/neighbor.hpp#L104) and the next is under [State](https://github.com/openthread/openthread/blob/dacdc7d36f0ad5b8b04b69c1e06efe22cd1993b8/src/core/thread/neighbor.hpp#L86)

I think 

> Get<ChildTable>().Iterate(Child::kInStateValid))

 would check StateFilter, and hence it does not have kStateRestored? Is that intended? It should be checking State and not StateFilter enum right?

abtink, 2024-01-26:
> I think there are 2 enums for kStateValid one is under [StateFilter](https://github.com/openthread/openthread/blob/dacdc7d36f0ad5b8b04b69c1e06efe22cd1993b8/src/core/thread/neighbor.hpp#L104) and the [...]

Here's a breakdown of the two enums and how they are used:

- `State`: This enum directly represents the possible states of a `Neighbor` (or `Child/Router`). A neighbor can only exist in a single state at a time.

- `StateFilter`: This enum acts as a filtering mechanism when searching or iterating through the child/neighbor table.  It works by specifying one or more acceptable `State` values. Some examples:
  - `kInValidState`: Accepts neighbors with the state `kStateValid`.
  - `kInStateValidOrRestoring`: Accepts neighbors that in states `kStateRestored`, `kStateChildUpdateRequest`, or `kStateValid`.
   - The names reflect their function (e.g., "Restoring," "Attaching"), enhancing code readability and simplifying maintenance if/when new states are introduced.
  
The line `for (Child &child : Get<ChildTable>().Iterate(Child::kInStateValid))` iterates over the child table, returning only `Child` entries whose current state is `kStateValid`.

sarveshkumarv3, 2024-01-28:
thanks @abtink The reason for security failure here here is that when we send a Child Update Request (with a Challenge TLV that is different from Parent Response) the Child ID request sends the Response TLV (based on Challenge TLV from Child Update Request instead of Parent Response) right? 

Had the child responded with a Child ID request based on the Challenge TLV from Parent Response this should not have occurred.

Handling of multiple parallel Challenge TLV's in order received would avoid this security failure right?

abtink, 2024-01-28:
I investigated the code related to where and how "Child Update Request" may be sent from parent to a child 

- If child is being restored, parent will send one "Child Update Request" to it right after parent becomes leader or router.
- Network Data changes will not trigger  a "Child Update Request" to a child that is being restored. It will be sent to a child  "valid" state  (already attached)
- The "Child Update Request" from parent to child will only include a "challenge" TLV if the child is not in "valid" state.

With this:
- I don't think the "Child Update Request" from parent to child was triggered by "Network Data" update.
- My theory is that the "Child Update Request" is the one to recover the child after role change.
- In this case, the child happened to send its "Parent Request" right in the small window of time where `Role detached -> leader` happened and before the "Child Update Request" is sent (triggered by the role change).
- So I think this situation is an edge-case.

Here are my thoughts/ideas on this:
- Since this seems to be an edge-case, we can let it be as is.
- A change is to save multiple "Challenges" per message type in the `Child` entry is not a good or desirable solution as it increases the `Child` entry size impacting overall Child Tale memory requirement to address an edge-case. 
- One idea (without adding any memory overhead):
    - We generate and save "challenge" when the `Child` entry is first initialized in `kRestostredState` 
    - We do not re-generate it when "Child Update Request" is being sent to restore the child.
   - This way the "Child Update Request"  to restore the child, 
        - will either use the "challenge" from initialization, 
        - or the one when a "Parent Response" was send (if this edge-case happens).

sarveshkumarv3, 2024-01-29:
thanks @abtink 

> * Since this seems to be an edge-case, we can let it be as is.

if Child Update Request is due to restore, this should occur more frequently than edge-case scenario right? Also the concern with letting it as is would be in scenarios with just 1 parent and 1 SED, the SED can potentially attach after very long time (upto 20mins) if the previous attempt fails? Considering the impact, do you think it is better to fix it? 

>     * will either use the "challenge" from initialization,
>     * or the one when a "Parent Response" was send (if this edge-case happens).

I think what you propose would be a good solution, but is there any concern from Security handling perspective if two different MLE messages (Parent Response and Child Update Request use the same Challenge TLV)?

abtink, 2024-01-29:
> if Child Update Request is due to restore, this should occur more frequently than edge-case scenario right?

For this to happen, a bunch of things should align and happen in certain order:
- Parent is reset and try to recover its former child.
- Child should also be detached and try to attach again.
   - Otherwise if child is not detached, it would respond to parent's "Child Update Request" and parent would be successful in restoring it as its child.
- And Then the child should happen to send its "Parent Request" such that it is received right after the role change to leader/router and within the time-tick window (average of ~0.5 sec) before the node prepares and sends the "Child Update Request" 

---

> I think what you propose would be a good solution, but is there any concern from Security handling perspective if two different MLE messages (Parent Response and Child Update Request use the same Challenge TLV)?

I don't know but I don't expect so.

---

> the SED can potentially attach after very long time (upto 20mins) if the previous attempt fails (especially in scenarios where parent would perform a software update and come back up after long time)? 

The 20 min is parmater the SED vendor should have considered when desingng their product and its battery capacity and expected lifetume. Unfortunately I think many vendors did mess this up (did not change the config values and used the default in OT which is intended as worse-case).

sarveshkumarv3, 2024-02-14:
@abtink can you please check the [PR](https://github.com/openthread/openthread/pull/9850) would address this issue? This is as per your proposed approach, to not generate a new challenge for Child Update Request if it is to child that is restored.

jwhui, 2024-03-05:
Resolved by #9850

closed 2024-03-05 by jwhui