openthread/openthread #9792: CLI unresponsive after multi-count ping command

bertoldi-silabs, 2024-01-23:
When issuing a ping command with a [count] argument greater than 1, the ICMP request is sent to the destination node and the reply is sent to the originating node, but the `>` CLI prompt does not return after the ping metrics output is printed, and additional commands are unable to be issued to the originating node. 

This behavior was observed on commit f75e2bb567bd4cbe7a0c8f79979e06ebaf0e4c1e, using the simulation platform.

To reproduce, follow the instructions of the [Simulating a Thread network with OpenThread](https://openthread.io/codelabs/openthread-simulation-posix#1) codelab up to the [Verify the network](https://openthread.io/codelabs/openthread-simulation-posix#2) step. When issuing the `ping` command, add a size, count, and interval argument, with a count >1, such as the following:
`> ping fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad 16 2 5000`

For issued ping commands with an unspecified count, or with a specified count of 1, the ICMP echo request is sent from the originating node, a reply is received from the target node, the originating node prints round trip metrics, the phrase "Done" is printed, and then the CLI `>` prompt is displayed. From this point, further commands can be issued from the CLI. I would expect similar behavior when using a count argument larger than 1, but the "Done" and prompt are not printed after the metrics.

```
> ping fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad
16 bytes from fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad: icmp_seq=1 hlim=64 time=2ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 2/2.0/2 ms.
Done
> ping fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad 1000 1 5000
1008 bytes from fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad: icmp_seq=2 hlim=64 time=42ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 42/42.0/42 ms.
Done
> ping fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad 16 2 5000
24 bytes from fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad: icmp_seq=3 hlim=64 time=1ms

```
I have not found a way to recover the affected node when in this state. The only workaround I have been able to use is closing the process and opening a new simulated node, as I am not able to issue a `factoryreset` in this state.

jwhui, 2024-01-24:
In your ping command:
```
ping fdbc:2581:4c06:dd9c:189:fbd6:2357:c9ad 1000 1 5000
```
The `interval` argument is in seconds and set to `5000`, which is quite a long time to wait. Try using a smaller value.

From the [`ping`](https://github.com/openthread/openthread/tree/main/src/cli#ping-async--i-source--m-ipaddr-size-count-interval-hoplimit-timeout) CLI documentation:
> interval: The interval between two consecutive ICMPv6 Echo Requests in seconds.

bertoldi-silabs, 2024-01-24:
Reducing the interval did address the "problem". That certainly makes sense why it would seem unresponsive if it is waiting for 5000 seconds instead of milliseconds!

Thank you for pointing that out.

closed 2024-01-24 by bertoldi-silabs