openthread/openthread #9800: test_srp_auto_host_address.py 5 failing /100 runs

librasungirl, 2024-01-25:
**Describe the bug** Unexpected test_srp_auto_host_address.py fail for https://github.com/openthread/openthread/pull/9795

**To Reproduce** Information to reproduce the behavior, including:

Local test on the PR code
```
MULTIPLY=100 ./script/test cert_suite ./tests/scripts/thread-cert/test_srp_auto_host_address.py 
PASS 95 FAIL 5 ./tests/scripts/thread-cert/test_srp_auto_host_address.py
```

Failing log
[test_srp_auto_host_address_72.log](https://github.com/openthread/openthread/files/14051276/test_srp_auto_host_address_72.log)
[image]

The pcap indicates the SRP update will select the manual added address `fd00:1:2:3:4:5:6:7` as source which is not an on-mesh address, so there won't be response, which explains the error step `AssertionError: 'Refreshing' != 'Registered'`

librasungirl, 2024-01-25:
A quick local validation with a build w/o https://github.com/openthread/openthread/pull/9634 1000/1000 PASS

librasungirl, 2024-01-25:
Manually added address is always preferred, and has global scope by default; MeshLocal address is of RealmLocal scope and init as non-preferred by default.
When SelectSource, in the for() loop, if Manually added address is after MeshLocal address in the interface's unicast address array, the Manually added address will win as it's preferred.

Here with https://github.com/openthread/openthread/pull/9801 which init MeshLocal address as preferred, this test can pass 1000/1000 runs。

Any more context to mark mesh local as non preferred？thoughts on the fix? @abtink

abtink, 2024-01-25:
Thanks @librasungirl for debugging this. 👍. 

I had noticed the frequent test failures and `AssertError(kRefresh)` state recently and was planning to investigate when I had some time. I suspected a timing issue where we might be checking for registration too early. Glad you found the root cause! 😃 

On https://github.com/openthread/openthread/pull/9801 and marking mesh local as preferred, it  makes sense to me, but I wonder why was it not set as such before?
- If I am not mistaken, this has been always the case in OT stack? Was it an omission or may be intentional? 
- I wonder if marking mesh-local as preferred may cause issues on platform side where Thread interface addresses are mirrored on `wpan` interface (having mesh-local addresses be marked as preferred?)
- I feel safer not change it without knowing more about the reasons behind this?

Related to PR https://github.com/openthread/openthread/pull/9634:
- It is a correct/important fix for a bug in `SelectSource()`.
- But I think the unfortunate side-effect of this bug was that it masked this situation making the `srp_auto_address` test work reliably.

Some ideas:
- We can update the test itself. It is adding a new address without adding a prefix for it in netdata? So I think the address is practically not usable/reachable? We can add a (non-SLAAC on-mesh) prefix for it in the test case?
- We can add a new rule to `SelectSource()` to always pick mesh-local source if dest is also mesh-local? 
   -  We already have a similar rule which picks RLOC when dest is RLOC. 
   - This seems reasonable to me and may be a good enhancement to add anyways. Thoughts?

abtink, 2024-01-26:
I did a bit of checking through the code/commit history. The mesh-local addresses were originally marked as preferred but this was changes in this PR (from ~May 2021)
- https://github.com/openthread/openthread/pull/6532

> This commit marks anycast addresses and mesh-local addresses as IPv6 non-preferred addresses.
> 
> The background is host IPv6 stack(such as lwIP, Linux kernel and so on) cannot tell these addresses to OMR addresses from the scope id. By marking these addresses non-preferred, the host IPv6 stack will not select these addresses as the source address of application layer traffic unless explicitly assigned.

jwhui, 2024-01-26:
A related issue:
- https://github.com/openthread/openthread/issues/8443

jwhui, 2024-01-30:
The proposed approach in https://github.com/openthread/openthread/issues/9800#issuecomment-1912684440 sounds good to me!

abtink, 2024-01-30:
Submitted 
- https://github.com/openthread/openthread/pull/9815

abtink, 2024-02-13:
We can close this as #9815 is merged.

closed 2024-02-13 by abtink