openthread/openthread #9820: Data poll delayed when receiving fragmented packets

CodingRays, 2024-02-01:
**Describe the bug**

We discovered this bug while working on #9763 . It appears that while receiving a fragmented packet the data poll for the next fragments can randomly be delayed by approximately 100ms. This matches the data poll timeout however no previous data poll was sent.
[image]

We have tested this inside a Faraday cage to eliminate any external causes and the bug seems to be easily reproducible. Out of 20 512byte packets 11 had at least 1 such delayed fragment.

After further testing we found that this only happens when logging is enabled at debug level. Which leads me to believe that it could be related to some very tight timeout. But how that would be possible is unclear to me. Further the bug only appears when attached as a SED not as a SSED.

**To Reproduce**

Git commit id: ce8bbfe05a888632b3908c2e0f330e150ed7e456 (Tested with multiple older commits as well)
Platform: FTD: nrf52840 dongle, MTD: nrf52840 dev kit
Topology: MTD attached as a SED to the leader
Other: poll interval set to 10s

**Expected behavior**

A data poll should be sent without waiting 100ms.

**Console/log output**

Unable to obtain due to messages overriding each other when logging at level 5. Logging at level 4 does not trigger the bug.

abtink, 2024-02-01:
It is been a while since I last read the `DataPollSender` and `Mac` code, but checking the code I dont see anything obvious that would lead to such a behavior.
- The `DataPollSender` should immediately `SendDataPoll()` from `ProcessRxFrame()` if it sees `aFrame.GetFramePending()` on the received frames.
- `Mac` should finish `kOperationWaitingForData` from `HandleReceivedFrame()` 

> Unable to obtain due to messages overriding each other when logging at level 5. Logging at level 4 does not trigger the bug.

Gotta love a [Heisenbug](https://en.wikipedia.org/wiki/Heisenbug#:~:text=In%20computer%20programming%20jargon%2C%20a,one%20attempts%20to%20study%20it.) 😄

jwhui, 2024-02-01:
@CodingRays , can you attach the pcap file?

CodingRays, 2024-02-02:
So heres a minimal setup to reproduce the issue i found. This uses only dongles:

From a fresh clone of the nrf52840:
Build using `./script/build nrf52840 USB_trans -DOT_BOOTLOADER=USB -DOT_LOG_LEVEL_DYNAMIC=ON`
On the mtd following commands:
```
log level 5
mode -
pollperiod 10000
```
On the ftd to test
```
udp send <mleid> 5000 -s 512
```
Since the bug appears ~50% of the time a few transmission might be required but it should trigger very reliably.

I sadly lost the pcap file from the image but here is a capture from a test using the above setup.
The bug appears between packet 172 and 173. Network key: `6ce43b5b088aa1a10db563e803d1a28b`
[data_poll_dongle.zip](https://github.com/openthread/openthread/files/14144682/data_poll_dongle.zip)

jwhui, 2024-03-05:
Looking at the packet trace again, it appears that the ACK was sent with the Frame Pending bit set, but no data frame transmission followed. The default Data Poll Timeout is set to 100ms:
https://github.com/openthread/openthread/blob/44c39060cd63bc478979de621719400afe3c0326/src/core/config/mac.h#L597

The Data Sequence Number jumps from 125 in frame 169 to 127 in frame 175. So it appears that it tried to send a data frame. But without more logging from the sender, it is difficult to determine exactly the root cause. It could be due to CCA failures. It could be that the frame was transmitted but a collision occurred. Both of which would be legitimate reasons for packet loss.

We may want to consider reducing the timeout period to something smaller.

CodingRays, 2024-03-07:
I can try to get the logs.

But its hard to believe its completely unrelated to the logging due to just how consistent it is. I have not had a single such failure during all my testing while logging was disabled while with it enabled its 50% of all ip packets / ~16% of all fragments.

CodingRays, 2024-03-08:
Ok so ive spent a while testing and i was no longer able to reproduce the bug. It did occur once and as previously when it did happen it was reliable until i changed the configuration. Here 10s after the change i continued testing and it immediately stopped. However without knowing any conditions to trigger it its hard to investigate. I also couldnt get logs from that test. One thing to note though is that now the wireshark logs show the data poll being sent but no response from the parent.

I am using different hardware now though and will try and test with the original hardware hopefully soon and get back after i do that.

CodingRays, 2024-03-22:
Alright i cant reproduce the bug anymore event with different hardware. I have no idea what changed.
I guess the issue can be closed then.

jwhui, 2024-06-10:
Closing stale issue.

closed 2024-06-10 by jwhui