openthread/openthread #9884: Segmentation fault occurs in CLI - CoAP Example when using block-wise.

eeumay, 2024-02-28:
**Describe the bug** A clear and concise description of what the bug is.
Segmentation fault occurs when using [CLI Example](https://github.com/openthread/openthread/tree/37fef5ed547421e627b69193798d2556b3cebe19/examples/apps/cli). I followed the [readme](https://github.com/openthread/openthread/blob/37fef5ed547421e627b69193798d2556b3cebe19/examples/apps/cli/README.md) to build and execute the axample and everything was well. And then I continued to run coap following another [readme](https://github.com/openthread/openthread/blob/37fef5ed547421e627b69193798d2556b3cebe19/src/cli/README_COAP.md). Coap worked well untill get with "block-" crashed the program.
Crash log is in the end of **Console/log output**.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
8bc25042ba5cde20605eec2a329c0dff575303f4 (thread-reference-20230706)
2. IEEE 802.15.4 hardware platform
simulation 
OS: Linux version 5.15.0-92-generic (buildd@lcy02-amd64-005) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #102-Ubuntu SMP Wed Jan 10 09:33:48 UTC 2024.
3. Build steps
[readme](https://github.com/openthread/openthread/blob/37fef5ed547421e627b69193798d2556b3cebe19/examples/apps/cli/README.md)
4. Network topology
2 devices build as FTD. One is leader and another is child.
 [CLI Example](https://github.com/openthread/openthread/tree/37fef5ed547421e627b69193798d2556b3cebe19/examples/apps/cli)

**Expected behavior** A clear and concise description of what you expected to happen.
Execute coap get with "block-" correctly or report a error.

**Console/log output** If applicable, add console/log output to help explain your problem.
Build:
ubuntu@xxxxx:~/workspace/openSource/openthread$ git status
HEAD detached at thread-reference-20230706
nothing to commit, working tree clean
ubuntu@xxxxx:~/workspace/openSource/openthread$ ./script/cmake-build simulation
+ OT_CMAKE_NINJA_TARGET=
+++ dirname ./script/cmake-build
++ cd ./script/..
++ pwd
+ OT_SRCDIR=/home/ubuntu/workspace/openSource/openthread
+ readonly OT_SRCDIR
+ OT_PLATFORMS=(simulation posix android-ndk)
+ readonly OT_PLATFORMS
+ OT_POSIX_SIM_COMMON_OPTIONS=("-DOT_ANYCAST_LOCATOR=ON" "-DOT_BORDER_AGENT=ON" "-DOT_BORDER_AGENT_ID=ON" "-DOT_BORDER_ROUTER=ON" "-DOT_CHANNEL_MANAGER=ON" "-DOT_CHANNEL_MONITOR=ON" "-DOT_COAP=ON" "-DOT_COAPS=ON" "-DOT_COAP_BLOCK=ON" "-DOT_COAP_OBSERVE=ON" "-DOT_COMMISSIONER=ON" "-DOT_COMPILE_WARNING_AS_ERROR=ON" "-DOT_COVERAGE=ON" "-DOT_DATASET_UPDATER=ON" "-DOT_DHCP6_CLIENT=ON" "-DOT_DHCP6_SERVER=ON" "-DOT_DIAGNOSTIC=ON" "-DOT_DNSSD_SERVER=ON" "-DOT_DNS_CLIENT=ON" "-DOT_ECDSA=ON" "-DOT_HISTORY_TRACKER=ON" "-DOT_IP6_FRAGM=ON" "-DOT_JAM_DETECTION=ON" "-DOT_JOINER=ON" "-DOT_LOG_LEVEL_DYNAMIC=ON" "-DOT_MAC_FILTER=ON" "-DOT_NEIGHBOR_DISCOVERY_AGENT=ON" "-DOT_NETDATA_PUBLISHER=ON" "-DOT_NETDIAG_CLIENT=ON" "-DOT_PING_SENDER=ON" "-DOT_RCP_RESTORATION_MAX_COUNT=2" "-DOT_REFERENCE_DEVICE=ON" "-DOT_SERVICE=ON" "-DOT_SNTP_CLIENT=ON" "-DOT_SRP_CLIENT=ON" "-DOT_SRP_SERVER=ON" "-DOT_UPTIME=ON")
+ readonly OT_POSIX_SIM_COMMON_OPTIONS
+ main simulation
+ [[ 1 == 0 ]]
+ local platform=simulation
+ echo simulation posix android-ndk
+ grep -wq simulation
+ shift
+ local_options=()
+ local local_options
+ options=('-DOT_SLAAC=ON')
+ local options
+ case "${platform}" in
+ local_options+=("-DOT_LINK_RAW=ON" "-DOT_DNS_DSO=ON" "-DOT_DNS_CLIENT_OVER_TCP=ON" "-DOT_UDP_FORWARD=ON")
+ options+=("${OT_POSIX_SIM_COMMON_OPTIONS[@]}" "${local_options[@]}")
+ options+=("-DOT_PLATFORM=${platform}")
+ options+=("$@")
+ build simulation -DOT_SLAAC=ON -DOT_ANYCAST_LOCATOR=ON -DOT_BORDER_AGENT=ON -DOT_BORDER_AGENT_ID=ON -DOT_BORDER_ROUTER=ON -DOT_CHANNEL_MANAGER=ON -DOT_CHANNEL_MONITOR=ON -DOT_COAP=ON -DOT_COAPS=ON -DOT_COAP_BLOCK=ON -DOT_COAP_OBSERVE=ON -DOT_COMMISSIONER=ON -DOT_COMPILE_WARNING_AS_ERROR=ON -DOT_COVERAGE=ON -DOT_DATASET_UPDATER=ON -DOT_DHCP6_CLIENT=ON -DOT_DHCP6_SERVER=ON -DOT_DIAGNOSTIC=ON -DOT_DNSSD_SERVER=ON -DOT_DNS_CLIENT=ON -DOT_ECDSA=ON -DOT_HISTORY_TRACKER=ON -DOT_IP6_FRAGM=ON -DOT_JAM_DETECTION=ON -DOT_JOINER=ON -DOT_LOG_LEVEL_DYNAMIC=ON -DOT_MAC_FILTER=ON -DOT_NEIGHBOR_DISCOVERY_AGENT=ON -DOT_NETDATA_PUBLISHER=ON -DOT_NETDIAG_CLIENT=ON -DOT_PING_SENDER=ON -DOT_RCP_RESTORATION_MAX_COUNT=2 -DOT_REFERENCE_DEVICE=ON -DOT_SERVICE=ON -DOT_SNTP_CLIENT=ON -DOT_SRP_CLIENT=ON -DOT_SRP_SERVER=ON -DOT_UPTIME=ON -DOT_LINK_RAW=ON -DOT_DNS_DSO=ON -DOT_DNS_CLIENT_OVER_TCP=ON -DOT_UDP_FORWARD=ON -DOT_PLATFORM=simulation
+ local platform=simulation
+ local builddir=build/simulation
+ shift
+ mkdir -p build/simulation
+ cd build/simulation
+ cmake -GNinja -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DOT_COMPILE_WARNING_AS_ERROR=ON -DOT_SLAAC=ON -DOT_ANYCAST_LOCATOR=ON -DOT_BORDER_AGENT=ON -DOT_BORDER_AGENT_ID=ON -DOT_BORDER_ROUTER=ON -DOT_CHANNEL_MANAGER=ON -DOT_CHANNEL_MONITOR=ON -DOT_COAP=ON -DOT_COAPS=ON -DOT_COAP_BLOCK=ON -DOT_COAP_OBSERVE=ON -DOT_COMMISSIONER=ON -DOT_COMPILE_WARNING_AS_ERROR=ON -DOT_COVERAGE=ON -DOT_DATASET_UPDATER=ON -DOT_DHCP6_CLIENT=ON -DOT_DHCP6_SERVER=ON -DOT_DIAGNOSTIC=ON -DOT_DNSSD_SERVER=ON -DOT_DNS_CLIENT=ON -DOT_ECDSA=ON -DOT_HISTORY_TRACKER=ON -DOT_IP6_FRAGM=ON -DOT_JAM_DETECTION=ON -DOT_JOINER=ON -DOT_LOG_LEVEL_DYNAMIC=ON -DOT_MAC_FILTER=ON -DOT_NEIGHBOR_DISCOVERY_AGENT=ON -DOT_NETDATA_PUBLISHER=ON -DOT_NETDIAG_CLIENT=ON -DOT_PING_SENDER=ON -DOT_RCP_RESTORATION_MAX_COUNT=2 -DOT_REFERENCE_DEVICE=ON -DOT_SERVICE=ON -DOT_SNTP_CLIENT=ON -DOT_SRP_CLIENT=ON -DOT_SRP_SERVER=ON -DOT_UPTIME=ON -DOT_LINK_RAW=ON -DOT_DNS_DSO=ON -DOT_DNS_CLIENT_OVER_TCP=ON -DOT_UDP_FORWARD=ON -DOT_PLATFORM=simulation /home/ubuntu/workspace/openSource/openthread
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
[... 342 lines cut ...]
85a27ed979449641df72115c63d6d6e3
0b29fa1c1e1a8a8c9be5f7a5fedc2ac3
ce250a2631871a983c97700aa8a2ea90
ab758c82db876ffac1bd0355ddb08722
e98daf45edb58467748dcdea69d557fc
22ca22842a92c388b70c154ee946a868
71fcff87619e59eacd9db58579a3e046
bc590e540a889eebdc0792c78ed3caf5
8b1cd80666dceaf1431766a661fa52c4
2aac0808dd2c9a7d6361bdb93ad695fb
69a32146fedc42d1953cf20dc0fed836
95899d43d5ec7ec26a59f4365c8f0748
884a17466e0df73e1ba901342702e278
899b261524e9346b9242b6134a98dd75
8512b1464248f3045374e893f4e7eed3
10c17e63c710e8fabb452402e17192da
2c7bae2279567ef5ee5e6190eb520879
ffe28c2a5e959f83e1b99597b427cfd3
29e9b93f47e1074e2dacc8a5b5587838
1f9d6577ff55d1fae91227ea1e608feb
0c26057477bbb9b95efaf0954c1ea145
ce940aSegmentation fault (core dumped)

**Additional context** Add any other context about the problem here.
I'm having trouble using coap block-wise in [gecko_sdk](https://github.com/SiliconLabs/gecko_sdk) with Silicon Labs EFR32 boards. And I tried the Example ot-cli-ftd in Simplicity Studio v5 and found block-wise work not correctly and break down before some intermediate request. Next, I tried simulate cli example. Finally, I found this.

eeumay, 2024-03-01:
I may have fixed this issue.
Here is diff:
```
ubuntu@xxxxx:~/workspace/openSource/openthread$ git diff
diff --git a/src/core/coap/coap.cpp b/src/core/coap/coap.cpp
index 696d0fc83..64e148792 100644
--- a/src/core/coap/coap.cpp
+++ b/src/core/coap/coap.cpp
@@ -655,7 +655,7 @@ Error CoapBase::PrepareNextBlockRequest(Message::BlockType aType,
         }
 
         // Copy option
-        SuccessOrExit(error = iterator.ReadOptionValue(&optionBuf));
+        SuccessOrExit(error = iterator.ReadOptionValue(optionBuf));
         SuccessOrExit(error = aRequest.AppendOption(optionNumber, iterator.GetOption()->GetLength(), &optionBuf));
     }
```
ReadOptionValue accepts a void* or &uint64_t type，and optionBuf is a uint64_t type. Obviously, this is a typo.

jwhui, 2024-03-04:
@eeumay , thanks for raising this issue and working to resolve it. Would you like to submit a PR with the fix?

eeumay, 2024-03-05:
> @eeumay , thanks for raising this issue and working to resolve it. Would you like to submit a PR with the fix?

I'm on it. I will submit a PR and I'm waiting for GitHub Actions. But for copying option correctly, optionBuf should be uint8_t array. I'm not sure how to define a fixed langth adapting to an uncertain length due to the theoretical max is too big, and I decide to use the max length of the uri path which is necessary. Do you have any advice?

jwhui, 2024-03-05:
> I'm on it. I will submit a PR and I'm waiting for GitHub Actions. But for copying option correctly, optionBuf should be uint8_t array. I'm not sure how to define a fixed langth adapting to an [...]

@eeumay , good observation. I believe the length issue was present in the original contribution:
- #4247 

@abtink , who last updated the `ReadOptionValue()` method, any thoughts on how to address?

abtink, 2024-03-05:
Using a fixed (max size) array can work but would be limited.

A more flexible (but more complex) approach can be to add a new `AppendOption()` flavor that directly copies the option value from a passed `Option::Iterator`. It can then copy the bytes from one message to another using `AppendBytesFromMessage()`, eliminating the need for an intermediate buffer.

eeumay, 2024-03-06:
@jwhui @abtink  Thanks for your advice. 
I submit a PR to fix copying option.[#9894](https://github.com/openthread/openthread/pull/9894)
But it still failed that Coap parsed the previous old request (in `CoapBase::PrepareNextBlockRequest()`), because `CoapBase::FindRelatedRequest()` did return a Message appending metaData, which cannot be directly iterated. In this case, `Option::Iterator::Read()` would reach metaData and probably return `kErrorParse`. 
I will open another issue to talk about it.

closed 2024-03-22 by jwhui